fairecasoimeme/ZiPulses #2: support de ZiPulses dans Zigbee2mqtt

jeanaaaa, 2023-03-23:
Est-ce que le support de ZiPulses dans Zigbee2mqtt est prévu (même en tant que convertisseur externe) ?

fairecasoimeme, 2023-03-23:
Voici la procédure à suivre pour éviter d'attendre la sortie et avoir la compatibilité Z2M sous HA : 

1- Dans le dossier de votre configuration de zigbee2mqtt, créer un dossier avec "converters"
2- Télécharger le fichier [lixee.js](https://github.com/ralmn/zigbee-herdsman-converters/blob/014fc397cd852a67b71e466f50b3989c90b77064/devices/lixee.js)
3- Modifier dans les require les "../" par "zigbee-herdsman-converters/"
4- Placer le fichier nouveau lixee.js dans le dossier "converters"
5-  Dans le fichier configuration.yaml ajouter:

```
external_converters:
  - converters/lixee.js

```
6- Relancer zigbee2mqtt

closed 2023-03-24 by jeanaaaa

sdalu, 2023-04-01:
Dispo depuis aujourd'hui dans la version 1.30.3.

Mais, lorsque je demande reporting pour: endpoint=1, cluster=seMetering, attribute=currentSummDelivered
j'obtient le message d'erreur:

`Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x00158d0009520282/1 seMetering([{"attribute":"currentSummDelivered","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')'`

De meme pour endpoint=1, cluster=Temperature, attribute=measuredValue
`
Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting 0x00158d0009520282/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'SOFTWARE_FAILURE')'
`