fairecasoimeme/ZiPulses #3: Le compteur ne prend pas en compte le facteur multiplicateur

jeanaaaa, 2023-03-24:
**Problème :**
Le compteur est incrémenté d'une unité au lieu de prendre en compte le facteur multiplicateur

**Contexte :**
- Configuration via Zigbee2mqtt
- J'utilise ZiPluses avec le compteur Gazpar et j'ai positionné le facteur multiplicateur à 10 et l'unité à L.

**Tests effectués sans succès :**
- Réinitialiser ZiPluses avec le bouton "Reset"
- Ré-appairer ZiPluses 
- Retirer la pile
- Changer le facteur multiplicateur à une autre valeur

**Données remontées via MQTT :**
Malgré le facteur multiplicateur configuré sur 10, le compteur "energy" n'est incrémenté que d'une valeur à chaque impulsion du compteur Gazpar.
```
Received 08:30:43

    QoS: 0
    Payload:

    divisor: 2
    energy: 5
    linkquality: 5
    multiplier: 10
    temperature: null
    unitOfMeasure: L (litre)
    voltage: null

Received 08:30:43

    QoS: 0
    Payload:

    divisor: 2
    energy: 6
    linkquality: 5
    multiplier: 10
    temperature: null
    unitOfMeasure: L (litre)
    voltage: null

```

fairecasoimeme, 2023-03-24:
Oui, c'est en cours de dev pour la prise en compte des facteurs multiplicateur / diviseur sur Z2M.

Il existe 2 solutions : 
- Soit Z2M fait lui même le calcul
- Soit Z2M va chercher le cluster : 0x0702 attribut : 0x0100 qui intègre déjà le calcul.

Sur Z2M, energy ne représente que l'index impulsionnel

Gonioul, 2023-04-04:
Pour moi même en insistant ça ne semble pas marcher.
De toutes les façons je n'arrive pas à changer l'unité, donc autant faire appel à des templates...

fairecasoimeme, 2023-04-05:
Bizarre avez-vous les encarts verts qui apparaissent lors de la reconfiguration ?

Gonioul, 2023-04-06:
Non, je n'ai pas eu l'impression...
Mais tant que je ne peux pas changer l'unité (ça revient systématiquement sur kWh après la sélection) ce n'est pas blocant, je me débrouille avec les templates.