fairecasoimeme/ZiPulses #4: Configuration avec Home Assistant 2023.3.6 et Zigbee2Mqtt 1.30.2-1

Gonioul, 2023-03-28:
Bon,
Mon chemin de croix se termine, c'est un peu la pêche aux infos à droite à gauche, mais si ça peut aider...

Impossible d’appairer le ZiPulse avec la version actuelle de Zigbee2Mqtt sans ajouter le converter:
https://github.com/fairecasoimeme/ZiPulses/issues/2
(à noter que le module a crashé une fois après cet ajout)

Ensuite, le capteur n'est pas proposé dans le dashboard énergie (problème d'unité je suppose), le multiplicateur ne fonctionne pas, etc..

Je pars du principe que:
- 1 impulsion sensor.gazpar_energy = 10 L
- 1 L = 0.01 m³

En approximant les prix et l'altitude:
- 1 m³ de gaz naturel = 10 kWh

Et donc:
- 0.01 m³ de gaz naturel = 0.1 kWh
- 1 L de gaz naturel = 0.1 kWh

Au final:
- 1 impulsion = 1 kWh

Et donc on se contente de customiser le sensor 'sensor.gazpar_energy':

```
homeassistant:
  customize_glob:
    sensor.gazpar_energy:
      device_class: energy
      unit_of_measurement: 'kWh'
```

ellmout, 2023-03-29:
Une impulsion = 0,10 kWh
Car une impulsion de gazpar = 10 L
Dans 1 m3, il y a 1 000 L
1 impulsion x 10L / 1000L pour avoir des m3 x 10 environ pour le coefficient = 0,1 kWh

Gonioul, 2023-03-29:
Effectivement j'ai été trompé par les données gdrf, mais bon pour l'instant le coefficient ne marche pas...

PEGARifrouf, 2023-03-29:
Bonjour,

Dans l'idée j'aurais créé un input number dans HA pour régler le coefficient de conversion (en attendant de pouvoir le régler directement avec Z2m)

Ensuite avec un sensor template on peut faire un conversion plus precise impulsion => kwh

En récupérant ici le facteur de conversion de votre commune :
[Lien vers GRDF](https://www.grdf.fr/particuliers/coefficient-conversion-commune)

Je reçois le Zipulse dans quelques jours je viendrais poster le template si ça marche bien.

ellmout, 2023-03-29:
Voici mes templates pour avoir en m3 et en kWh, qu'on peut ajouter au tableau energie.

```yaml
template:
   - sensor:
      - name: "Gaz Energy"
        unit_of_measurement: 'kWh'
        state: "{{ states('sensor.zipulses_energy') | int * 10 / 1000 * 10.92 | round(3) }}"
        device_class: energy
        state_class: total_increasing
   - sensor:
      - name: "Gaz Volume"
        unit_of_measurement: 'm³'
        state: "{{ states('sensor.zipulses_energy') | int * 10 / 1000 | round(3) }}"
        device_class: gas
        state_class: total_increasing
```

10.92 = mon facteur de conversion, il est aussi sur sa facture de Gaz

PEGARifrouf, 2023-03-29:
Petite question, vous avec écrit votre converter vous même pour Z2M ?
Je n'ai pas vu de demande d'ajout sur leur GitHub.

ellmout, 2023-03-29:
fairecasoimeme a donné une procédure temporaire dans une issue mais elle est cloturée :

https://github.com/fairecasoimeme/ZiPulses/issues/2#issuecomment-1481206759

au 1er avril, le converter sera dans z2m.

Gonioul, 2023-03-30:
Ah... Mais le coefficient de conversion n'arrête pas de changer... 11.15, 11.23, 11.26 sur la dernière facture.
Et impossible d'ajouter une entité en m3 sur le dashboard énergie.

PEGARifrouf, 2023-03-30:
Pour le coefficient j'ai pris la valeur moyenne trouvée sur le site de GRDF et je compte la changer rarement.
Le template de sensor fourni plus haut fonctionne, les données s'importent dans le dashboard.

Gonioul, 2023-03-30:
Il semble que dans les versions récentes de HA, les m³ ne soient plus acceptés pour le gaz.

PEGARifrouf, 2023-04-02:
Hello,
Je viens de vérifier, c'est bien un template en m3 qui remonte dans le dashboard énergie dans ma config.

Gonioul, 2023-04-02:
J'ai pas le choix, pas le template en volume, et le popup ne parle pas de m³.

[image]

Est-ce que tu n'aurais pas des warnings dans les logs système te disant que les m³ ne sont pas une unité valide?

Sinon avec le nouveau Zigbee2Mqtt du 1er avril, plus besoin du convertisseur, et avec la miniature qui va bien.

PEGARifrouf, 2023-04-04:
Non pas de warning dans les log disant que ça va changer, par contre j'ai bien l'unité qui s'affiche sur le formulaire:
[image]
La note dit bien qu'on ne peut pas mélanger quantité d'énergie et volume. N'as tu pas un autre capteur déjà configuré en quantité d'énergie ?

Gonioul, 2023-04-04:
Ah bah forcément j'ai un zlinky aussi :)

Gonioul, 2023-04-04:
Bizarre, j'ai retiré le zlinky et le gas_energy, et j'ai bien pu ensuite remettre le zlinky et le gas_volume... Va comprendre...

PEGARifrouf, 2023-04-05:
J'ai le Zlinky en Kwh aussi