fairecasoimeme/ZiPulses #5: Pourcentage batterie?

Gonioul, 2023-03-30:
Je sais que le ZiPulses peut être aussi alimenté par autre chose, mais dans le cas où il est alimenté par une pile, il n'y aurait pas moyen d'avoir d'autres informations que la tension?
Zigbee2Mqtt est incapable de donner le pourcentage.

[image]

fairecasoimeme, 2023-04-02:
Pour le moment pas de remontée d'un pourcentage, uniquement la tension car compliqué de faire un pourcentage sur une pile lithium mais en plus lorsqu'il peut y avoir plusieurs sources d'alimentation.
Cependant, je le garde comme possible amélioration du firmware

Gonioul, 2023-04-02:
Et en demandant à Zigbee2Mqtt de faire le calcul, ça serait peut-être plus simple?

fairecasoimeme, 2023-10-18:
J'ai rajouté le pourcentage de batterie dans la version 3.0.
les données (temperature / voltage / pourcentage) remonte toutes les 2 heures mais aussi 1 minute après premier démarrage

closed 2023-10-18 by fairecasoimeme