fairecasoimeme/ZiPulses #6: ZiPulse - compteur eau Elster V200 + Honneywell PR6

paugebertrand, 2023-04-02:
Bonjour,

J'utilise le ZiPulses sur mon compteur d'eau Elster V200. J'ai connecté l'Emetteur d'impulsion ELSTER PR6 qui fournie une impulsion par litre. L'impulsion est un créneau bas de 80ms. Le ZiPulse ne détecteur pas cette impulsion (pas de flash bleu).
Est-ce parce que le pulse est trop court ? si oui, quelle est la durée minimum détectable par le ZiPulse ?
(la sortie du Elster PR6 est un transistor MOS open-drain avec une résistance 100 ohms en série)

Je ne sais pas si cela est liée mais le compteur 'energy' reste bloqué sur N/A même en générant des impulsions valides avec un shunt sur l'entrée de ZiPulses (flash bleu visible).

Merci pour votre aide.

nicolas51270, 2023-04-02:
Salut. 

De mon côté je l’ai installé sur home assistant via z2mqtt. Avec un compteur d’eau de marque 
GIOANOLA avec capteur ILS

Pareil toutes les valeurs reste sur N/A ou sur aucune valeurs. je n’ai cependant pas de flash lorsque le compteur tourne (il fonctionne, testé avec un multimètre) et pas de flash également en shuntant les entrées. 

Tu l’utilise sur pile ou via une alimentation ?

j’ai l’impression que le lixee est capricieux

paugebertrand, 2023-04-02:
Bonjour,

Je me suis aperçu que la connexion au bornier n'était pas correcte. Le shunt ne marchait pas à tous les coups. J'ai soudé un fil directement du bornier au module pour assurer un bonne connexion. C'est peut être ton problème !
Je suis en alimentation sur pile.

nicolas51270, 2023-04-02:
Merci pour l’info. Tu a donc enlevé le bornier vert ? 

Je test dans l’après-midi voir déjà si cela résout le problème de flash lors des impulsions. 

Reste le problème que tout reste en N/A

[image]

gignonje, 2023-04-02:
> Bonjour,
> 
> Je me suis aperçu que la connexion au bornier n'était pas correcte. Le shunt ne marchait pas à tous les coups. J'ai soudé un fil directement du bornier au module radio pour assurer un bonne connexion. C'est peut être ton problème ! Je suis en alimentation sur pile.

Bonjour, même problème ici, c'est très compliqué de mon côté pour détecter une impulsion, shunter ne fonctionne que très rarement. Comment as-tu relié le bornier au module radio ? Sinon malheureusement ça sent le retour pour moi :(

paugebertrand, 2023-04-02:
J'ai soudé un fil entre le bornier et le module radio.
Je suspecte que la connexion casse quand on sert la vis du bornier.
[image]

nicolas51270, 2023-04-02:
Bingo. Après avoir soudé un fil rapidement entre le bornier et le module, j’ai bien la led qui indique réceptionner l’impulsion qui s’allume. 
merci paugebertrand. 

Me reste plus qu’à comprendre pourquoi tout reste sur N/A
[image]

paugebertrand, 2023-04-02:
Ravi de t'avoir aidé nicolas51270
Pour le reste, il n'y a que le concepteur qui pourra nous aider !

Pour info, au cas où ça peut aider au debug, j'ai également the Zlinky_TIC dans mon installation (qui fonctionne très bien).

D'après la description sur github :
ZiPulses reçoit des impulsions : 
    Le capteur est reveillé. Il incrémente l'index et l'envoie au coordinateur

En regardant le log (onglet Journaux dans le zigbee2mqtt/dashboard), je ne vois aucun message de ZiPulses quand de provoque une impulsion. Pourtant, la connexion est valide puisque je vois bien la confirmation quand j'applique mes paramètres de diviseur, multiplicateur, ...

fairecasoimeme, 2023-04-02:
Bonjour,

Bizarre pour le coup du pulse et du soudage. N'hésitez pas à m'envoyer un mail pour faire un échange standard.  

Pour le problème de non réception de l'information de l'impulsion, il y a deux possibilités : 
- La portée (ce qui ne semble pas être le cas pour @paugebertrand )
- Le bind du cluster qui n'est peut-être pas passé. En effet, il est indispensable d'appliquer un bind sur le cluster 0x702 et 0x402. Soit vous avez la possibilité de forcer un Bind avec Z2M, soit il faut supprimer puis réappairer le module.

nicolas51270, 2023-04-02:
> Bonjour,
>
> Bizarre pour le coup du pulse et du soudage. N'hésitez pas à m'envoyer un mail pour faire un échange standard.
>
> Pour le problème de non réception de l'information de l'impulsion, il y a [...]

Alors oui bizarre mais ça fonctionne comme ça. Soit comme le dit Bertrand une faiblesse lorsque l’on sert le bornier soit je pense à un défaut du circuit imprimé. 

Je reçois enfin bien les infos de comptage dans z2mqtt, après avoir supprimé et reapairer a plusieurs reprise le module, il m’a fallu me rapprocher de mon coordinateur aussi. 

L’antenne est peut être également un peut faible. A voir. 

Par contre le diviseur ou multiplicateur ne fonctionne pas, quoi qu’il arrive une impulsions incrémente de 1 mon compteur.

fairecasoimeme, 2023-04-02:
> Par contre le diviseur ou multiplicateur ne fonctionne pas, quoi qu’il arrive une impulsions incrémente de 1 mon compteur.

Oui je travaille sur le sujet.

nicolas51270, 2023-04-02:
> > Par contre le diviseur ou multiplicateur ne fonctionne pas, quoi qu’il arrive une impulsions incrémente de 1 mon compteur.
> 
> Oui je travaille sur le sujet.

Ok super. 

Vous êtes le créateur du module lixee?

fairecasoimeme, 2023-04-02:
> > > Par contre le diviseur ou multiplicateur ne fonctionne pas, quoi qu’il arrive une impulsions incrémente de 1 mon compteur.
> > 
> > 
> > Oui je travaille sur le sujet.
> 
> Ok super.
> 
> Vous êtes le créateur du module lixee?

oui

paugebertrand, 2023-04-02:
Merci fairecasoimeme.

Mon problème n'est pas la portée, 141lqi
J'essaie de supprimer et réappairer le module.

Quand est-il pour un pulse de 80ms comme expliqué dans mon premier post ?

nicolas51270, 2023-04-02:
> > > > Par contre le diviseur ou multiplicateur ne fonctionne pas, quoi qu’il arrive une impulsions incrémente de 1 mon compteur.
> > > 
> > > 
> > > Oui je travaille sur le sujet.
> > 
> > 
> > Ok super.
> > Vous êtes le créateur du module lixee?
> 
> oui

Enchanté,

je vous ai envoyé un mail via votre site 

https://faire-ca-soi-meme.fr

N’hésiter à revenir vers moi.

fairecasoimeme, 2023-04-02:
> Merci fairecasoimeme.
> 
> Mon problème n'est pas la portée, 141lqi J'essaie de supprimer et réappairer le module.
> 
> Quand est-il pour un pulse de 80ms comme expliqué dans mon premier post ?

En regardant la datasheet de votre compteur impulsion, j'ai bien peur qu'il ne soit pas compatible avec le ZiPulses.
En effet, votre compteur n'utilise pas un capteur ILS (lamelles métalliques) qui joue le rôle de contact magnétique. 
Il nécessite malheureusement une charge de contact ce qui n'est pas approprié avec le fonctionnement du ZiPulses

paugebertrand, 2023-04-02:
Je pense que le module détecte des fronts descendant et montant. 
Le transistor dans mon compteur d'impulsion fonctionne comme un ILS. J'ai d'ailleurs vérifié à l'oscilloscope.

Existe-t-il un filtrage numérique dans le modude Zigbee de type anti-rebonds qui filtrerait un pulse de 80ms ?

Gonioul, 2023-04-02:
Léger aparté, pour les tensions et températures qui sont à N/A, c'est normal au début, ce n'est transmis qu'occasionnellement (toutes les 2h si j'ai bien compris la doc)

gignonje, 2023-04-02:
> J'ai soudé un fil entre le bornier et le module radio. Je suspecte que la connexion casse quand on sert la vis du bornier. [image]

Je confirme, une fois le fil soudé entre le bornier et le module, l'impulsion est correctement détectée

fairecasoimeme, 2023-04-03:
> Enchanté,
>
> je vous ai envoyé un mail via votre site
>
> https://faire-ca-soi-meme.fr
>
> N’hésiter à revenir vers moi.

il vaut mieux passer directement par le mail contact[at]lixee.fr

fairecasoimeme, 2023-04-03:
Pour le problème des coefficients sur Z2M : 

https://github.com/fairecasoimeme/ZiPulses/blob/master/README.md#z2m

nicolas51270, 2023-04-04:
> Pour le problème des coefficients sur Z2M :
> 
> https://github.com/fairecasoimeme/ZiPulses/blob/master/README.md#z2m

Salut. 

Parfait cela fonctionne, 
Ce n’est pas que c’est compliqué à faire mais dommage que l’on ne puisse pas le faire directement via l’appareil

paugebertrand, 2023-04-04:
Bonjour fairecasoimeme,

Mon hypothèse s'est vérifiée, le pulse de 80ms doit être filtré par le debounce.
Je pense que le paramètre ci-dessous doit pouvoir agir sur ce filtrage :
     (app_pulses_buttons.c)
     /* Defines the number of bits sampled for a button debounce event. One sample taken for each bit set */
     #define APP_BUTTON_SAMPLE_MASK          (0x3)

Pour contourner, j'ai ajouter 20uF en parallèle ce qui provoque un pulse suffisamment long pour être comptabilisé.
Je teste pendant plusieurs jours pour voir si je ne rate pas de pulse et confirmer le workarround. Le bémol de ce workarround est qu'il va impacter la consommation.

fairecasoimeme, 2023-04-05:
En effet, En attendant, le condensateur est une solution, en espérant qu'il n'y est pas 2 pulses trop rapprochés. (Pas sur que ça impacte la durée de vie sur pile). De mon côté je vais investiguer pour réduire le temps du rebond et je vous tiens au courant.

En tous les cas, un énorme merci pour la contribution.

fairecasoimeme, 2023-04-05:
> > Pour le problème des coefficients sur Z2M :
> > https://github.com/fairecasoimeme/ZiPulses/blob/master/README.md#z2m
> 
> Salut.
> 
> Parfait cela fonctionne, Ce n’est pas que c’est compliqué à faire mais dommage que l’on ne puisse pas le faire directement via l’appareil

Malheureusement, je ne maitrise pas assez Z2M et cette procédure un peu étrange ne concerne pas les autres plugins. Je suis tombé sur cette procédure à force de faire des tests mais sur ZHA par exemple, pas de problème, lorsqu'on modifie un attribut, il est pris en compte de suite dans le logiciel. 

Pour tous les appareils sur pile, il est indispensable de les réveiller pour chaque action (Ce n'est pas propre au ZiPulses). C'est plus laborieux que les appareils sur secteur mais malheureusement pas le choix.

paugebertrand, 2023-04-05:
> En effet, En attendant, le condensateur est une solution, en espérant qu'il n'y est pas 2 pulses trop rapprochés. (Pas sur que ça impacte la durée de vie sur pile). De mon côté je vais investiguer [...]

Je suis ravi si j'ai pu contribuer.
Je conserve le workarround avec le condensateur jusqu'à l'amélioration de la gestion du rebond alors...

fairecasoimeme, 2023-04-06:
Bonjour,
Avant de mettre en ligne, pouvez-vous tester ce firmware qui améliore la gestion du rebond ?
[ZiPulses_v4.0.zip](https://github.com/fairecasoimeme/ZiPulses/files/11168571/ZiPulses_v4.0.zip)

paugebertrand, 2023-04-06:
J'essaie ce firmware ce week-end et je vous tiens au courant.

paugebertrand, 2023-04-08:
Je viens de mettre à jour mon ZiPulse.
Les impulsions sont bien vues et je n'ai pas d'impulsions ratées avec un fort débit d'eau. Je vérifierai dans quelques jours si j'en ai loupé.
Le workarround avec le condensateur n'est pas validé. J'ai perdue 600 impulsions sur la semaine.
Merci pour cette mise à jour.

paugebertrand, 2023-04-23:
Bonjour fairecasoimeme,
Comme discuté, je reviens avec des résultats. Je constate qu'en 14 jours, 8 impulsions sur 7080 ont été perdues.

fairecasoimeme, 2023-04-24:
Bonjour,

Merci beaucoup pour le retour. Je devrais recevoir un générateur de fonction afin de mieux développer la gestion des pulses du firmware.
Je vous tiendrai au courant et communiquerai sur la mise à jour du firmware

fairecasoimeme, 2023-04-27:
Bonjour,

J'ai modifié la gestion des pulses dans le nouveau firmware ici : 
https://github.com/fairecasoimeme/ZiPulses/releases/tag/v5.0

Normalement détection des pulses au dessus de 20ms

paugebertrand, 2023-04-30:
Bonjour fairecasoimeme,
Je viens de flasher la version 5.0, je vous tiens au courant...

paugebertrand, 2023-05-13:
Bonjour fairecasoimeme,
Comme discuté, je reviens avec des résultats. Je constate qu'en 13 jours, 7 impulsions sur 6149 ont été perdues.

fairecasoimeme, 2023-05-16:
Bonjour, Merci pour le retour, je vais mettre en route le générateur de fonctions pour pousser les tests et comprendre/trouver pourquoi il peut y avoir des impulsions perdus.
(7 impulsions sur 6149, ça fait 0.11 % d'erreurs)

JojoNenez, 2023-06-23:
Bonjour tout le monde.
Tout d'abord, bravo à Fairecasoitmeme pour son travail :-) Tout n'est pas parfait, mais il faut reconnaitre que c'est un sacré boulot...

Voici donc un petit retour sur mon expérience avec ce module :

- J'ai acheté ce module récemment et j'ai eu le même problème que certains ici, à savoir que la piste reliant le bornier Pulses au module Zigbee n'a pas tenue plus de deux vissages/dévissages avant de se couper. j'ai donc ponté celle-ci.
- J'ai aussi eu le problème de largeur d'impulsion : mon compteur est un "Maddalena MVM plus C" et je l'ai équipé de son capteur d'impulsions, un "Maddalena Flow Pulse" version 5 fils. L'ensemble fonction bien, mais il y a un hic... Pour un débit inférieur à 4 impulsions par secondes, la largeur de celle-ci est de 125ms. Pour un débit supérieur, elle est encore moins large. Donc avec mon module Lixee en version V4000-0003, ces impulsions ne sont pas prises en compte. 
J'ai fait le test sur table avec un générateur de signaux à la fréquence de 0.5hz en jouant sur la largeur de la "fermeture" de l'impulsion, c'est à dire sur la largeur de l'état 0V. Il en résulte qu'en dessous de 190ms, l'impulsion n'est pas prise en compte. en revanche, à partir de 200ms, ça fonctionne bien.
Je vais donc m'équiper pour flasher mon module en V5, en espérant que tout rentre dans l'ordre :-)

Merci encore pour votre travail.

paugebertrand, 2023-07-30:
Bonjour fairecasoimeme,
Avez-vous apporté des mises à jour depuis la v5 ?
De mon côté, j'ai toujours une petite erreur (0.14% sur 41312 litres) ce qui est acceptable.
Je pense que l'erreur peut aussi venir de l'Emetteur d'impulsion ELSTER PR6.
Est-ce que les autres utilisateurs ont cette erreur ?

fairecasoimeme, 2023-07-31:
Bonjour,
Non je n'ai pas encore pu faire des tests poussés avec le générateur. je vous tiendrai au courant.

paugebertrand, 2023-10-25:
Bonjour fairecasoimeme,
Avez-vous avancé sur ce sujet ?

fairecasoimeme, 2023-10-25:
Bonjour,
je suis en test actuellement : 
[image]
[image]
Pas de pertes pour le moment mais je laisse tourner

paugebertrand, 2023-10-26:
Bonjour,
J'attends les résultats avec impatience ;)
Pouvez-vous déterminer la largeur mini du pulse sans pertes ?
Je pourrais en déduire un débit max détectable.
Cordialement.

fairecasoimeme, 2023-10-27:
Bonjour,

Je suis allé jusqu'à 21952 impulsions avant la nuit. La largeur du pulse est de 50ms.
Je vais refaire les tests car il me manque un câble pour consolider et éviter les faux contacts. En effet, après une nuit et 68000 impulsions, il y a eu un décalage de plus de 1000 impulsions, ce qui ne me semble pas possible. J'avais déjà eu le souci car j'ai un câble qui fait à peine contact et avait provoqué des milliers de pulses (côté compteur/générateur)

paugebertrand, 2023-10-27:
Bonjour,
Le pulse de 50ms est un pulse bas ou haut ?
Quelle est la fréquence du signal de test ? 1/(2x50ms) = 10Hz?

fairecasoimeme, 2023-10-30:
Je vais lancer de nouveaux tests, voici les captures sur le setup :
[image]

Je suis passé à des pulses de 30ms de large toutes les secondes.  (Les pulses se déclenchent aux alentours de 25ms)

[image]

fairecasoimeme, 2023-11-03:
Voici le résultat des tests avec le setup ci-dessus : 

|Jour|Heure|Nb impulsions reçues (imp)|Nb de secondes depuis l'initialisation(sec)|Nb d'impulsions loupées|Remarques|
|----|------|---------------------|-------------------------------------|-------------------------|----------|
|J1|16:36:00| 0|0|0||
||17:06:00|1800|1800|0||
||17:18:00|2520|2520|0||
||17:46:00|4220|4220|0||
||19:06:00|9000|9000|0||
|J2|08:22:00|56756|56756|0||
||11:12:00|66944|66960|16|J'ai effectué un reset parce que je ne voyais plus HA bouger|
||13:24:00|74864|74880|16||
||18:12:00|92084|92100|16||

PS: Le firmware a été modifié pour remonter des trames zigbee (temperature / tension) toutes les 5 minutes pour augmenter les probabilités d'impulsions loupées.

paugebertrand, 2023-11-03:
C'est étonnant ce reset nécessaire du ZiPulses.
S'il n'y avait pas eu de reset manuel, aurait-il redémarrer tout seul ? Watchdog ?

JojoNenez, 2023-11-03:
Il veut dire un reset de Home Assistant je pense.

diliak, 2023-11-03:
Cela rejoindrait-il mon souci de "disparition" au bout de quelques jours ?
Je ne voyais plus "Domoticz" bouger ?

fairecasoimeme, 2023-11-03:
Non rien à voir ... c'est mon pc de test (ou HA) qui a freeze. J'aurai du d'abord redémarrer HA mais dans la panique j'ai reset le ZiPulses qui au final n'a perdu que 16 pulses dû au reset. Si c'était vraiment de la faute du ZiPulses, on aurait eu beaucoup plus de pertes d'impulsions

diliak, 2023-11-03:
OK.
En ce qui me concerne, je ne cherche pas à "incriminer" le Zipulses. Le souci potentiel que j'ai peut très bien venir d'une mauvaise gestion côté plugin. En tous cas, il fonctionne à priori bien depuis le 21 Octobre avec le nouveau firmware. Je fais quelques impulsions manuellement tous les jours.
Fin su HS ;)

paugebertrand, 2023-11-03:
Comment apparaît la version pour le FW v5 dans Zigbee2mqtt ?
j'ai : 
Date du firmware
20230222
Version du firmware
4000-0003

paugebertrand, 2023-11-03:
Si votre setup est toujours en place, pourriez-vous faire un essai avec une periode de 200ms, svp ? Cela simulera un débit de 5 l/s.

fairecasoimeme, 2023-11-07:
Bonjour,
Oui je peux refaire les tests avec une période de 200ms mais quel est l'objectif pour vous?

paugebertrand, 2023-11-07:
Bonjour,
Le but serait de vérifier que ZiPulses peut compter sans perte avec un débit de 5l/s.

fairecasoimeme, 2023-11-07:
ok j'essaie ASAP

paugebertrand, 2024-01-02:
Bonjour fairecasoitmeme,
Vous avez corrigé la détection de pulse pour convenir à 20ms et le décalage de comptage n'est pas visiblement pas dû à ZiPulse au regard de vos essais.
Je vais clore ce ticket.
Merci pour votre support.

closed 2024-01-02 by paugebertrand