fairecasoimeme/ZiPulses #8: ZiPulse ne joint pas Z2M

ValentinAUCLERC, 2023-04-19:
Hello

Je viens de recevoir mon ZiPulse, j'ai essayé de l'ajouter sur Z2m, en version 1.30.3, mais je n'arrive pas à le faire join.

=> Il est alimenté par pile
=> Quand je clic sur le bouton, j'ai bien la led qui clignotte
=> Quand je laisse cliqué pour réinitialiser, j'ai également un clignotement 
=> Absolument aucun log côté z2m qui est bien en acceptation de nouveau perif
=> Je continue à avoir des remontés de mes autres devices

Une idée ?

fairecasoimeme, 2023-04-20:
Bonjour,
Si le ZiPulses clignote, c'est qu'il n'est pas associé. Il faut bien vérifier que le coordinateur est bien en mode appairage puis re appuyer sur le bouton link du ZiPulses à proximité du coordinateur pour commencer.

Gonioul, 2023-04-20:
J'ai toujours eu des soucis d'appareillage avec Z2M et HA.
Souvent, redémarrer Z2M dans son menu outils ou redémarrer le module complémentaire Z2M peut aider avant de retenter un nouvel appareillage.

ValentinAUCLERC, 2023-04-24:
> J'ai toujours eu des soucis d'appareillage avec Z2M et HA. Souvent, redémarrer Z2M dans son menu outils ou redémarrer le module complémentaire Z2M peut aider avant de retenter un nouvel appareillage, surtout si le précédent n'a pas marché (désassocier de force avant). Se mettre aussi à proximité du dongle mais surtout à distance des émetteurs de même fréquence (wifi, bluetooth).

Effectivement le problème venait bien de z2m... désolé pour l'issue

closed 2023-04-24 by ValentinAUCLERC