fairecasoimeme/ZiPulses #9: ZiPulses (BAT) - GAZPAR - Pas de valeur energy

MrRometo, 2023-04-25:
Bonjour,

Je ne comprends pas je suis sur HomeAssistant / Zigbee2MQTT, je viens de recevoir et d'installer le Zipulses mais je n'ai pas toutes les informations qui remontent.

Et la principale (celle qui m'intéresse) je ne l'ai pas --> ENERGY

fairecasoimeme, 2023-04-26:
Bonjour,

Le coefficient multiplicateur ne doit pas être à 0 car impulsions x 0 donnera 0

fairecasoimeme, 2023-04-27:
Après avoir paramétré le multiplicateur, il faut bien faire la manipulation suivante : 
https://github.com/fairecasoimeme/ZiPulses#z2m en vérifiant bien que les deux encarts vert apparaissent.

Il faut aussi vérifier que le ZiPulses s'allume quand vous relier les deux bornes du "pulses" (pour simuler une impulsion). Si ce n'est pas cas, il y a un problème matériel sur le ZiPulses.

MrRometo, 2023-04-27:
Je ne comprends pourquoi je récupère les informations du capteur (voltage, temperature, linkquality) mais pas celle du compteur à impulsion (energy).

Alors que l'impulsion à l'air d'arriver.

MrRometo, 2023-04-28:
Bon après 5 ou 6 désappairage réappairage en suivant la doc à chaque fois ça fonctionne ! Je ne touche plus à rien!

[image]

Merci pour votre aide.

closed 2023-04-28 by MrRometo