fairecasoimeme/ZiPulses #11: Pile vide au bout d'un mois

jeanaaaa, 2023-05-29:
J'utilise ZiPluses sur un compteur Gazpar avec une pile cr2450 et la durée de vie de la pile n'excède pas le mois.

Est-ce que d'autres personnes ont le même souci ?

PEGARifrouf, 2023-05-30:
Bonjour,
J'ai effectivement déjà changé la pile une fois. Peut être qu'elle tiendra plus longtemps en été avec moins d'impulsions.
Je pense quand même le passer sur une alimentation secteur.

fairecasoimeme, 2023-06-01:
Bonjour,

Pouvez-vous déterminer combien d'impulsions a été enregistré sur la vie d'une pile ? (consommation / coefficient)
J'ai eu le cas d'un problème matériel qui surconsommait

jeanaaaa, 2023-06-01:
> Bonjour,
> 
> Pouvez-vous déterminer combien d'impulsions a été enregistré sur la vie d'une pile ? (consommation / coefficient) J'ai eu le cas d'un problème matériel qui surconsommait

3900 évènements pour un mois

PEGARifrouf, 2023-06-04:
Bonjour !
J'ai dû la changer vers 4200 impulsions, le Zipulse est assez proche d'un routeur et dans un cave donc tempérée 15-16 pendant la période de la pile.

fairecasoimeme, 2023-06-05:
> > Bonjour,
> > Pouvez-vous déterminer combien d'impulsions a été enregistré sur la vie d'une pile ? (consommation / coefficient) J'ai eu le cas d'un problème matériel qui surconsommait
> 
> 3900 évènements pour un mois

Bonjour,
en effet, ça me parait bizarre, je vous contacte par mail pour que je puisse récupérer le matériel et faire des tests

benooye, 2023-06-20:
Bonjour,
Je constate le même problème : pile HS rapidement
Je vais la changer et voir si le cela se reproduit

thunbergi, 2023-06-27:
Bonjour à tous,

Alors je ne sais pas si je suis tombé sur une mauvaise pile mais à 2v apres 3 jours de tentative d'appareillage...

Je la change et je vais voir... La nouvelle est bien à 3.24V

ça m'embête tout ça :)

thunbergi, 2023-06-27:
Et bien ça alors ! je viens de revenir, je teste la pile, 2.7v.... ça ne risque pas de marcher...

fairecasoimeme, 2023-06-28:
Attention, une pile lithium de type 2450 a une tension (sorti du carton) de 3.2V mais dès qu'on tire un peu d'énergie chute. Ensuite, dans le cas du ZiPulses, il y a une diode (pour des raisons de sécurité) qui engendre une baisse de tension. A l'usage, la tension "normale" d'une pile neuve est de 2.7 ou 2.8V.
Cependant, il est aussi possible que vous ayant un problème matériel si vous avez une des premières versions du ZiPulses.
Tenez moi au courant de l'évolution.

thunbergi, 2023-06-28:
En fait la tension je l'ai toujours prise en la retirant. Je n'ai par ailleurs jamais reussi à faire fonctionner le Zipulses, meme apres l'avoir mise à jour.  Là je viens de retirer la pile, 1.9V, sans rien faire ! :(

fairecasoimeme, 2023-06-28:
Ok dans ce cas là contactez-moi par mail pour faire un échange. contact[at]lixee.fr

PEGARifrouf, 2023-08-10:
Bonjour,
J'en suis à 2200 impulsions depuis le dernier changement. A terme j'installerai un alim secteur si ça tient dans les 4000 impulsions comme la première fois.
Je viendrai poster quand j'aurai de nouveau besoin de la changer.

Sinon ca marche vraiment très bien 👍

PEGARifrouf, 2023-08-11:
Si c'est l'envoi des données en zigbee qui consomme, il faudrait pouvoir choisir tout les combien d'impulsion sont envoyé les données voir même combiner avec un temps minimum entre chaque rafraîchissement.

Pour le gazpaz quand on utilise de l'eau chaude les plus petits incréments sont de 0.11kwh.

Voici a quoi ressemble une consommation d'eau chaude.

[image]

fairecasoimeme, 2023-08-21:
Bonjour,

Ce n'est pas forcément l'émission qui consomme le plus. C'est plus les réveils (dus aux impulsions) qui utilisent le plus d'énergie mais j'ai prévu de faire des recherches dans ce sens.

Fred

raspbeguy, 2023-08-21:
J'ai connecté mes deux zipulses à une pile 9V chacun. Ça a duré moins de 15 jours.

fairecasoimeme, 2023-08-21:
C'est normal.
si vous avez avez branché la pile 9V sur les plots vin/gnd, cela passe par un régulateur qui consomme bcp d'énergie. Le switch sur REG ne permet pas de conso limité.
Seule le switch sur bat et l'utilisation du support de pile permet une consommation limitée.

raspbeguy, 2023-08-29:
Merci de ces infos.

> PS 2 :De mon côté je vais travailler sur un nouveau régulateur (à découpage) pour avoir une conso moindre et utiliser les bornes VIN et GND

S'agit-il d'un changement matériel ou firmware?

fairecasoimeme, 2023-08-29:
> S'agit-il d'un changement matériel ou firmware?

Non forcément un changement matériel. Il faut remplacer le régulateur linéaire par un autre régulateur à découpage avec une consommation plus appropriée.

raspbeguy, 2023-08-29:
Sera-t-il possible d'effectuer la modif facilement ?

raspbeguy, 2023-08-29:
Merci de ces précisions.

Gilles2senior, 2023-08-31:
Par ailleurs, bien qu'il soit indiqué que le ZiPulses est supporté nativement par ZHA, la donnée "battery" ne remonte pas nativement sur ZHA et indique "NA". Il faut aller contrôler l'attribut de cette entité pour connaître la tension de la batterie.
Ce serait bien d'avoir le retour "battery" standard dans ZHA.

biwizit, 2023-09-24:
Bonjour, j'ai vidé ma pile CR2450 en 3 mois avec un ZiPulses qui n'a pas eu à envoyer des tonnes d'impulsions Gazpar (je n'ai pas le total car j'ai essayé tout un tas de choses pour améliorer la portée et j'ai souvent réinitialisé malgré moi le compteur interne). 

Dans les journées les plus froides d'hiver, mon Gazpar va envoyer 1500 impulsions par jour (soit environ une par minute). Du coup, je suis en train de passer sur deux LR20 Nimh rechargeables 10 000mAh en série pour augmenter la durée de vie (j'espère que ça le fera pour la tension...). Mais si une CR2450 de 500mAh ne supporte réellement que 4000 impulsions, je perds mon temps car je ne passerai clairement pas l'hiver... 

Je n'ai pas vérifié dans mon ZiPulses mais si j'en crois le Github, la puce utilisée est une NXP JN5189 et d'après la spec, la partie radio semble quand même consommer bien plus qu'un simple réveil en mode low-power pour aller stocker les impulsions en mémoire, non ? N'ai-je vraiment pas grand chose à gagner à demander l'envoi des infos du compteur interne que toutes les 30 impulsions ou toutes les 30 minutes (min interval) ?

Pour la partie régulation, c'est sûr que si on pouvait passer sur un plus grand éventail de batteries (comme du Li-Ion) sans les vider juste parce qu'elles dépassent 3V, ça aiderait aussi (mais moins qu'une gestion efficace de la conso quand le module ne fait que stocker les impulsions je pense).

Merci d'avance pour vos réponse et votre travail.

fairecasoimeme, 2023-09-25:
Bonjour,
Pour la consommation, il faut prendre en compte l'énergie consommé ET le temps dans lequel l'énergie est consommée. En effet, la transmission zigbee peut consommer jusqu'à 20mA mais le délai de transmission est de quelques millisecondes. 
Lorsque le zipulses se reveille, il consomme 8mA pendant 1-2 secondes. et il se réveille toutes les minutes.
la procédure de réinitialisation consomme aussi car le ZiPulses reste éveillé (8mA) + transmissions (20mA) pendant 15 secondes au moins permettant le bind, le reporting et les différents échanges radios liés à l'appairage.

Je pense qu'avec des LR20, vous passerez largement l'hiver puisque dans vos conditions (avec bcp de reinit qui consomme bcp) la capacité est 20 fois supérieures (soit 3 mois x 20 = 60 mois). même si on divise par 2 liés à plus de consommations et des piles rechargeables pas rechargé au max, ça passera vraisemblablement.

Enfin, pour la régulation, je travaille sur le changement du régulateur pour que la consommation soit moins importe que l'actuelle. Mais les piles lithium ne changeront pas non plus la vie. Ca reste une technologie avec des avantages et des inconvénients.

biwizit, 2023-09-25:
J'aurais imaginé un réveil depuis un (deep) sleep mode grâce à une des I/O du chip connectée à l'entrée du ZiPulses, plutôt qu'un réveil systématique toutes les minutes. J'ai l'impression que la puce le permet. Ceci explique peut-être cela...
  
Mes 3 mois se sont faits avec uniquement de l'eau chaude sanitaire, soit environ 50 impulsions par jour (on retrouve les 3x30x50=4500 impulsions pour une CR2450). Quand le chauffage va se mettre en route, on passera à 1 500 impulsions par jour ! Du coup, le rapport 20 des LR20 ne suffira pas à compenser ce rapport 30 de l'hiver (et pas 2), l'autonomie sera même plus courte... Mais on verra ce que ça donne très bientôt (à partir de mi-Octobre).

Merci.

fairecasoimeme, 2023-09-25:
Bonjour,

Il y a plusieurs choses qui réveille du deep sleep: 
* Une impulsion (I/O). On ne peut pas comptabiliser une impulsion sans réveiller le micro. (Du coup, ne faire des réveils que toutes les minutes ne suffit pas.)
* Un appui sur le bouton (I/O)
* L'horloge (RTC interne)

Combien de fois avez-vous fait l'appairage sur la pile ? Dans votre calcul vous ne prenez pas en compte cette partie non négligeable. Un appairage utilisera 0,2mAh environ alors qu'une émission utilisera 0.0003 mAh soit environ 650 fois moins énergivore. (un réveil utilisera 0.07mAh)

Pour le moment, il n'y a pas de reporting modifiable justement pour éviter selon la config de ne pas maitriser les piles. Le reporting se fait toutes les 2 heures pour les impulsions, la température et la tension de pile et à chaque impulsion.

Le travail actuel est surtout sur la consommation du réveil (en limitant le réveil de certaines interfaces) et à la limite de prolonger le deepsleep (il faut que je vérifie la possibilité dans le code car je ne suis pas sur mais il me semble qu'on ne peut pas rester en deepsleep perpétuel). Je pense plutôt que c'est sur ces points qu'il faut que je travaille plutôt que configurer le nombre d'émissions. 
Après il est vrai que pour des cas avec 1500 impulsions par jour, on ne peut pas faire de miracles avec 500mAh en suivant ce système mais comme le ZiPulses a énormément de cas d'usage. C'est pour cette raison que je regarde pour une alimentation avec un régulateur plus performant.

biwizit, 2023-09-26:
Bonjour, oui, j'ai fait quelques appairages dans mes essais d'améliorations de portée. Espérons que je puisse en effet gagner de ce point de vue-là. 

J'ai deux questions : 
1. Pourquoi y-a-t-il un réveil toutes les minutes ? Il pourrait ne se faire qu'à la réception d'impulsions, non ? 
2. Quand vous dites que pour le moment, il n'y a pas de reporting modifiable, vous voulez dire de façon "directe/facile" comme on peut le faire pour la partie unité/multiplieur/diviseur? Mais il est bien possible de le faire d'un point de vue "indirect/zigbee" en jouant avec la partie reporting de zigbee2mqtt ou zigbee4domoticz ? Je n'ai pas encore essayé de modifier les valeurs mais on voit bien des champs "min interval", "max interval' et "change" modifiables pour seMetering/currentSummDelivered dans zigbee2mqtt par exemple.

Merci pour vos réponses.

PEGARifrouf, 2023-09-27:
Bonjour,

Pour info sur une pile (la marque n'était peut être pas la même).
De mémoire j'ai fait 4500 impulsions sur une période d'hiver et 2500 sur une période avec peut être un poil de chauffage puis de l'eau chaude sanitaire seulement.
Comme dit avant, en hiver la pile tiendra plus qu'en extrapolant les chiffres d'été.

ellmout, 2023-09-27:
Bonjour,

Avec une pile CR2450 VARTA, depuis le 23/03/2023, j'en suis à 32 094 impulsions.

fairecasoimeme, 2023-09-27:
>     1. Pourquoi y-a-t-il un réveil toutes les minutes ? Il pourrait ne se faire qu'à la réception d'impulsions, non ?

Il y a un réveil toutes les minutes car la RTC me l'impose (il faut que je vérifie)
 
> 2. Quand vous dites que pour le moment, il n'y a pas de reporting modifiable, vous voulez dire de façon "directe/facile" comme on peut le faire pour la partie unité/multiplieur/diviseur? Mais il est [...]

Quand je dis qu'il n'y a pas de reporting modifiable, c'est qu'on ne peut pas jouer avec le config report pour changer les délais d'émissions. (Comme pour la plupart des objets zigbee sur pile) le fait de rendre disponible et modifiable les délais sur le reporting entrainerait une non maitrise (plus importante qu'actuellement) de la consommation des piles.

biwizit, 2023-09-27:
Merci pour vos réponses. C'est dommage pour cette partie config report non modifiable car je dirais que ceux qui vont jouer avec cet aspect-là dans les plugins savent en général ce qu'ils font. Et surtout, selon moi, vous vous êtes mis dans le cas le plus défavorable d'un point de vue autonomie avec envoi à chaque impulsion (avec peut-être un min interval d'une minute ? je ne suis plus sûr).
 
C'est grâce à cette approche qu'il est possible de faire disparaitre les quelques reproches qu'on peut faire à un device sur pile comme la sonde de température déportée Owon THS-317-ET. En mettant "max interval" et "change" à des valeurs plus adaptée (mais raisonnables), cette sonde devient bien plus réactive et intéressante, sans vraiment sacrifier l'autonomie (avec LR03 Eneloop, c'est bonheur en fait).

Et dans le cas du ZiPulses, avec un "min interval" à 30 min et un "change" à 10, on pourrait imaginer gagner en autonomie dans l'utilisation d'un Gazpar (il reste cette histoire de réveil toutes les minutes qui me semble contre-intuitif...).

Est-ce que ce sont des choses qui pourrait évoluer dans le futur ? Avec par exemple un firmware custom partagé à des utilisateurs assumant le risque d'un effet négatif quelconque ? 

Pour finir, je suis surpris (mais content) qu'on ait un device à 32 094 impulsions (série en cours) avec une pile de 500mAh. Ca sera clairement une bonne nouvelle pour certains (j'imagine que le device doit être très proche d'un coordinateur ou routeur, la conso semble aussi liée aux dbm dans la spec...).

Encore merci.

ellmout, 2023-09-27:
Le compteur est à 15 m du coordinateur dans un boitier sur la rue, après j'ai une vingtaine de routeur dans la maison, plus au moins près, donc je ne sais pas avec qui y parle exactement !
Je suis moi même très étonné de la longévité de la pile !
Après on a seulement commencé au 23/03, d'après le reporting, le plus froid que j'ai eu dans le boitier, c'est 7°C.

biwizit, 2023-09-27:
Ca me donne beaucoup d'espoir car je suis dans le même cas (moins de 15m en fait) et j'ai souvent eu un ZiPulses qui arrêtait de communiquer avec le coordinateur (Sonoff ZBDongle-E). C'est en cherchant tout un tas de moyen pour améliorer la situation que j'ai multiplié les appairages (forcer le passage par une ampoule Tradfri qui éclaire le devant de la maison, etc...). Je me dis naïvement que la multiplication des impulsions (passage de 50 par jour à bientôt 1500) évitera peut-être cette "perte de réseau" pendant les périodes où il n'y avait un échange que toutes les 2h (correspondant à la remontée d'info tension/température). Etant donné l'autonomie de ma première CR2450, j'ai commencé à creuser la partie LR20 pendant que la 2ème se vide (à 2.6V au bout d'une semaine, avec un seul apparairage et toujours 50 impulsions par jour, je crois très peu à une changement radical d'ailleurs...). Merci pour les infos.

ellmout, 2023-09-27:
Tu utilise Zigbee2Mqtt ?
Si c'est le cas, la clé Sonoff ZBDongle-E est affiche comme "expérimentale" dans la page des adapteurs supportés : 
https://www.zigbee2mqtt.io/guide/adapters/
Moi j'avais une conbee II avant, et j'avais régulièrement des problèmes.
Je suis passé sur une Sonoff ZBDongle-P (basé sur TI CC2652), et depuis plus aucun problème...

biwizit, 2023-09-27:
J'utilise depuis toujours conbee2/deconz sur Raspberry Pi4 au coeur de mon salon. Mais sans routeurs supplémentaires (beaucoup de thermomètres et de capteurs d'ouverture de porte, détecteur de présence, de fuite, etc... uniquement sur piles). J'avais mis un thermomètre aqara puis un thermometre Sonoff dans le boitier Gazpar côté rue pour vérifier que ça captait et tout marchait nickel sur plusieurs jours. Du coup, j'ai acheté le ZiPulses et là, douche froide : il ne tenait pas plus d'une journée dans le réseau (grosse déception, on parle d'un device 10x plus cher...).  

Donc, je me suis dit que j'allais utiliser un Raspberry Pi3 que j'avais en stock, positionné beaucoup plus près de la fenêtre, avec un coordinateur dédié et utiliser Zigbee2mqtt pour envoyer l'info à Domoticz qui est sur mon raspberry Pi 4. J'ai effectivement pris le dongle le plus récent de Sonoff qui se trouve être en expérimental sur Zigbee2mqtt... Tous les objets nouveaux que j'ai eu depuis 3 mois ont été mis sur ce combo (pour me faire une idée) et je n'ai eu absolument aucun problème. Par contre, le ZiPulse a malheureusement continué à passer hors ligne régulièrement. Mon dernier essai a été de rajouter un routeur (ampoule Tradfri) à l'extérieur, en me disant que ça pourrait améliorer la situation. Au bout de quelques jours, le ZiPulses a décidé de repasser sur le coordinateur directement et je vais bien voir si ça finit comme avant d'ici quelques jours...

Je peux évidemment tenter d'acheter le Sonoff ZBDongle-P mais quand tout le reste marche parfaitement bien, je me dis quand même... Bref, voilà où j'en suis... Merci pour le retour en tout cas.

benooye, 2023-09-27:
Bonjour, pour info de mon côté j'utilise également un SonOff Zigbee Dongle-E et j'avais pas mal de déconnexion du ZiPulse, en particulier au redémarrage de Home Assistant.
J'ai mis à jour le firmware du dongle via https://github.com/darkxst/silabs-firmware-builder (EmberZNet NCP)
Réinitialisé le Zigbee puis refait un appairage en mettant bien le ZiPulse à sa place finale près du GazPar.
Depuis plus de problèmes...

biwizit, 2023-09-27:
Intéressant, merci. Du coup, si le ZiPulses finit par disparaître, j'en profiterai certainement par mettre le firmware du Zdongle-E à jour du coup. Surtout que ça a l'air d'être très facile maintenant (directement dans Chrome).

diliak, 2023-09-29:
Hello,
Pile vide au bout d'un mois ? Moi c'est une semaine environs.
Deuxième pile HS ce jour.
Installation le 17/09 et on est le 29... Grosso modo, une pile par semaine.
Installation avec un compteur d'eau acheté sur Domadoo (y compris le Zipulses.

Le système en lui-même est génial si c'est alimenté en 12V (ce que j'ai fait) mais avec des piles....

diliak, 2023-10-02:
Bonjour,

Ce matin, zipulses en demande d'appairage (Bleu clignotant). Plus rien ne remonte.
Je refais des essais de RAZ divers ce soir. Il y a un bouton reset sur la carte du zipulses. Je vais tester voir si ça améliore les choses pour ensuite appairer à nouveau. Curieux, je n'avais pas tous ces soucis avec les piles même si ça en bouffe. Peut-être réduire la tension d'alim en mettant un régulateur...

fairecasoimeme, 2023-10-02:
Bonjour,

@diliak et @biwizit 
Si les piles se vident en 1 semaine, c'est qu'il y a un problème avec l'appareil. Je veux bien en faire l'analyse et vous le remplacer par un nouveau. pour cela, il suffit de me contacter par mail contact[at]lixee.Fr

diliak, 2023-10-02:
Bonjour,

C'est noté pour moi et c'est très gentil de votre part.
Je vais refaire des tests ce soir et essayer d'épurer tout ça et repartir au propre.
A la limite, si je peux faire fonctionner celui-ci avec ma source 12v, pas de pb pour moi.
Par contre, dans mes essais hier, quand le capteur ILS reste fermé (compteur d'eau), la LED bleue reste allumée. Je ne me souviens pas si c'était le cas avec la pile mais cela pourrait expliquer la conso excessive. Il faut aussi que je résolve les pertes de connexion (pourtant, à 5m du coordinateur à travers une cloison bois & vérifs bonne connexion antenne correcte sur le zipulses).

diliak, 2023-10-02:
Tests effectués. Je prends contact en mail. J'ai peut-être identifié un souci.

fairecasoimeme, 2023-10-18:
Bonjour,
avec la version 6.0, il y a optimisation de la consommation, notamment si le compteur eau s'arrête pile sur l'aimant et que l'impulsion reste en contact.
il y avait aussi une légère sur consommation après l'appuie du bouton qui était corrigé. Il faut quand même faire attention de ne pas mal remettre le boitier qui engendre l'appuie permanent du bouton.
Enfin, j'ai fini le prototype avec le nouveau régulateur et lancé la production du nouveau ZiPulses permettant d'utiliser les bornes VIN / GND avec toutes sources d'alimentations entre 3.3V et 24V avec une surconsommation de seulement 4uA.
Il sera disponible mi-novembre.

diliak, 2023-10-18:
Bonjour,
Ah merci beaucoup ! Je vois que mes remontées ont été utilies ;) .
Pour le bouton bloqué, j'ai eu le coup, je confirme. C'est un peu ce qui m'a décidé à ne pas le renvoyer inutilement.
Pour moi, je pense laisser comme ça et faire la mise à jour uniquement si je fais un changement majeur. 
En tous cas, super idée que ce zipulses.

Je commanderai certainement un V6 pour mon Gazpar.

diliak, 2023-10-21:
> J'utilise depuis toujours conbee2/deconz sur Raspberry Pi4 au coeur de mon salon. Mais sans routeurs supplémentaires (beaucoup de thermomètres et de capteurs d'ouverture de porte, détecteur de [...]

Marrant, je rejoins le même club mais avec une clé Conbee2 ... Notons que je n'ai aucune erreur dans le plugin Zigbee for Domoticz.
Disparitions régulières du Zipulses donc plus aucune remontée et obligé de reset...

diliak, 2023-10-21:
https://github.com/fairecasoimeme a dit :

> Bonjour, avec la version 6.0, il y a optimisation de la consommation, notamment si le compteur eau s'arrête pile sur l'aimant et que l'impulsion reste en contact. il y avait aussi une légère sur [...]

Bonjour,
J'ai mis à jour le Zipulses. Je le mets en tests sur table.
Effectivement, le voyant s'éteint maintenant même si l'ILS est fermé.
Je fais des impulsions à la main, tout fonctionne bien. Des fois, une impulsion ne passe pas mais est retransmise avec une nouvelle impulsion. C'est donc très bien, cela veut dire que le zipulses compte bien et mémorise bien les impulsions.

Dans Domoticz sur pile :
Tension 3,3V
Température 21°
Batterie level à 200% (???)

J'attends 2h, pour voir. Ensuite, c'est peut-être une modif dans le plugin de Domoticz à faire.

fairecasoimeme, 2023-10-24:
Bonjour,
les rebonds sont gérés et même s'ils ne l'étaient pas, ça ne fait pas "planter", la seule chose, coté ZiPulses, qui peut le faire sortir du réseau c'est le bouton "link" resté appuyé. 
J'ai déjà effectué énormément de tests sur les rebonds et simuler énormément d'impulsions et à aucun moment le ZiPulses plante ou sort du réseau. Quand vous pensez qu'il plante, la led ne s'allume plus lors d'appuie sur bouton ou impulsion ?

Quand vous le faites "planter" et que vous mettez votre coordinateur en appairage. Le ZiPulses rerentre-t-il ? si c'est le cas le ZiPulses a été remis à zéro sinon c'est le coordinateur qui l'a éjecté sans l'avertir.

Enfin, vos interventions ne semble plus correspondre avec l'issue. Veuillez recréer une issue correspondant à vos problèmes.

diliak, 2023-10-24:
Merci pour ces infos.
Quand je dis "planter" c'est que les remontées ne fonctionnent plus (plus de LED bleue allumée lors d'un pulse).
Et effectivement, on sort du sujet. Je ferai un autre topic plus tard.

Gilles2senior, 2023-10-29:
Une question sur la pile bouton du Zipulses:  je suis la valeur de la tension de la pile sur HA. Mais quelle est la tension mini qui doit m'indiquer qu'il faut penser à changer cette pile? Ceci me permettrait de mettre une alerte sur Home Assistant.
A ce jour, la tension de ma pile est à 2.6V, mais quelle est la valeur à ne pas dépasser à la baisse? 
Merci

biwizit, 2023-10-29:
La limite basse est 2.4V. Il est possible de trouver l'info dans les spécifications de certaines CR2450. Et je l'ai vérifié empiriquement.

Gilles2senior, 2023-10-29:
> La limite basse est 2.4V. Il est possible de trouver l'info dans les spécifications de certaines CR2450. Et je l'ai vérifié empiriquement.

Merci beaucoup pour cette info

ellmout, 2023-10-29:
Ma pile est à 1.8V et mon Zipulse communique toujours !

biwizit, 2023-10-29:
Euh... Même le microcontrôleur Zigbee ne peut pas être alimenté en dessous de 1.9V (dans ses spécifications également). Bizarre donc.

ellmout, 2023-10-29:
[image]

Je vais devoir aller changer la pile alors !

biwizit, 2023-10-29:
Je n'ai pas dit que je ne croyais pas ce que vous lisiez. Mais la mesure est sûrement bidon, en effet.

biwizit, 2023-10-29:
Par exemple, la très très classique Energizer que j'utilise a coupé à 2.4V. Et il est strictement impossible d'être en dessous de 2V. 

Voir sa spécification : [https://data.energizer.com/pdfs/cr2450.pdf](https://data.energizer.com/pdfs/cr2450.pdf)

biwizit, 2023-10-29:
Donc il va falloir aller la changer, oui ! :-)

diliak, 2023-10-29:
Tu es passé sur le dernier firmware ?

ellmout, 2023-10-29:
Qui moi ? Non je ne touche à rien, il marche trop bien.

fairecasoimeme, 2023-10-30:
Bonjour,

La pile bouton est protégée par un MOSFET pour la polarité mais aussi une diode qui engendre une baisse de tension de 0.5V environ. (ça ne joue pas sur la conso). Dans la prochaine version du produit, il n'y a plus la diode qui engendre la baisse de tension. En dessous d'une certaine tension aux alentours de 2V, la led ne s'allume plus mais le ZiPulses continue d'émettre les trame Zigbee.

ellmout, 2023-12-03:
Tjrs pas changé la pile :
[image]

[6 more replies]