fairecasoimeme/ZiPulses #12: Zipulses configuration Z2M

nlindblo, 2023-06-13:
Hello,

I have installed Zipulses in my Gazer meter and configured zigbee2mqtt in a way that I believe is correct. I am receiving pulses but the values are in whole numbers and the energy sensor does not have a unitOfMeasure, meaning I dont seem to be able to add it to the Energy dashboard. Here is my configuration and output if someone have an idea of where I am going wrong, my objective would be to have the numbers to come in as m3 into HA, is there any further configuration I need to do ?

Apologies for writing in English as I'm not French speaking, feel free to respond in French and I will Google translate

[image]

[image]

[image]

[image]

fairecasoimeme, 2023-06-13:
hi, you have to reconfig the device as explained here : 
https://github.com/fairecasoimeme/ZiPulses#z2m

nlindblo, 2023-06-13:
I actually had followed the instruction for the reconfiguration, but to confirm I re-paired the device and did it another time and now the values seem to come through in the right decimal format. However, the energy sensor does still not have a UnitOfMeasurement or device_class, is this supposed to be configured manually or come from Z2M ? These are the sensors that have been created

[image]

raspbeguy, 2023-07-26:
Can confirm the behaviour. If you're using Home Assistant, workaround is to create another device in HA config file:

```yaml
template:
  - sensor:
      - name: "Gas_meter"
        unit_of_measurement: "m³"
        device_class: "gas"
        state_class: "total_increasing"
        state: "{{ states('sensor.compteur_gaz_energy')|float }}"
        availability: "{{ states('sensor.compteur_gaz_energy') not in ['none', 'unknown', 'unavailable'] }}"
```

beauby, 2025-01-23:
I am facing the same issue: setting `unitOfMeasure` to `m3` does not cause the `energy` entity to have `unit_of_measurement` set to `m3`. Am I misunderstanding the role of `unitOfMeasure`?

nlindblo, 2025-01-24:
Its been some time since I looked at this, but I checked just now and I have the unit set to m3 in Z2M and then I have this template sensor, and it seem to generate an accurate result

````
    - sensor:
        - name: "Engie Gas kWh"
          availability: "{{ ((states('sensor.zipulses_device_energy') | is_number )) }}"
          icon: mdi:gas-burner
          unit_of_measurement: "kWh"
          device_class: energy
          state_class: total_increasing
          state: >
            {% set gasm3 = states('sensor.zipulses_device_energy') | float %}
            {% set coeff = states('input_number.grdf_gas_coefficient') | float %}
            {{ (gasm3  * coeff) | round(2, default=0) }}
````

beauby, 2025-01-24:
Thanks for the workaround @nlindblo, I ended up with something similar on my end and can confirm it works.
I am still confused about the role of the `unitOfMeasure` property since it does not seem to be used by the ZiPulses internally, nor for the ˋenergy` field.