fairecasoimeme/ZiPulses #13: ZiPulses pas reconnu dans Z2M

mrgnsrl, 2023-06-16:
Je n'arrive pas à associer mon ZiPulses dans Z2M, il est reconnu comme non supporté :
[image]

Z2M est en version 1.31.2-dev :
[image]

Gonioul, 2023-06-16:
Entretien terminé = Faux.
-> Il faut recommencer l'appairage.

Je conseille de supprimer l'appareil, redémarrer l'addon zigbee2mqtt et de réessayer.

mrgnsrl, 2023-06-17:
Merci, j'ai du sélectionner l'équipement le plus proche pour l'appairage sinon l'entretien échouait systématiquement.

Tout fonctionne parfaitement !

closed 2023-06-17 by mrgnsrl