fairecasoimeme/ZiPulses #15: Configuration Home Assistant échoué

Lens40, 2023-07-12:
J'ai pu appairer le zipulse avec la clé USB sonoff qui est aussi connecté avec Lixee (qui marche parfaitement) mais sur la configuration cela échoue. J'ai un binding et le reporting affiché en vert mais pour le reste j'ai que des croix orange. 

Donc je peux voir le zipulse dans Home Assistant mais je n'ai aucune entité.

fairecasoimeme, 2023-07-13:
Si la demande de configuration est effectuée en dehors du processus d'appairage, il faut actionner le bouton du ZiPulses pour le réveiller. Si vous n'avez aucune entité, je pense qu'il faut refaire la procédure d'appairage.

Lens40, 2023-07-13:
Merci, le zipulse clignote en continue mais après plusieurs minutes (5min et plus) il n‘est toujours pas configuré après un nouveau appairage. Aussi le lancement d’une nouvelle configuration a échoué.

fairecasoimeme, 2023-07-13:
Si le zipulses clignote, c'est qu'il n'arrive pas à s'appairer. Du coup, il y a un problème de portée. Soit distance, soit perturbation.
Pour vérifier qu'il fonctionne correctement, il faut commencer par le rapprocher du coordinateur. Si l'appairage se fait correctement, il faut "analyser" le réseau pour régler le problème de distance ou perturbation

Lens40, 2023-07-13:
Merci. Les deux sont à 10 cm. Donc pas de problème à ce niveau la. L‘appairage fonctionne, il est bien reconnu mais il ne fait pas la configuration. Le RSSI est à -3 et le LQI a 255.

fairecasoimeme, 2023-07-13:
Vous utilisez ZHA ou Z2M avec quelle version ?
Pouvez-vous faire une capture d'écran ?

Lens40, 2023-07-13:
Merci pour votre temps. J‘utilise ZHA. Et j‘ai la dernière version de Home Assistant
[image]
[image]
[image]

Lens40, 2023-07-18:
Il y a du nouveau? Comment puis-je le faire fonctionner?

fairecasoimeme, 2023-07-19:
Désolé mais les captures d'écrans n'apportent rien. vous devez "slider" vers le bas pour voir les capteurs

Mika59940, 2023-10-05:
Bonjour, en appairant le zipulses j'arrive à configuer le multiplicator et le divisor par contre impossible à modifier l'unité de mesure ,  je n'ai pas la tension de la pile et température. Voici ce que j'ai trouvé dans le journal de Zigbee2MQTT:

`2023-10-05 08:05:43Failed to configure '0x00158d0005ce760f', attempt 1 (Error: Read 0x00158d0005ce760f/1 seMetering(["divisor","unitOfMeasure","multiplier"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":25784,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))`

fairecasoimeme, 2023-10-05:
Bonjour, la valeur modifiée doit être un entier en respectant le tableau des unité de mesures : https://github.com/fairecasoimeme/ZiPulses#tableau-dunit%C3%A9-de-mesure

fairecasoimeme, 2023-10-10:
Bonjour,
Il est normal de ne pas avoir de rapport.
Pour l'unité de mesure, avez-vous suivi le tutoriel ?
https://github.com/fairecasoimeme/ZiPulses#z2m

Mika59940, 2023-10-10:
Bonjour  @fairecasoimeme , j'ai bien suivi le tuto car 'Multiplier' et 'divisor' se modifient  sans problème . Seulement 'UnitOfMeasure' que je n'arrive pas à modifié .

Info sur les versions:

- ZiPulses: 4000 0004

- Zigbee2MQTT: 1.33.1-1

tomj44300, 2023-11-03:
Bonjour @fairecasoimeme je viens de recevoir mon zipulses ce midi avec la fw 4000 0006 et utilise Z2M 1.33.2 , j'ai exactement le même problème que @Mika59940 pour configurer en L avec mon compteur d'eau Diehl izar pulse I
As tu depuis trouver une solution ?
Et oui j'ai bien suivi le tuto et comme Mika je peut changer multiplier et divisor
Merci pour ton aide

fairecasoimeme, 2023-11-03:
https://github.com/fairecasoimeme/ZiPulses/issues/21#issuecomment-1788820015

rs, 2023-12-28:
Je rencontre un problème similaire à @Lens40. Lors de l'ajout dans home assistant via ZHA, la phase de configuration ne termine jamais et le device est ajouté de façon incomplète à home assistant. Seul la température est disponible et j'obtiens des erreurs de ce type dans les logs :

```
2023-12-28 15:08:28.457 INFO (MainThread) [zigpy.endpoint] [0x3676:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, dev
ice_type=263, device_version=1, input_clusters=[0, 1, 3, 1026, 1794], output_clusters=[4, 3, 1794])
2023-12-28 15:08:28.459 DEBUG (MainThread) [zigpy.zcl] [0x3676:1:0x0000] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOB
AL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=False, *is_genera
l=True), tsn=94, command_id=<GeneralCommand.Read_Attributes: 0>, *direction=<Direction.Server_to_Client: 0>)
2023-12-28 15:08:28.460 DEBUG (MainThread) [zigpy.zcl] [0x3676:1:0x0000] Sending request: Read_Attributes(attribute_ids=[4, 5])
2023-12-28 15:08:28.462 DEBUG (MainThread) [zigpy.device] [0x3676] Extending timeout for 0x5e request
2023-12-28 15:08:28.463 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(timestamp=datetime.datetime(2023, 12, 28, 14, 8, 28, 463113,
 tzinfo=datetime.timezone.utc), src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, add
ress=0x3676), dst_ep=1, source_route=None, extended_timeout=True, tsn=94, profile_id=260, cluster_id=0, data=Serialized[b'\x00^\x00\x04\x00\x05\x00'], tx_option
s=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-12-28 15:08:28.464 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setExtendedTimeout: (00:15:8d:00:05:d2:5e:4f, True)
2023-12-28 15:08:28.466 DEBUG (MainThread) [bellows.uart] Sending: b'570e21a92a2a5aec8b914aa8bf559342497e'
2023-12-28 15:08:28.472 DEBUG (MainThread) [bellows.uart] Data frame: b'760ea1a92a2ada0f7e'
2023-12-28 15:08:28.472 DEBUG (MainThread) [bellows.uart] Sending: b'8070787e'
2023-12-28 15:08:28.472 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setExtendedTimeout: []
2023-12-28 15:08:28.473 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x3676, EmberApsFra
me(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupI
d=0, sequence=94), 95, b'\x00^\x00\x04\x00\x05\x00')
2023-12-28 15:08:28.474 DEBUG (MainThread) [bellows.uart] Sending: b'600f21a9602a15c46f904b25aa5493099d4e27f5b2c967d5fdc2638cfcd03f7e'
2023-12-28 15:08:28.485 DEBUG (MainThread) [bellows.uart] Data frame: b'070fa1a9602a15c607867e'
2023-12-28 15:08:28.485 DEBUG (MainThread) [bellows.uart] Sending: b'8160597e'
2023-12-28 15:08:28.485 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendUnicast: [<EmberStatus.SUCCESS: 0>, 116]
2023-12-28 15:08:28.528 DEBUG (MainThread) [bellows.uart] Data frame: b'170fb1a96b2a15c46f904b25aa5493099d4e27dfb2ce677b6c7e'
2023-12-28 15:08:28.528 DEBUG (MainThread) [bellows.uart] Sending: b'82503a7e'
2023-12-28 15:08:28.530 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>
, 13942, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DIS
COVERY: 320>, groupId=0, sequence=116), 95, <EmberStatus.SUCCESS: 0>, b'']
2023-12-28 15:08:28.530 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 1
3942, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOV
ERY: 320>, groupId=0, sequence=116), 95, <EmberStatus.SUCCESS: 0>, b'']
2023-12-28 15:08:28.548 DEBUG (MainThread) [bellows.uart] Data frame: b'270fb1a9112a15b658944a24ab5593499c80ab149bf89874e7de3d88f87e3fe5ee81b737ea9ac2dbd59061d6
2f73dc80055ed6cb4a7e'
2023-12-28 15:08:28.549 DEBUG (MainThread) [bellows.uart] Sending: b'83401b7e'
2023-12-28 15:08:28.552 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAS
T: 0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, group
Id=0, sequence=206), 140, -65, 0x3676, 255, 255, b'\x18^\x01\x04\x00\x00B\x05LiXee\x05\x00\x00B\x08ZiPulses']
2023-12-28 15:08:28.553 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST:
0>, EmberApsFrame(profileId=260, clusterId=0, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=
0, sequence=206), 140, -65, 0x3676, 255, 255, b'\x18^\x01\x04\x00\x00B\x05LiXee\x05\x00\x00B\x08ZiPulses']
2023-12-28 15:08:28.554 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2023, 12, 28, 14, 8, 28, 554327, tzin
[... cut ...]
2023-12-28 15:08:28.617 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (00:15:8d:00:05:d2:5e:4f, 1, 0, 4, 'LiXee', datetime.dateti
me(2023, 12, 28, 14, 8, 28, 563343, tzinfo=datetime.timezone.utc)) params: FOREIGN KEY constraint failed
2023-12-28 15:08:28.621 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (00:15:8d:00:05:d2:5e:4f, 1, 0, 5, 'ZiPulses', datetime.dat
etime(2023, 12, 28, 14, 8, 28, 563446, tzinfo=datetime.timezone.utc)) params: FOREIGN KEY constraint failed
…
```

Il y a d'avantage de logs dispo si besoin.

fairecasoimeme, 2024-01-02:
Bonjour,
Il faut simplement appuyer régulièrement sur le bouton "Link" du ZiPulses lors de l'appairage pour le réveiller. Sinon refaire la procédure d'appairage.

rs, 2024-01-03:
Le réveiller permet effectivement d'arriver au bout de l'appairage, mais ensuite les valeurs reste "unavailable" malgré plusieurs pulses de test: 
[image]

fairecasoimeme, 2024-01-04:
Quel est le LQI de l'appareil ?

rs, 2024-01-04:
LQI: 120
RSSI: -70

fairecasoimeme, 2024-01-05:
Pour tester, pouvez-vous refaire l'appairage à côté du coordinateur zigbee ?

rs, 2024-01-08:
Cela fonctionne mieux effectivement. Je l'ai ensuite branché au gazpar, mais il ne semble enregistrer aucune pulsation. L'entity `sensor.lixee_zipulses_summation_delivered` reste invariablement à 0. Le sensor de température lui semble fonctionner.

fairecasoimeme, 2024-01-10:
Attention, il y a aussi un sens dans le branchement au Gazpar
https://github.com/fairecasoimeme/ZiPulses#compteur-gazpar