fairecasoimeme/ZiPulses #16: Comptage eau reste à Zero + Device type

kfran78, 2023-09-01:
Bonjour

Je suis sous ZHA et j’utilise un compteur  impulsion Gianola (1 impulsion = 1 litre) 
Appairage ok. 
Changement des unit en litre au lieu de W/h = Ok
Je vois bien l'appareille se réveiller à chaque impulsion (diode bleu s'allume et s’éteint)

1/ Mais le comptage reste à Zéro
[image]

2/ Le device type est  "electric metering" et il faudrait le changer pour être reconnu dans "Energie" en tant que consommation Eau. Comment faire ?
[image]

fairecasoimeme, 2023-09-04:
Bonjour,

Il y a deux possibilités :
* Soit il y a un problème de portée. Il faut vérifier si les autres valeurs remontes. Vérifier la valeur du LQI
* Soit le bind n'est pas passé lors de l'appairage. Il faut donc refaire la procédure d'appairage

fairecasoimeme, 2023-09-20:
Bonjour,
Malheureusement le zipulses peut compter de l'eau / du gaz ou de l'électricité donc difficile de mettre un type de device qui convienne à tout le monde.
Cependant, j'ai vu passer sur des forum de home assistant la possibilité de modifier le YALM du capteur pour changer de "class"
(Désolé je ne suis pas expert HA)

tomj44300, 2023-11-05:
Bonjour j'ai un peu galérer voici ce qui marche pour moi dans mon fichier de conf HA.
J'ai pu ajouter ce sensor en litre dans la carte energy et y mettre un prix et avoir les graphs qui vont bien
[image]
Chez moi c est le zinky qui fait routeur pour le zipulses qui lui est semi enterrer sous une plateau en fonte 
ps je suis sur Z2M

template:
  - sensor:
      - name: "Consommation eau"
        unit_of_measurement: "L"
        state: "{{ states('sensor.idzipulses_energy')}}"
        device_class: water
        state_class: total_increasing

utility_meter:
  eau_mois:
    source: sensor.consommation_eau
    cycle: monthly

  eau_semaine:
    source: sensor.consommation_eau
    cycle: weekly

  eau_quotidien:
    source: sensor.consommation_eau
    cycle: daily

  eau_heure:
    source: sensor.consommation_eau
    net_consumption: true
    cycle: hourly