fairecasoimeme/ZiPulses #19: Zipulses + EMETTEUR D'IMPULSION DIEHL IZAR MBUS COMPACT I --> pas d'incrément

Pigouchet, 2023-09-17:
Bonjour,

Je tente de configurer le Zipulses avec le module l'émetteur d'impulsions Diehl (2 fils) mais il semblerait que Zipulses n'arrive pas à capter les impulsions du compteur.
D'après mes différents tests :

- Le Zipulses s'incrémente bien lors d'un contact (lien entre les 2 bornes pulse)

[image]

- Le Zipulses est bien vu sur le réseau Zigbee lorsqu'il est dans la boite près du compteur  

[image]

Est-ce qu'il peut y avoir une incompatibilité sur la réception des impulsions (fqz...) ? un paramétrage spécifique au niveau Z2M ?

P.S. : 

Ma config : 

- HA : 2023.9.2
- Z2M : 1.33.0-1
- Zipulses FW : Version 0005 après flash abouti à 100% mais la remontée dans Z2M indique toujours 0004 (??)

[image]

fairecasoimeme, 2023-09-18:
Bonjour,
quel est le modèle de votre compteur impulsion ? est-il polarisé ?
Fred

Pigouchet, 2023-09-18:
Bonsoir,
Le modèle est le module [IZAR MBUS COMPACT i ](https://cdn.store-factory.com/www.compteur-energie.com/media/DIEHL-IZAR-MBUS-COMPACT-i-compteur-energie.pdf). Non il est d'une sortie 2 fils non polarisés.
Effectivement j'en étais arrivé à la même conclusion mais je me demandais s'il n'y avait la possibilité de mauvaise "interprétation" des impulsions par le Zipulses ou des réglages supplémentaires à tester (e.g. jouer sur la valeur du debounce).
On est d'accord sur le papier cette configuration est censé fonctionner ? Pas de restriction technique identifié ?
Merci pour votre retour dans tous les cas.

fairecasoimeme, 2023-09-19:
Ok, je n'avais pas fait attention au modèle dans le titre de l'issue mais le matériel que vous avez communique en MBUS. Du coup, votre émetteur n'est pas compatible avec les impulsions.
je pense que c'est ce produit qui sera compatible : 
https://www.compteur-energie.com/eau-emetteurs-impulsions-sappel-izar-pulse-i.htm

Pigouchet, 2023-09-19:
Ok merci de votre retour je vais le changer en espèrent que ça fonctionnera.
Merci.

closed 2023-09-20 by fairecasoimeme