fairecasoimeme/ZiPulses #21: Instructions upgrade firmware ZiPulses

Gonioul, 2023-10-18:
Bonjour,

Les instructions ne me semblent pas claires.

Déjà est-ce qu'on doit retirer toute alimentation avant (pile) ?

Le lien vers la maj Zigate v2 pointe vers des distributions rasberry obsolètes: Pas de Bullseye ou de Bookworm
Et pas de binaires x86 linux non plus.

Cdlt,

Mika59940, 2023-10-19:
bonjour @fairecasoimeme, pour la mise à jour du firmware ZiPulses, faut-il utilisé l'outil de flashage [https://github.com/fairecasoimeme/ZiGatev2/raw/master/Tools/JN-SW-4407-DK6-Flash-Programmer.zip](url).
Cordialement.

Mika59940, 2023-10-20:
Bonsoir @Gonioul, j'ai réussi avec Powershell de windows 10 

- Télécharger le firmware en (.bin)
- Installer le pilote de la clé USB CP2102
- Télécharger l’outil flash (JN-SW-4407-DK6-Flash-Programmer.zip)
- Connecter la clef USB.
- Lancer Powershell :

> Pour lister le port COM
`DK6Programmer -l`
> Puis utilisé le port pour la mise à jour:
`DK6Programmer -s COM4 -P 500000 -e FLASH -Y -p C:\NXP\DK6ProductionFlashProgrammer\Zipulses_v6.0.bin`

Gonioul, 2023-10-20:
Le premier lien ne marche pas sans copier-coller du texte.
Sinon pour le reste je préférerais me passer de windaube :p

Mika59940, 2023-10-20:
Voici le lien pour la clé USB CP2102 [https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads](url)

diliak, 2023-10-21:
Hello,
Je l'ai fait sans la pile (zipulses alimenté par l'adaptateur USB/TTL) et sans l'option FLASH. Ca a fonctionné sans soucis.
Pour LINUX, effectivement, je n'ai rien trouvé de probant.

pdecat, 2023-10-22:
Sous linux, il faut contacter @fairecasoimeme pour obtenir l'outil servant au flash qui ne peut être distribué publiquement.

Gonioul, 2023-10-22:
Malheureusement même sous windows rien à en tirer, le CP2102 clignotte en rouge et bleu mais finit par un read error...

diliak, 2023-10-23:
Question à deux balles (sait-on jamais...) N'avez-vous pas mis le CP2102 à l'envers ?

diliak, 2023-10-23:
Dernier truc (encore une fois, peut-être l'avez-vous fait...) comme dit dans l'explicatif :

> Une fois que les branchements sont OK, il suffit d'insérer sur votre ordinateur la clef USB **en maintenant le bouton Flash** puis relacher. L'utilisation d'une rallonge USB peut se révéler plus pratique pour faire la mise à jour.

Si ça peut aider...

diliak, 2023-10-23:
Ben, je sais pas mais pour moi, j'ai eu la même fenêtre sur powershell et je vous garantis que ça a fonctionné. Comment je le sais ?
Le voyant bleu s'allume et s'éteint si on fait des impulsions en faisant un pont avec un fil. Avec la V5, ça restait allumé si on restait en contact permanent. 
Faites le test.

Pigouchet, 2023-10-23:
@diliak, Merci pour votre retour. Effectivement pour moi aussi le contact permanent n'a plus l'air d'allmuer la diode en continu, ça serait juste un bug d'affichage dans Z2M ?
J'ai tenté de réappairer Zipulses dans Z2M mais pas mieux même moins bien car depuis je ne peux plus re positionner l'unité de mesure en "L" à la place de "kWh" :

[image]

diliak, 2023-10-23:
Ah ben là, je ne pourrai pas vous aider mais à priori le firmware a bien été installé.

Gonioul, 2023-10-24:
Les bornes semblent correspondre en ce qui me concerne.
Doit y avoir un soucis de driver.

Gonioul, 2023-10-24:
Question bête, Je ne vois que la led rouge fixe GND et la led bleue qui clignotte du TX, est-ce qu'il doit y avoir une led sur le RX? Si oui je ne la vois pas...

diliak, 2023-10-25:
J'ai le même modèle acheté sur Amazon et à priori c'est le bon avec les pinoches correspondantes. Curieux.
Avez-vous bien vérifié que ça vous ouvre un port com en le branchant ?
Dans gestionnaire de périphériques, catégorie ports (COM et LPT).

fairecasoimeme, 2023-10-25:
si la led bleue clignote très rapidement sur le TX c'est qu'il fait des demandes pour flasher mais que ça ne passe pas.
Etes-vous sur que le port COM n'est pas déjà utilisé par une autre application ?
Avez-vous retiré la pile ?
La procédure est : On appuie sur le bouton flash en le maintenant, puis on insère le module USB (dans le bon sens) puis on relache le bouton et tout en maintenant le module USB, on lancer la commande de flash

diliak, 2023-10-25:
Après, si tout est bon de partout, il y a peut-être une mauvaise soudure/soudure sèche du module sur le Zipulses.
Personnellement, j'ai soudé des pinoches (4) sur le Zipulses au pas standard (2,4mm, de mémoire) et j'utilise des fils avec connecteurs adéquats de bout en bout.

fairecasoimeme, 2023-10-25:
Il n'y a pas de soudure côté ZiPulses à faire
(le pas est de 2.54mm)
[image]

diliak, 2023-10-25:
Curieux, en ce qui me concerne, le module nageait dans les plots. C'est pour ça que j'ai soudé.
Au moins, comme ça, je n'aurai jamais de soucis.

florianbelhomme, 2023-10-25:
J'ai aussi des échecs de mon côté. Après un premier essai, sans pile, j'ai eu une `Error opening connection:`, au deuxième essai j'ai eu `Error opening connection: Read error`. Maintenant quand je branche la clef dans Windows (même seule) le système me dit que "le périphérique se comporte anormalement" et la led du ZiPulses ne s'allume plus du tout. Je pense qu'il c'est passé quelque chose qui a abimé les deux cartes.

diliak, 2023-10-25:
Des fois, en cas de mauvais fonctionnement d'un périph' USB, celui-ci se met en mode dégradé avec l'objet connecté.
Essayez de changer de port USB. Vérifiez aussi que le module est bien emboité dans le zipulses et que les contacts se font bien.
(Perso, j'ai ajouté un support soudé) et SURTOUT, assurez-vous d'avoir les pilotes du module utilisé.

Il faut toujours brancher le module seul afin de voir s'il monte un port série. Vérifier ici : 
Bouton droit sur "Ce PC",
Dans gestionnaire de périphériques, catégorie ports (COM et LPT).

Ensuite, débrancher le module, brancher ce qu'il faut programmer dessus et faire les manips.

florianbelhomme, 2023-10-25:
Merci pour les infos. J'avais vu ce thread avant de tenter la MAJ, j'avais bien les pilotes de silabs.com et la clef apparaissait correctement dans la liste du matériel.
J'ai essayé différents ports USB. Je ne serais pas dire ce qui c'est passé, mais maintenant quand je branche la clef seule, windows me dit que le périphérique fonctionne anormalement. De son côté, le ZiPulses ne donne plus de signe de vie.

diliak, 2023-10-26:
Si ça se trouve, il y a eu un mauvais contact à un moment lors de l'upgrade et le Zipulses n'a pas reçu tout le firmware.
Je ne sais pas si cette situation est possible. Le créateur du Zipulses en sait peut-être plus que moi à ce sujet.

Perso, ce que je ferais c'est rachat d'un autre module de prog (moins de 10€ les deux sur Amazon), tests à nouveau en le branchant seul sur le PC avant tout. (là, vous verrez si ça vient du PC ou du module, d'ailleurs, je suppose qu'en branchant autre chose, le port USB fonctionne ?)

Ensuite, je souderais des pinoches sur le zipulses et je recommencerais l'upgrade avec des fils type Dupont entre le Zipulses et le module. 
Je n'arrive pas à comprendre comment les contacts peuvent être bons entre le Zipulses et le module ou alors, celui que j'ai eu avait un défaut de fabrication mais le module nageait sérieusement dans les perçages. Pourtant les connecteurs sont au pas standard et en épaisseur standard également. (peut-être les puits pas assez étamés ?)
Si j'avais tenté un upgrade comme ça, je pense que j'aurais probablement eu le même problème.

Bien entendu, ce que je dis n'engage que moi d'autant que j'ai peut-être perdu la garantie sur le Zipulses (mais ça j'assume).

fairecasoimeme, 2023-10-26:
Avec le logiciel de flash, cela se fait en deux étapes, la programmation et la vérification de la programmation. Si la programmation va au bout, cela garanti que le ZiPulses est correctement programmé.
Bien entendu que les pins du module USB "nage" dans le port de programmation du ZiPulses, il faut simplement le maintenir en le penchant pour les contacts se fassent. C'est comme cela que ça se fait en général. Vous pouvez souder un connecteur si cela vous arrange mais ce n'est pas obligatoire.

florianbelhomme, 2023-10-31:
J'ai du nouveau.

J'ai HA 2023.10.5 / Z2M 1.33.1-1 / Mosquitto broker 6.3.1 / Clef SONOFF ZigBee 3.0 EFR32MG21

Mon aventure "so far". J'ai commandé sur Lixee deux ZiPulses y a quelques mois. Ils ont arrivé en firmware 4. J'en ai mis un en fonctionnement et j'ai entreposé l'autre pour plus tard en attendant des travaux. Quand le firmware 6 est arrivé, j'ai voulu profiter de l'évolution au niveau du pourcentage de la batterie. J'ai acheté une clef `CP2102 USB 2.0 to TTL UART Serial Converter Module 5Pin STC PRGMR` en noname sur Amazon pour faire la MAJ. J'ai essayé plusieurs fois de flasher le ZiPulses, sans succès (voir messages ci dessus) jusqu'au moment où le ZiPulse et la clef ont cessé de donné signes de vie.
Ne voulant pas rester sur un échec, j'ai recommandé un ZiPulses et la clef de MAJ sur Lixee. Je les ai reçu aujourd'hui. Le nouveau ZiPulses est arrivé en firmware 6.

J'ai tenté la MAJ du ZiPulses qui était en fonctionnement. Le logiciel a indiqué une réussite.

Seulement je suis perplexe par rapport au résultat. Quand je l'associe de zéro (désappairage forcé + oubli puis nouvel appairage). Z2M m'indique que le firmware est en 4 MAIS affiche le pourcentage de la batterie ce qui ne faisait pas avant. Et il m'est maintenant impossible de définir "UnitOfMeasure" j'ai une erreur systématiquement.

[image]

[image]

Sinon le nouveau ZiPulse a la même erreur quand j'essaie de définir "UnitOfMeasure". Problème lié au Firmware ?

PS: j'ai essayé de flasher le ZiPulse qui ne donnait plus signe de vie avec la nouvelle clef, ça a indiqué "read error" du coup je crois c'est mort mort ^^

florianbelhomme, 2023-11-01:
@Pigouchet pour le soucis "reading zcl" (cf image précédente) il semble que ce soit un PB de l'interface Z2M. J'ai pu contourner le soucis avec la "Console dev". Elle permet de lire et écrire les éléments.

[image]

"unitOfMeasure" semble être une valeur numérique basée sur le menu déroulant de "Expose" ex: kwh = 0, m3 = 1 (j'ai donné un mauvais exemple dans l'image, pour les m3 j'ai réglé sur "multiplier 1 divisor 100 unitOfMesure 1")

florianbelhomme, 2023-11-05:
[image]
[image]

Ça confirme que c'est bien un probème de Z2M. Bien que j'avais fait une "suppression forcée / nouvel appairage", on dirait qu'il avait gardé cette information du précédent appairage, y a peut être du cache quelque part.

ortigosano, 2023-11-15:
Hello everyone!

[image]
I have purchased two pulse water meters model 

https://shop.smarthome-europe.com/fr/suivi-energie/2127-gioanola-compteur-d-eau-chaude-a-impulsion-3-4-1-imp-litre.html

/[ manufacturer Gianola]([image])

[](https://shop.smarthome-europe.com/fr/suivi-energie/2127-gioanola-compteur-d-eau-chaude-a-impulsion-3-4-1-imp-litre.html )

And two units of ZiPulses.

I thought this came configured to work more easily, like any other ZigBee device (plug and play).

My intention is to be able to visualize in HomeAssitant the liters of water consumed rather than cubic meters (although the "analog" water meter shows cubic meters). 

I do not know the equivalence in pulses to liters.

The problem I have is the pulse meter (ZiPulses) is configured to measure kwh (energy) instead of pulses.

Unfortunately, I do not know French. I would like to know what parameters I have to configure in the "Dev Consle" tab of Zigbe2mqtt to be able to count liters.

Thanks

florianbelhomme, 2023-11-16:
In order to ZiPulses to report values, you must set the "multiplier", "divisor" and "unitOfMesure". Actually there is a bug in Z2M that prevent to set those parameters through the UI.

[image]

Follow the number in this image and select the 3 parameters. Once done, set the right values for "multiplier", "divisor" according the "tic" from your water meters Gianola. For "unitOfMesure" look at this table : https://github.com/fairecasoimeme/ZiPulses#tableau-dunit%C3%A9-de-mesure for liter it's 7.

After that, click on "Write", the red button then push the button on the ZiPulse (a very short push to make it blink blue once).
Then you should have a succesfull  green message on HA.

Then if you push "Read" (the green button) then push the button on the ZiPulse (a very short push to make it blink blue once), it should display the setted values in JSON. And if you go back to the tab "Expose" it should display the values you setted.

The ZiPulses is ready, so you can connect it to your water meter or wait 2H for it to report the index, temperature, etc...
(If the index value is less than 1 (between 0 and 1), you should go to the "Parameter (specific)" tab to set "Energy precision" to 2 to have 2 number after the dot and be able to have values like 0.25 and confirm than the ZiPulses is working).

ortigosano, 2023-11-16:
> In order to ZiPulses to report values, you must set the "multiplier", "divisor" and "unitOfMesure". Actually there is a bug in Z2M that prevent to set those parameters through the "Expose" tab. But you can set them manually like that :
> 
> [image]
> 
> Follow the number in this image and select the 3 parameters. Once done, set the right values for "multiplier", "divisor" according the "tic" from your water meters Gianola. For "unitOfMesure" look at this table : https://github.com/fairecasoimeme/ZiPulses#tableau-dunit%C3%A9-de-mesure for liter it's 7.
> 
> After that, click on "Write", the red button then push the button on the ZiPulse (a very short push to make it blink blue once). Then you should have a succesfull green message on HA.
> 
> Then if you push "Read" (the green button) then push the button on the ZiPulse (a very short push to make it blink blue once), it should display the setted values in JSON. And if you go back to the tab "Expose" it should display the values you setted.
> 
> The ZiPulses is ready, so you can connect it to your water meter or wait 2H for it to report the index, temperature, etc... (If the index value is less than 1 (between 0 and 1), you should go to the "Parameter (specific)" tab to set "Energy precision" to 2 to have 2 number after the dot and be able to have values like 0.25 and confirm than the ZiPulses is working).

Thanks for your quick answer!
I did it as you described it but after clicking again  into "dev console" it seems the values are missed. 
Even after getting the green messages when pushing the button. Which means, in theory it seems to have gone well, but in practice I don't really know.

I have try with the Edge version of zigbee2mqtt and everything works as it should , therefore I think it may be more worthwhile to wait for the new zigbee2mqtt release. Don't you think?  or maybe switch to the Edge version for ever, but may have more stability issues...

Merci