fairecasoimeme/ZiPulses #22: mise à jour OTA

mickae1, 2023-10-21:
bonjour il ne semble pas avoir de firmware pour mise a jour OTA

fairecasoimeme, 2023-10-23:
Bonjour,
comme la plupart des appareils zigbee sur pile, il n'y a pas de mise à jour OTA, sous peine de vider les piles.

pdecat, 2023-10-23:
Hmm, la plupart de mes appareils sur piles supportent l'OTA, notamment ceux de marques Ikea, Legrand et Aqara/Xiaomi.

fairecasoimeme, 2023-10-23:
Les capteurs aqara comme les capteurs de temperature / humidité ?
J'ai pas l'impression (mais peut-être que je me trompe) que les capteurs avec des piles boutons permettent les update OTA.

pdecat, 2023-10-23:
Typiquement, j'ai eu plusieurs mises à jour pour ce modèle, de capteur TVOC Aqara, et heureusement car il était très instable au début : https://www.zigbee2mqtt.io/devices/VOCKQJK11LM.html

Ce n'est qu'un exemple, il y en a plein d'autres.

mickae1, 2023-10-23:
> Bonjour, comme la plupart des appareils zigbee sur pile, il n'y a pas de mise à jour OTA, sous peine de vider les piles.

oui enfin je préfère changer une pile que payer l'adaptateur pour la mise a jour...

pdecat, 2023-10-23:
D'autant que j'utilise des piles rechargeables au lithium en remplacement de celles d'origine.

Gonioul, 2023-10-24:
Les boutons/interrupteurs Ikea font OTA par exemple.

fairecasoimeme, 2023-10-25:
Le module usb CP2102 coute moins chère qu'une pile ...
Normalement les piles boutons ne sont pas faites pour qu'on tire 20 mA pendant plusieurs minutes d'affilées. 
Dans tous les cas, si je rajoute l'OTA dans la prochaine version, il faudra flasher avec un module USB...

amy92000, 2023-11-26:
> Le module usb CP2102 coute moins chère qu'une pile ... Normalement les piles boutons ne sont pas faites pour qu'on tire 20 mA pendant plusieurs minutes d'affilées. Dans tous les cas, si je rajoute l'OTA dans la prochaine version, il faudra flasher avec un module USB...

Bonjour je voudrais savoir si l'OTA dans la version 06 ?? Merci

fairecasoimeme, 2023-11-27:
non pas encore d'OTA sur la version v6

amy92000, 2023-11-27:
> non pas encore d'OTA sur la version v6

Merci pour votre réponse j'espère que vous allez l'inclure dans la version 7 ça sera plus facile pour nous de faire la mise à jour comme vous l'avez fait avec celui du zlinky

albanleandri, 2023-11-27:
Bonsoir, je serais aussi en faveur du support de l'OTA même s' il faut brancher via USB l'appareil le temps de la mise a jour pour lui fournir une alimentation plus stable, ne serait ce que par ce que je trouve cela plus simple que le mode de mise a jour non-OTA

MilesTEG1, 2024-10-16:
Bonsoir ,
Je viens de voir qu'il y avait une version 7 du firmware.
Est-elle possible en OTA celle-ci ?

Gonioul, 2024-10-19:
Sans vouloir abuser, est-ce qu'on a une visiblité sur une mise à jour OTA ou USB?
Merci d'avance.

devildant, 2025-02-02:
Bonjour, 

je relance le sujet de la mise a jour OTA, ca serait vraiment un plus, j'ai plusieurs zipulses, certains sont relier au secteur via alim 5v et d'autre via pile a l'extérieurs de chez moi, faire les mises a jour est extrêmes pénible car je doit tout débrancher a chaque fois un vrai calvaire.

comme il sont souvent raccorder au compteur de gaz ou d'eau il sont dans des endroits pas forcement facile d'accès, et devoir tout démonter a chaque fois pour faire une maj c'est un enfer...

abatista75, 2025-12-14:
Bonjour,
Je suis dans le même cas : mon ZiPulses est connecté à une source externe permanente (excellente fonctionnalité, et merci d'avoir proposé plusieurs options d'alimentation).

Par contre, démonter le ZiPulses n'est vraiment pas pratique, et la mise à jour OTA serait vraiment pratique.

Merci