fairecasoimeme/ZiPulses #23: Arrêts de transmissions vers Domoticz sans explications

diliak, 2023-10-24:
Bonjour à tous,
Je remonte cela car j'ai un souci assez curieux.

Avec le firmware V5 :
Tout fonctionne parfaitement pendant 15 jours environs et soudain, plus rien ne remonte. Notons que je n'ai aucune erreur dans le plugin Zigbee for Domoticz. Le Zipulses est toujours vu en ligne... (peut-être que le plugin attend les deux heures afin de le déclarer hors ligne ?)
Je  reset le tout, ça tient à nouveau un une poignée de jours et puis hop... Gérard Majax passe par là...

Le Zipulses est installé à environs 5m derrière un panneau bois donc j'ai écarté "pour l'instant" un souci de communications radio.
J'ai essayé de traiter des éventuels soucis de micro-coupures secteur mais peine perdue. Rebelotte.

Une fois, il m'a suffit d'appuyer une fois sur le bouton et tout s'est mis à jour (donc en réveillant le Zipulses).
Les autres fois, j'ai été obligé de reset (réappairage sans perte des objets). Ca met le souc dans les stats de conso mais bon...

Avec le firmware V6 :
Je mets le Zipulses en tests sur table.
Effectivement, le voyant s'éteint maintenant même si l'ILS est fermé.
Je fais des impulsions rapides à la main, tout fonctionne bien. Des fois, une impulsion ne passe pas mais est retransmise avec une nouvelle impulsion. Si on "secoue" un peu trop (ce qui n'arrive jamais en relève normale) le Zipulses arrête les remontées mais ça refonctionne quelques secondes après. C'est donc très bien, cela veut dire que le zipulses compte bien et mémorise bien les impulsions.

Dans Domoticz, Zipulses sur pile :
Tension 3,3V ou 3,4V (Comment est-ce possible ???) 
Température 21° (pas de surprise...)
Batterie level à 200% (???)
Peut-être une mise à jour du plugin à faire donc pas lié au Zipulses ?

Pour l'instant, difficile d'incriminer le Zipulses ou autre cause.  je cherche à comprendre.

diliak, 2023-10-27:
7 jours après : toujours là.
Je fais entrer des impulsions manuellement de temps en temps.
Posé sur le bureau à environs 3m du coordinateur et environs 1m d'une prise routeur.

diliak, 2023-11-03:
Il fonctionne à priori toujours bien depuis le 21 Octobre avec le nouveau firmware. Je fais quelques impulsions manuellement tous les jours.

florianbelhomme, 2023-11-04:
En firmware 6 sur mon ZiPulse et branché sur mon Gazpar, j'ai un fonctionnement décrit dans la doc. CAD, un ping toutes les 2H en l'absence d'impulsions et un ping quand des impulsions passent.

Avec HA et Z2M, il m'est arrivé que des appareils Zigbee "freeze". Un redémarrage de MQTT et Z2M a corrigé ça, ou un réappairage forcé. Avez-vous tenté ces opérations ? (ou équivalentes dans Domotiz)

diliak, 2023-11-04:
Bonjour,
Merci pour votre réponse.
J'ai globalement tout essayé.
Je n'ai aucun souci avec les autres objets sur l'installation.
Le Zipulses est directement sur le coordinateur. Installation provisoire sur le bureau à environs 3m. L'installation prévue est à moins de 5m.
Pour l'instant, depuis le 20 Octobre, fonctionnement nickel.

diliak, 2023-12-21:
Des news.
Fonctionnement toujours OK. Batterie OK. Mêmes conditions de travail (3m du coordinateur).
Le seul truc que j'ai fait c'est l'upgrader en version 6 du firmware et le rapprocher.
Ce soir, je le remets au même endroit qu'au départ et ferai les impulsions "à la main" comme je le fais actuellement pendant 10/15 jours.
A suivre...

diliak, 2024-01-28:
Re,
Tests toujours fonctionnels. Je pense clôturer ce topic d'ici quelques jours.

diliak, 2024-02-28:
Re :
Pile "enfin" HS. Bon, je ne l'utilisais que peu en tests.
Plus de perte dans Domoticz.
Je vais clôturer ce topic.

closed 2024-02-28 by diliak