fairecasoimeme/ZiPulses #24: Impossible de mettre à jour le firmware Zipulses.

Pigouchet, 2023-10-24:
Je suis dans l'incapacité de mettre à jour le firmware de mon Zipulses.
J'ai tenté sur plusieurs environnement Win10 ou Win11 

Ligne(s) de commande(s) exécutée(s) en tant qu'Administrateur en Powershell (après avoir connecté le Zipulses + Appui sur bouton reset) :

`.\DK6Programmer.exe -s COM3 -P 500000 -e FLASH -Y -p C:\Users\toto\Documents\Zipulses_Bis\ZiPulses_v6.0.bin`

idem avec la commande :

`.\DK6Programmer.exe -s COM3 -P 500000 -Y -p C:\Users\toto\Documents\Zipulses_Bis\ZiPulses_v6.0.bin`

Retour, la mise à jour semble réussir ;

[image]

Mais la version du module Zipulses reste désespérément la même dans Z2M (j'ai dû faire x20 la manip mais pas mieux) : 

[image]

J'ai loupé un truc ????

_Originally posted by @Pigouchet in https://github.com/fairecasoimeme/ZiPulses/issues/21#issuecomment-1775650983_

Gonioul, 2023-10-24:
En ce qui me concerne je n'ai pas le "memory programmed sucessfully"
Je pense donc plutôt vers un problème Z2M.
Essayer de faire une reconfiguration de l'appareil dans Z2M.
Voire redémarrer Z2M avant pour être sûr (module supplémentaire et/ou paramètres)

diliak, 2023-10-25:
C'est ce que je pense également. Votre mise à jour à priori s'est bien passée.

fairecasoimeme, 2023-10-25:
Je confirme que la mise à jour s'est bien passée vu le screenshot. Le problème vient de Z2M

closed 2023-10-25 by fairecasoimeme