fairecasoimeme/ZiPulses #25: Mise à jour du firmware depuis Linux

jiquem, 2023-11-25:
La documentation ne mentionnant pas d'utilitaire à utiliser sous Linux, k'ai tenté la mise à jour du Zipulse avec l'utilitaire trouvé ici : https://zigate.fr/wp-content/uploads/2021/05/x86_64.tar
Mais j'ai un message d'erreur  m'indiquant qu'il n'arrive pas lire sur le device USB. (J'ai vérifié je suis bien sur le bon)
Est-ce que j'ai "loupé" qqchose ?

```
user@nuc:~/zipulse/x86_64$ ./DK6Programmer -s /dev/ttyUSB0 -P 115200 -Y -V 2 -p ZiPulses_v6.0.bin
/dev/ttyUSB0: Error opening connection: Read error
```

fairecasoimeme, 2023-11-27:
Pour la ZiGate USB+ /ZiGate-Ethernet et la ZiGate-DIN (pilote FTDI)

Pour les pilotes FTDI, il est nécessaire d’utiliser la valeur ZIGATE+ à la place du port série.
Pour être sur de la valeur, il est possible de lancer la commande :

$ sudo ./DK6Programmer -l

Sans remise à zéro (Conserve les appareils déjà appairés):

$ sudo ./DK6Programmer -s ZIGATE+ -P 115200 -Y -p <CHEMIN_BINAIRE.bin>

Avec remise à zéro (Efface toutes les données (appairages compris)):

$ sudo ./DK6Programmer -s ZIGATE+ -P 115200 -e FLASH -Y -p <CHEMIN_BINAIRE.bin>

nlindblo, 2024-02-24:
@jiquem was this resolved for you ? I am experiencing the same Error opening connection: Read error message. Using the ZIGATE+ as port as suggested here just gives me an "Error opening connection". I have confirmed that my port is /dev/ttyUSB0 and operating system id Debian 10
[image]

jiquem, 2024-02-26:
Hello
I've sent it back, and it was fixed by the provider

jiquem, 2024-02-27:
The ZLinky_TIC, after exchanging through the contact https://lixee.fr/nous-contacter