fairecasoimeme/ZiPulses #28: Zipulses - plus de mesure depuis la mise à jour en version 6

ekurgan, 2023-12-06:
J'ai mis à jour le firmware du Zipulses de la version 4 à 6 afin de récupérer l'état de la batterie mais depuis la valeur "energy" ne se met plus à jour.
J'ai tenté une mise à jour du firmware avec reset, exclusion/réinclusion mais rien n'y fait.
Je suis sur HomeAssistant et Zigbee2MQTT. Le zipulses est connecté au compteur gaz et a bien fonctionné en version 4 durant 6 mois. Je reçois des mises à jour des autres valeurs régulièrement mais energy reste désesperement à 1:
{
    "battery": 100,
    "divisor": 100,
    "energy": 1,
    "last_seen": "2023-12-06T13:48:08+01:00",
    "linkquality": 135,
    "multiplier": 1,
    "temperature": 11.37,
    "unitOfMeasure": "m3",
    "voltage": 3200
}

Ce que je trouve bizarre est que Zigbee2MQTT me dit que je suis toujours en version 4
[image]

Une idée de ce qui pose problème ?

ElGibs, 2023-12-19:
Bonjour,
Je me permets de rajouter ici, dans le doute que ce soit lié peut-être à un souci de détection spécifique à une source Gazpar, le même problème: j'ai 2 ZiPulses depuis 2 semaines. Les deux mis à jour au firmware 0.6 dès leur mise en service (quasi impossible à appairer sous HA/ZHA sinon, très instables et gouffres à piles). Hardware PCB v.0.5.
Le premier (eau) fonctionne parfaitement (compteur type Gianola).
Le second (Gazpar), compte "correctement" lorsque je le stimule avec un fil pour simuler des impulsion (au bout d'un moment j'ai bien vu remonter une valeur incrémentée). Une fois tout éteint et monté près du compteur, tout démarre bien, j'ai bien le niveau de batterie et la température de façon régulière. Mais aucune détection d'impulsion du Gazpar. Dans le doute d'un souci de polarité (ne sachant pas vraiment la nature du stimulus côté Gazpar) j'ai tenté d'inverser les fils venant du Gazpar, rien non plus.
J'ai tenté de désassocier, et réassocier, rien non plus.

Des idées ?

Merci !

florianbelhomme, 2024-01-01:
> mais energy reste désesperement à 1

Est-ce toujours à 1 ? Si tu n'a pas augmenté le nombre de chiffres après la virgule, si tu consomme moins d'un m3 de Gaz, le compteur reste à 1 et on dirait qu'il ne se passe rien. Si les autres chiffres remontent correctement, c'est bon a priori.