fairecasoimeme/ZiPulses #29: ZiPulse - Comptage impulsions Gazpar ne fonctionne pas

ElGibs, 2023-12-20:
Bonjour,

J'ouvre une nouvelle issue car cela semble différent de la situation dans laquelle j'avais initialement posté.

J'ai 2 ZiPulses depuis 2 semaines. Les deux mis à jour au firmware 0.6 dès leur mise en service (sinon ils étaient très difficiles à appairer sous HA/ZHA, instables et gouffres à piles). Hardware PCB v.0.5.

- Le premier (eau) fonctionne parfaitement (compteur type Gianola).

- Le second (Gazpar), compte "correctement" lorsque je le stimule avec un fil pour simuler des impulsion (au bout d'un moment j'ai bien vu remonter une valeur incrémentée). Une fois tout éteint et monté près du compteur, tout démarre bien, j'ai bien le niveau de batterie et la température de façon régulière (toutes les 2h). Mais aucune détection d'impulsion du Gazpar.

Tentatives et recherches:
- Dans le doute d'un souci de polarité (ne sachant pas vraiment la nature du stimulus côté Gazpar) j'ai tenté d'inverser les fils venant du Gazpar, rien ne change(*).
- J'ai tenté de désassocier, et réassocier du Zigbee, rien non plus.
- J'ai tout réinitialisé, réassocié, puis remis le diviseur (100) et l'unité (m3 - valeur 1), rien non plus.
- J'ai vérifié la continuité électrique depuis le bout de la prise JAE (câble de 1 mètre) jusqu'aux pins chipset(+) et GND sur le PCB, sans anomalie constatée.
- J'ai mesuré environ 3V DC aux bornes de comptage du ZiPulses

Y a-t-il moyen de tester d'autres réglages de durée d'impulsion, ou de faire clignoter la LED en cas d'impulsion (je suis maintenant en v0.6 elle est donc éteinte) le temps d'un test ?

Je suis à court d'idées. La seule possibilité que je vois est que le premier branchement du ZiPulses a été fait en inverse de la polarité proposée. La première photo [https://github.com/fairecasoimeme/ZiPulses?tab=readme-ov-file#exemple-de-schema-montage](url) montrait en effet un câble marron comme GND et gris comme "impulsions".
Il m'a semblé lire quelque part (mais je ne retrouve plus où) que le Gazpar pourrait condamner sa prise en cas d'anomalie. Avez-vous un test simple à proposer pour m'assurer qu'il fournit bien une impulsion, éventuellement ?

Merci !

ElGibs, 2023-12-28:
Bonjour,

Pardon de relancer, mais personne n'a d'idée(s) pour investiguer ?

J'ai bien les températures:
[image]

J'ai bien la batterie:
[image]

Mais pas le comptage:
[image]

Le "pic" que l'on voit correspond à mes tests d'impulsions avec un simple bout de câble, avant de définir le diviseur sur 100 et de mettre en place le ZiPulses sur le Gazpar.

Merci et bonnes fêtes de fin d'année.

florianbelhomme, 2023-12-30:
Par rapport aux tests que vous avez effectué, il ne reste plus que quelques éléments non testé :
- un problème de connexion entre votre câble JAE et la prise du Gazpar
- un problème de la sortie d'impulsion du Gazpar
- la pile interne du Gazpar est morte (mais bon, j'imagine que le distributeur de gaz serait passé vérifier)

Avez vous la possibilité de contrôler la sortie du câble JAE (avec le câble bien branché dans prise du Gazpar) avec un multimètre pendant qu'une consommation de gaz est en cours ? (chauffage ou eau chaude). En fonction du résultat, testez également directement la prise du Gazpar pour éliminer les derniers défauts possible.

Fabhito, 2024-01-02:
Bonjour à tous 
j'ai le même problème que @ElGibs sauf que j'allai installer le ZiPulses. 

J'ai acheter le module ZiPulses en avril 2023 et j'étais en recherche de solution pour l'alimenter. Je pensai utilisé la ligne de mon interphone avec un régulateur LM2596S DC-DC mais j'ai vu que ce n'était pas possible, j'ai donc opté pour de la pile 18650 (2,6v-4,2v)  et réutilisé mon régulateur pour brider la tension de sortie de la pile. 

Avant montage de l'ensemble j'ai MaJ le ZiPluses vers le firmware de 0,5 à 0,6 et ce sans trop de problème.
J'ai bien fait attention à ne pas alimenter le module à plus de 3,2v en passant par les bornes de prog histoire de ne pas avoir la surconsommation du régulateur.
J'associe le ZiPulses à mon réseau domotique, fait les réglages de comptage (m3) et diviseur, le ZiPulses se réveil bien quand on le sort de veille par le bouton d'association (la led s'allume) mais après pas de comptage quand je shunt les bornes. 
Je démonte mon montage et tente de l'alimenter en 5v via le régulateur interne et en 3v par la pile mais dans des cas de figure tester le comptage est incrémenté. 

Le module en lui même fonctionne bien et donne la température depuis 2/3 jours mais pas les impulsions qu'il n'enregistre pas. 

Je ne sais plus qui tester et j'espère que je n'ai pas grillé. 

Si @fairecasoimeme à une idée d'autre test a faire je suis preneur.

ElGibs, 2024-01-02:
Attention si tu as mis le diviseur sur 100, il faut logiquement 100 impulsions minimum avant de voir "1" (m3). J'ai fait mon test avant de mettre le diviseur.
Pas chez moi cette semaine, merci pour les suggestions, j'espère voir au multimetre les impulsions du Gazpar, à mon retour.

Radeon24, 2024-01-06:
Bonjour,
Je suis également sur le dernier firmware et j'ai l'impression d'être dans un cas de figure similaire. Sauf que tout a fonctionné "correctement" durant plusieurs semaines. Enfin presque, juste une journée avec une mise à jour de 14m³ de trop par rapport au relevé fournisseur.
Mais depuis deux semaines, aucun index remonté alors que la température se mettait à jour correctement. Et hier soir le Zipulses a envoyé 90m³ d'un seul coup. Même cette valeur n'est pas cohérente avec l'index de retard des deux dernières semaines (consommation réelle <1m³ par jour).
[image]
J'essayerai d'aller vérifier la connectique dans les prochains jours, mais ça ressemble quand même plus à un bug du Zipulses.

florianbelhomme, 2024-01-06:
@Radeon24 par rapport au relevé fournisseur : vérifiez le coefficient qui doit être présent sur votre facture de consommation, c'est un multiplicateur a appliquer au relevé du Gazpar, qui correspond a l'élévation de votre commune.

Sinon pour les "saut", il est possible que ça corresponde à une période où le ZiPulse n'a pas réussi a contacter son noeud Zigbee pour rapporter la consommation. Il a continué a compter l'index et plus tard, quand la connexion à été rétablie, la mise-a-jour a pu se faire et ça se traduit par une marche. Il faut voir si la marche correspond à votre consommation potentielle par rapport à la période où aucune MAJ n'a été faite. Eventuellement la pile est peut être faible ?

ElGibs, 2024-02-05:
Suite et fin:
Pour diverses raisons, j'ai basculé de ZHA à Zigbee2MQTT. Durant la bascule, j'ai totalement sorti mes équipements Zigbee, puis fait l'appairage 1 par 1. Aucun problème pour le Zipulses de mon compteur d'eau (avec appui sur le bouton), par contre totalement impossible de faire rejoindre mon ZiPulses dédié au compteur de gaz (celui qui ne remontait rien d'autre que voltage et température). Y compris en le ramenant près du coordinateur Zigbee.

DONC...
J'en ai commandé un autre, directement chez Lixee. Cette fois c'est une "v2" (v0.7 sur le PCB et non v0.5). Appairage instantané, j'ai un peu lutté pour l'alimentation sur piles, vu que mon bloc piles fournit du 3v je pensais naïvement avec la v2 que ce voltage était pris en charge via le bornier ;-) Bref, une fois ressoudée mon bloc pile au bornes 3v3/gnd, et branché, miracle !

Pourtant, je suis persuadé que mon ZiPulses v0.5 fonctionnait, il remontait bien des incréments du cumul 'energy' en shuntant les bornes 'pulse'  de façon répétée. J'émets donc l'hypothèse que la v0.5 est /possiblement/ différente de la v0.7 du point de vue de la captation des impulsions ? @fairecasoimeme  vous confirmez ?

Merci.

ElGibs, 2024-02-07:
> Le régulateur utilisé au travers du bornier, démarre à partir de 3.3V minimum. c'est pour cela qu'il est marqué de 5V à 12V

Oui oui, c'était pourtant logique, je me suis trouvé bête 😅

> La version v0.5 utilise une résistance de pullup de 10k alors que la v0.7 utilise une résistance interne de pullup. C'est la seule différence côté hardware mais n'influe pas sur le fonctionnement de la captation des impulsions.

Merci pour cet éclaircissement, effectivement présenté comme ça, mon hypothèse est fausse. En tout cas ça fonctionne en v0.7 sans autre formalité, avec le même câble JAE, mêmes piles et même compteur. Je vais clore mon issue.

closed 2024-02-07 by ElGibs