fairecasoimeme/ZiPulses #30: hors ligne

sergelhr, 2024-03-06:
Alors que je n'ai aucun problème avec ZLinky_TIC, le ZIpulses me prends la tête depuis le début.
Cela fonctionne un certain temps et je me retrouve ensuite hors ligne  .
Sous zigbee2mqtt/HA dernière version et probablement dernier firmware du zipulses , du moins je le pense, affichage de version 04 pour un firmware 06 par mes soins.
Comment faire un diagnostique pour comprendre la raison du hors ligne? alimentation en 5v via alim et compteur eau capteur ils.
[image]

fairecasoimeme, 2024-03-07:
Bonjour,
le LQI à 0 semble bizarre.
La seule chose qui peut rendre le ZiPulses hors ligne c'est qu'il n'ait pas pu communiquer 1 seule fois depuis 2-3j. Si c'est le cas quand vous appuyez sur le bouton link, clignote-t-il rapidement.
Si ce n'est pas le cas, c'est autre chose.
Je ne sais pas comment Z2M calcule la disponibilité. Le ZiPulses communique toutes les 2 heures s'il n'y a pas d'impulsion.
si vous simuler une impulsion, le produit revient-il disponible ?

sergelhr, 2024-03-07:
en ce moment la led clignote , une fois par seconde

fairecasoimeme, 2024-03-07:
S'il se met a reclignoter rapidement, c'est qu'il sort du réseau zigbee
A la vue de la network map, le ZiPulses passe par le ZLinky qui a un LQI faible.
Les pertes de données sont probablement liées à ça. Pour régler le problème, il faut probablement améliorer la portée du ZLinky

sergelhr, 2024-03-07:
[image]

sergelhr, 2024-03-07:
après suppression dur réseau et nouvel appairage, par contre pou l'unité et div et mult j'ai bien suivi le tuto mais je dois être nul  retour msg en rouge

fairecasoimeme, 2024-03-08:
il faut bien appuyer sur le bouton link lors de l'envoi de requête de modification. Si cela ne fonctionne toujours pas, essayer de rapprocher le ZiPulses pour la modification