fairecasoimeme/ZiPulses #31: Zipulse mort après 2h de fonctionnement

Snoofy26, 2024-03-28:
J'ai mis mon zipulse sur un compteur gazpar après appairage.
Tout fonctionnait nickel mais au bout de 2h plus rien.
Je me disais que c'était un pb de portée mais lorsque je l'ai remis à côté du coordinateur et que j'ai appué sur le boutton d'appairage rien ne s'est passé. La led ne s'est même pas allumée.
J'ai d'abord pensé à la pile mais j'ai pris sa tension au multimètre mais la pile est à 3.2V donc ce n'est pas ça.
Un reset n'a rien fait non plus.
Qu'est ce que je loupe ? Un composant aurait grillé ?

fairecasoimeme, 2024-04-02:
Avez-vous tenté d'appuyer une dizaine de secondes sur le bouton d'appairage ? le switch est bien positionné sur BAT ?

Snoofy26, 2024-04-21:
C'est bon, le switch BAT/ REG n'était pas au bout de sa course, en le rouvrant j'ai du le bouger sans faire attention.
Ca refonctionne désolé pour le dérangement

closed 2024-04-21 by Snoofy26