fairecasoimeme/ZiPulses #33: Update unit_of_measure in ZHA

fredoche1810, 2024-06-13:
Hello,
I try to change the value of the unit of measure as explained here : https://github.com/fairecasoimeme/ZiPulses?tab=readme-ov-file#zha
When I go in Home Assistant on the Zipulse to reconfigure I can find the parameters.
When I read the information of the unit_of_measure, I receive the following value : enum8.undefined_0x00
When I try to change it to "7" as value to have Liters, the value is send without error but it doesn't change anything. Moreover, when I read it again, it's always the default value enum8.undefined_0x00
Do you have to insert the value in a specific format ?
It's the first time for me that I have to use enum8 as format.
Thanks in advance,
[image]

fairecasoimeme, 2024-06-17:
Did you get the Green V, when you click on "écrire l'attribut" ?
Did you push the ZiPulse's link button ?

fredoche1810, 2024-06-17:
Hello @fairecasoimeme,

Yes I have a green V when i click on "écrire l'attribut". 
Yes I pushed on the ZiPulse to wake it up. Without that no green V. 
I think that the format of the value isn't given correctly due to enum8. Only int 7 for liters isn't enough no ?

I suppose when I see your nickname that we can also do it in French.

Thanks,

fairecasoimeme, 2024-06-17:
il me semble qu'en mettant juste 7 dans l'étiquette valeur suffit.

fredoche1810, 2024-06-17:
C'est ce que j'ai fait ... Je n'ai pas d'erreur lorsque je fais "écrire l'attribut" mais quand je relis la valeur elle n'a pas été modifiée malheureusement