fairecasoimeme/ZiPulses #34: Déconnexions

helepeut, 2024-06-15:
Bonjour, 

J'ai acheté début avril un ZiPulses pour mesurer la consommation de Gaz. 
Le compteur de gaz se trouve à l'extérieur de la maison, à quelques mètres. Pour maximiser la portée, j'ai installé à l'intérieur de la maison un ZBDongle E (Sonoff) configuré en routeur.
Le ZiPulses utilise 2 piles LR20. 
C'est une configuration Zigbee2Mqtt.

Depuis l'installation, j'ai assez régulièrement des absences de remontée d'information. Parfois j'ai pu relancer la connexion en appuyant sur le bouton du ZiPulses, d'autres fois en retirant et en remettant les piles, car j'ai vu qu'à la mise sous tension il fait un "rejoin". 

[image]

Mais dernièrement, impossible de récupérer les donnée de consommation, je ne parviens pas à trouver la bonne manip. Le Zipulses s'allume en réponse aux impulsions.
- un clic sur le bouton ? (il s'allume 1 fois et s'éteint)
- retrait et remise en place des piles ?
- clic de 3 secondes ? 
- clic de 10 secondes ?

Il semble pourtant que lorsque je fais mes manips, il est bien vu par Zigbee2Mqtt et met à jour l'information 'Last Seen'.

J'ai peur de supprimer le ZiPulses du réseau et de le réintégrer en appuyant 10 secondes, car à la fois précédente il s'est réinitialisé et a remis le compteur à 0.

J'ai deux questions : 
- dans ces cas là, quelle serait la bonne méthode pour relancer la connexion, ou l'appairage sans réinitialiser les données comptées
- et comment préserver le ZiPulses des connexions ? sachant que pendant presque 1 an j'ai utilisé un interrupteur ikea comme compteur d'impulsion, mais pas assez fiable sur le long terme car des impulsions étaient perdues.

Merci de votre aide !

biwizit, 2024-06-16:
Ah les déconnexions du Zipulses... ça m'aura bien occupé pendant des mois et je ne suis pas sûr d'avoir réussi à le dompter...

J'ai déjà observé les mêmes soucis que vous.

Si le click ne produit qu'un allumage, c'est qu'il se considère toujours dans le réseau.

Si éteindre et rallumer ne change rien (normalement, ça entraîne un rejoin), il faudra passer par un appui de 10 secondes pour sortir du réseau afin d'y revenir après ça. Avec comme conséquence en effet, une réinitialisation du compteur.

Pour éviter ces déconnexions, j'ai vraiment essayé beaucoup beaucoup de choses. Comme passer du Zdongle E au Zdongle P pour ne plus être dans un mode expérimental de Zigbee2mqtt, etc... 

Ce qui m'a un peu démoralisé il y a quelques jours, c'est qu'entre deux essais, j'ai laissé mon Zipulses dans le salon, vraiment pas loin du coordinateur et le device a fini par se mette en hors-ligne, ce qui n'avait aucun sens.

Depuis quelques jours, j'utilise un ZBMini de Sonoff (dans une boîte etanche de la même marque) comme routeur intermédiaire au plus près de mon Zipulses qui se trouve côté rue, pour avoir une liaison assez robuste (LQI de 130), en faisant en sorte de mettre mon coordinateur assez loin à l'intérieur de la maison pour que mon Zipulses n'essaie pas de s'y connecter.

Comme les mystères du Zigbee sont impénétrables, il lui arrive souvent d'aller chercher un routeur au LQI beaucoup plus faible (dans les 50, avec mon Zlinky par exemple), jusqu'à en perdre la connexion... Du coup, j'ai essayé de faire un réseau Zigbee dédié le plus petit et déterministe possible pour éviter de telles situations.

Mais c'est très frustrant de voir ce device disparaître aussi facilement d'un réseau quand un Aqara à 5€ (acheté à ce prix il y a longtemps, j'avoue :- ) arrive à résister des mois sans souci dans les mêmes conditions.

helepeut, 2024-06-24:
Merci @biwizit, je me sens un peu moins seul !

Entre temps, j'ai réinstallé le routeur ZB Dongle E qui fait relai au plus proche du Zipulses (mais 10 mètres tout de même, car je n'ai pas encore envisagé de prendre la pelle pour tirer de l'électricité et le mettre dans le jardin, et comme vous : un interrupteur ikea était beaucoup plus stable dans la même configuration, et pourtant dénué d'antenne dédiée).
Et il a assez souvent des absences, bien qu'il soit sensé faire un rapport toutes les 2 heures ?

Alors en passant devant, je me suis dit que j'allais appuyer sur le bouton pour le réveiller, et là il se réveille et me remonte 100m3 de consommation d'un coup (bien évidemment, avec 2 douches uniquement, ca me parait cher la douche), comme vous. J'avais observé cela déjà dans mes précédents tests, mais comme je testais pas mal de choses j'avais cru que c'était ma faute. Mais là je n'ai fait que cliquer sur le bouton, 2 fois. 

Alors que le Zipulses aille faire un tour de temps en temps, et que l'on n'a pas une vue vraiment réaliste de la consommation à un instant t, admettons. Mais si quand on le réveille il se met à remonter n'importe quoi et fausse toutes les statistiques, ca commence à être vraiment décourageant.  

Pour ma part je suis à 2,8V / 2,9V avec les 2 piles LR20 (non rechargeables), est-ce trop peu ? Car j'ai du mal à comprendre la documentation : une pile CR2450 (3V) ou une alimentation externe de 5V à 12V. Et plus loin dans la documentation, avec une alimentation externe avec 2 piles, on est sur du 3V théorique. 

Depuis que vous avez augmenté la tension des piles, est-ce que cela va mieux ?

net1957, 2024-07-12:
Même soucis chez moi!

Pour le prix ce produit est très capricieux. Cela va se terminer par un montage filaire ou wifi avec un ESp32 ou un vulgaire bouton zigbee avec un jeu de piles  + puissantes

biwizit, 2024-07-20:
@helepeut depuis mon dernier message, le Zipulses a par deux fois disparu un ou deux jours pour finir par réapparaître, avec une valeur de compteur beaucoup plus importante. Comme s'il l'impulsion avait duré une éternité... Du coup, pour ces jours là, pas de mesure (ou mesure, mais à effacer). Pour ce qui est de l'alimentation externe, le schéma et les composants ont changé au cours du temps et au final, ça fonctionne  effectivement avec deux piles mais j'ai préféré en mettre plus pour lever l'incertitude. En vain... 

Et @fredoche1810 a peut-être raison, mais de mon côté, pas de Home Assistant dans le paysage pour le moment, car je voulais rester sur une solution qui ne m'avait jamais posé de souci jusqu'à maintenant (Raspberry + ZDongle + Zigbee2mqtt nourrissant Domoticz et Home Assistant sur un autre Raspberry). Je n'ai pas eu à redémarrer l'ensemble et j'ai pourtant vécu ce comportement effectivement capricieux du Zipulses...

helepeut, 2024-07-21:
Merci @biwizit ! 
De mon côté, rien changé dans la configuration ni dans l'alimentation. Je l'ai relancé avec une pression, hop 127m3 sont apparus d'un coup en 24h. Il est retombé quelques jours après, un clic n'a rien fait, et j'ai replacé les piles + un clic, et je me suis pris encore 80m3 supplémentaires en 6 jours cette fois (oui, j'ai mis 6 jours à relancer, tellement je suis résigné quant au résultat).
Concernant HomeAssistant, il me semble qu'une fois la donnée remontée dans les statistiques d'énergie, plus possible de corriger.

@fredoche1810, en dehors de la déconnexion, vous avez également un problème de remontées fausses ? Idem pour vous @net1957 ? Et, oui, très agaçant vu le prix en effet.

Car à niveau là, autant faire un thread séparé entre les déconnexions et les remontées erronées au réveil. Si au moins les données n'étaient pas fichues au réveil, ca serait déjà un moindre mal, et peut-être que cela peut être corrigé au niveau firmware.

biwizit, 2024-07-21:
@helepeut pour corriger une erreur dans les statistiques, il suffit d'aller dans "Outils de développement" puis "Statistiques" et choisir le bon sensor d'énergie. Il y aura un petit icône tout à droite pour "Ajuster une statistique". Il suffira alors de cliquer sur "Valeurs aberrantes" en bas et il sera facile de se débarrasser de ces données qui ne correspondent à rien. 

Mais en effet, solution pas très utilisable en l'état...

net1957, 2024-07-25:
Eh oui, encore 2 fois cette semaine. En appuyant sur le bouton, il communique une fois, sans remonter de valeur, on re-appui, la led clignote, mais pas de connexion vue de z2m. Puis plus tard,  il remonte des valeurs incongrue puis communique normalement pendant un certain temps plus ou moins long (de quelques heures a quelques jours).
Ce n'est donc pas utilisable. Le site d'enedis donne des informations plus fiable mais au bout de 2 jours!!
Que faire?

helepeut, 2024-09-01:
Bonjour, de retour de vacances, et malheureusement le Zipulses s'est déconnecté. 
Pourtant il continue de compter les impulsions, je vois la led bleu quand je tire sur le gaz. 
Alors je le relance avec 1 petit clic dessus (pour une fois il va vite), et hop ca n'a pas loupé, 60m3 remontés d'un coup (pour une chaudière éteinte, il va de soi).
@fairecasoimeme Nous sommes plusieurs à avoir ce problème : D'abord les déconnexions au bout de quelques jours ou quelques semaines, et surtout le fait qu'au réveil (forcé) les remontées sont totalement fausses. Je ne sais pas si c'est lié à un diviseur ou un multiplicateur qui n'est pas pris en compte, ou bien si c'est le compteur qui s'est emballé, mais serait-il possible de regarder comment cela peut se produite ?

fredoche1810, 2024-09-01:
Bonjour,
@helepeut oui j’ai eu des remontées farfelues lorsque je poussais sur le bouton pour soit disant le reconnecter. J’ai décidé de passer supprimer Zha de home assistant et de tout faire passer par Z2M. Depuis que j’ai fais cela, je n’ai plus eu des déconnexions… j’ai même pu changer l’unité de mesure en litre ce que je n’arrivais pas à faire non plus sous ZHA. Donc pour l’instant je ne rencontre plus de problème mais je ne comprend pas pourquoi

biwizit, 2024-09-01:
Je suis sur Z2M et je continue d'avoir des déconnexions et des remontées farfules. Donc je dirais : ne pas lui donner de faux espoirs ! :-) Mais tant mieux pour vous si vous avez trouver un équilibre stable. De mon côté, jJ'ai abandonné pour le moment.

helepeut, 2024-09-01:
@biwizit, on lui dit que ca risque de ne pas tenir ? ;-)
@fredoche1810, depuis combien de temps avez-vous changé vers Z2M ? car je suis aussi sur Z2M !

fredoche1810, 2024-09-02:
@biwizit @helepeut, ok j'ai donc peut-être trop d'espoir que cela change quelque chose :-( 
J'ai migré vers Z2M début août donc je dirai 4 semaines sans coupure pour l'instant.
Si cela déconne encore trop souvent, je pense que je vais passer à un ESP32 dédié à cela.

fairecasoimeme, 2024-09-02:
Bonjour à tous,
je vais reprendre les tests pour tenter de reproduire les problèmes.
ce que je retiens :
* Il y a des déconnexions
  * s'il y a une déconnexion du réseau zigbee, c'est qu'il y a eu des trames qui n'ont pas abouti sur une grande période. Cependant comme le device communique que toutes les 2 heures, s'il n'y a pas d'impulsions, ça augmente le phénomène.... je vais regarder pour limiter ou stopper le phénomène.
* Lors de la reconnexion dans le réseau, il y a un grand nombres d'impulsions qui ne semblent pas être cohérent.
  * Même si l'appareil sort du réseau, il continue à compter les impusions et va remonter le nouvel index lors du réappairage. Je vais tenter de reproduire pour voir ce qui ne va pas.

TGuimbert, 2024-09-02:
Bonjour à tous.
Je constate les mêmes soucis que tout le monde ici mais uniquement sur un de mes deux ZiPulses.
Celui qui ne fonctionne pas bien est situé sur un compteur de gaz à l'extérieur de la maison avec une force de signal de mauvaise qualité.
Celui qui fonctionne parfaitement est connecté sur un compteur de panneaux solaire à un mètre environ du coordinateur Zigbee et a donc un très bon signal. Il n'a jamais été affecté par un redémarrage de Z2M ou Home-Assistant.
Les deux ont la même version de firmware 4000-0006 et fonctionne sur pile bouton.

helepeut, 2024-09-04:
Merci @fairecasoimeme !
Si ca peut aider, mon ZiPulses s'est à nouveau déconnecté. Est-ce que vous voulez que je prenne des logs au moment où je vais le réveiller manuellement ?

fairecasoimeme, 2024-09-05:
Bonjour,

J'ai effectué des tests et voici quelques précisions :
* si 3 messages consécutifs n'ont pas reçu d'acquittement, le ZiPulses se met en mode "déconnecté"
* le ZiPulses fait une procédure de "rejoin" uniquement si on appuie sur le bouton
* le Zipulses continue de compter les impulsions même en "déconnecté" et remonte tout d'un coup lorsqu'il a "rejoin" + nouvelle impulsion ou 2h après

PS: Je n'ai pas observé de problèmes d'impulsion délirant ou anarchique. (A chaque fois j'ai compté les impulsions)

Pour régler le problème de gestion des déconnexions j'ai : 
* rajouter une procédure de "rejoin" (si "déconnexion") automatique avant chaque détection d'impulsions (ou toutes les 2 heures)
* si le "rejoin" est effectué, cela remonte toutes les données dans la foulée.

Normalement, avec ces modifications, mêmes s'il y a des "déconnexion" cela reviendra à la normale automatiquement (sans être obligé de cliquer sur le bouton)
le seul inconvénient, avec ce mécanisme, est que s'il y a beaucoup de "déconnexions" (mauvaise qualité radio par exemple) la consommation d'énergie sera plus importante. (le "Rejoin" consomme de l'énergie)

pour ceux qui veulent tester avant que je mette en production, voici le fichier binaire
[ZiPulses_v7.0.zip](https://github.com/user-attachments/files/16887710/ZiPulses_v7.0.zip)

biwizit, 2024-09-06:
Merci pour les explications et la tentative d'amélioration.

Normalement, j'ai tout pour mettre à jour le firmware. Quelle valeur (et à quelle adresse) doit-on avoir pour cette nouvelle version une fois la mise à jour non-OTA effectuée ? Histoire d'être sûr que la procédure s'est bien passé. Merci.

fairecasoimeme, 2024-09-07:
normalement, cluster 0x0000 et attribut 0x4000 doivent donner : 4000-0007 en version

biwizit, 2024-09-07:
Firmware mis à jour, l'attribut me donne bien 4000-007. J'en profite pour remettre un petit coup de charge dans mes 4 accus LR20 cet après-midi pour et je vous tiens au courant. Merci.

fredoche1810, 2024-09-18:
@helepeut @biwizit vous avez raison. Cela n’aura pas tenu plus d’un mois sous Z2M.
@fairecasoimeme je confirme que les valeurs folles remontées après déconnexion sont correctes pour le cumul total. Les valeurs ne sont donc pas erronées mais cumulées et enregistrées sur le moment de la reconnexion. 
De mon côté, ce que je ne comprend pas, c’est pourquoi il ne sait soit disant plus communiquer avec le dongle alors qu’ils sont à 4 mètres de distance, dans la même pièce sans obstacle… 
La mise à jour est la solution magique ?

biwizit, 2024-09-18:
De mon côté, ce n'était clairement pas le cumul mais bien une valeur erronée.
Depuis que j'ai vécu moi aussi ces déconnexions à 4m sans obstacle, j'ai compris que c'était certainement peine perdue.
La mise à jour en 4.7 n'a rien changé (en fait si : j'ai l'impression d'avoir eu des reconnexions mais avec un compteur qui ne bouge plus... avant d'arrêter de communiquer définitivement au bout d'un certain temps).
Bref, en cette période où je viens de rallumer ma chaudière, c'est la déception.

fairecasoimeme, 2024-09-19:
Bonjour,
Merci pour le retour. j'ai réussi à reproduire le blocage de communication (uniquement après une longue désactivation du coordinateur zigbee).
En déconnectant le coordinateur quelques minutes et rebranchant la connexion se fait avec remontée des impulsions. Le blocage de communication ne s'est produit qu'après une longue déconnexion (1 nuit)
Je travaille dessus et le fait de pouvoir reproduire va me permettre de corriger.

fairecasoimeme, 2024-09-19:
@biwizit @helepeut pouvez-vous retester avec le firmware suivant : 
* Augmentation du nombre de messages simultanés + ACK
* Amélioration de la gestion du mode sommeil
* Reconnexion automatique au réseau si trop de pertes 

Bien entendu, tout cela n'impacte pas le comptage des impulsions

Je me suis aperçu que même reconnecté au réseau, parfois l'erreur 0x89 (Transmit failed) persiste, parfois non. Pour régler le problème, après 5 envois ratés, je reinitialise la pile pour que ça reparte.

[ZiPulses_v7.1.zip](https://github.com/user-attachments/files/17062831/ZiPulses_v7.1.zip)

La version à verif est : 4001-0007

biwizit, 2024-09-19:
Merci pour cette mise à jour.
A priori, tout s'est bien passé pour la prise en compte du nouveau firmware mais je n'arrive plus à l'inclure comme avant. Il n'y a que le last seen et le link quality qui apparaissent, le reste est  "null". J'ai essayé sur deux réseaux Zigbee différents avec des clés qui n'ont rien à voir et j'ai le même résultat. Une idée pour que je puisse aller plus loin ? Merci.

[image]
[image]

biwizit, 2024-09-19:
A force de persévérance (redémarrage du serveur Zigbee2mqtt,...), j'ai réussi. Je vous tiens au courant.

helepeut, 2024-09-20:
Merci beaucoup pour la mise à jour, je ne suis pas chez moi en ce moment, je n'ai pas encore pu l'essayer, mais je le ferai dans les prochains jours. Je vous tiens au courant.
Merci encore !

biwizit, 2024-09-20:
La bonne nouvelle, c'est que ça tient toujours au bout de 24h, ce qui n'a pas toujours été le cas. Par contre, je n'ai plus le suivi de la température (et de la tension), peut-être à cause d'une réinclusion imparfaite après la mise à jour du firmware (j'ai lutté, bizarrement). Le suivi de la température est généralement un bon moyen de voir si la communication est solide car la valeur change forcément sur un intervalle de 2h (et si ça reste plat, c'est clairement le signe d'une déconnexion). On verra bien dans le temps.

biwizit, 2024-09-22:
Hello,

Après 3 jours, c'est toujours nickel ! Et surtout, cette fois-ci, le ZiPulses a même été se connecter à un routeur ZB Mini que j'avais mis pas loin à l'extérieur pour faire relais avec mon coordinateur qui est à l'intérieur. C'est exactement le comportement que j'attendais depuis des mois, mais jusqu'à maintenant, le device préférait disparaitre que d'aller chercher une meilleur liaison (lqi dans les 120 en faisant le changement de chemin). Attendons encore quelques jours avant de crier victoire mais je trouve qu'il y a vraiment du mieux depuis les quelques modifs du 4001-0007. Il faudrait que quelqu'un confirme qu'il arrive à remonter les infos de températures et de tensions via zigbee2mqtt. Les champs sont bien dans le payload, mais systématiquement à null. 

Merci.

biwizit, 2024-09-23:
J'ai systématiquement fait mes appairages à l'intérieur de la maison, à 2m de ma clé ZDongle-P donc aucune chance qu'on puisse associer ce souci de binding au problème de perturbations que l'on essaie de régler avec le nouveau firmware. Je verrai effectivement ce que ca donne au prochain réappairage. Si ca fonctionne pour vous (avec Z2M ?), c'est déjà une bonne nouvelle.

Et en parlant de bonne nouvelle, pour le coup, la dernière version du firmware a vraiment l'air magique. J'ai suivi d'un peu plus près les logs de Zigbee2mqtt et quand le device routeur ZBMini a un coup de moins bien (perte du réseau quelques minutes à cause d'un LQI un peu faible), le ZiPulses arrive à se reconnecter au coordinateur ZDongle-P, pour ensuite repasser par le relais ZBMini Sonoff quand il réapparait. 

Je continue de vous tenir au courant.

fairecasoimeme, 2024-09-23:
Attention, j'ai eu des cas de perturbations radios à quelques centimètres du dongle Zigbee. Par exemple, quand ce sont des perturbations WiFi ou quand le dongle est trop proche d'un disque SSD, les perturbations sont les mêmes loin ou proche du dongle.

En tous les cas, que le ZiPulses gère mieux ces problèmes sont une bonne chose... même si tous les paquets n'atteignent pas le coordinateur zigbee.

helepeut, 2024-09-24:
Bonjour, juste pour vous tenir au courant et ne pas vous faire croire que j'ai abandonné.
J'ai passé le firmware 7.1 hier (merci), le ZiPulses est réapparu lorsqu'il était à côté du routeur zigbee, mais les valeurs sont restées en l'état. Je l'ai repositionné sur le compeur de gaz, et rien ne se passe depuis 24h, il a disparu.

En forçant un peu ce matin, il a refait signe à Z2M (last seen) , mais toujours pas de changement de valeur (temperature, battery et impulsion).

Cependant je me pose une question : mon ZiPulses est connecté à 2 piles LR20 en mode REG. J'ai testé les piles elles sont bien à 1,5V, mais est-ce suffisant ? ou dois-je doubler pour passer à 6V? (car la led bleue est assez faible comparée à mes souvenirs des derniers mois, et quand je l'ai flashée)

fairecasoimeme, 2024-09-24:
> Cependant je me pose une question : mon ZiPulses est connecté à 2 piles LR20 en mode REG. J'ai testé les piles elles sont bien à 1,5V, mais est-ce suffisant ? ou dois-je doubler pour passer à 6V? [...]

En mode REG, il faut normalement être entre 5 et 12V. ça passe à partir de 3.5V environ mais ce n'est pas dans les clous. Du coup, les 2 piles LR20 ne suffisent pas.

helepeut, 2024-09-24:
Zut, confusion à la lecture de la doc. 
Il me semble avoir une v0.7 et la doc indique "⚠️ Avec la version 2 (v0.7 sur PCB), il est désormais possible d'utiliser ces supports avec les bornes classiques VIN et GND de l'appareil" dans la section alimentation externe (custom).

C'est donc ce que j'ai fait en m'inspirant des indications de la v0.5.
D'après vous, si je peux éviter de mettre 4 LR20, est-ce que : 
- je dois basculer le switch sur battery en restant avec les 2 LR20 sur les bornes vin et gnd ?
- je dois brancher le porte pile sur les plots 3v3 et GND de la carte, en laissant le switch sur REG ?

Merci

fairecasoimeme, 2024-09-24:
si vous voulez garder le support 3V, il faut utiliser les plots 3v3 et GND et peu importe le switch. Par le port de programmation l'alimentation ne passe par aucun régulateur mais aussi sans protection de tension

helepeut, 2024-09-24:
Merci, c'est ce que je viens de faire, il est revenu à la vie et envoie ses informations à nouveau ! (après un reset total).
Juste le moment de doute lors de l'application des diviseurs / multiplicateurs, mais je regarderai plus tard.
Pas encore d'information de batterie, voltage (peut-être normal ?) ni de température pour le moment.

Je vous tiens au courant, merci !

biwizit, 2024-09-24:
Normalement,il faut attendre 2h pour la remontée de ces infos température et tension. Et c'est ensuite mis à jour toutes les deux heures.
Je confirme que tout va toujours très bien de mon côté avec ce 4001-0007, (ça en est presque frustrant de ne pas avoir été écouté plus tôt car il y avait bel et bien une grosse amélioration possible côté firmware pour la communication Zigbee ! :-)

helepeut, 2024-09-24:
Ahah, 4 piles LR20, il va falloir que je monte une cabane à côté de mon compteur de Gaz car je ne vais plus savoir ou les mettre.
Pour le moment, j'ai basculé en direct sur la carte avec le 3v3 et le gnd. Au pire, j'ai un support de pile en plus qui traine pour compléter et repasser en Reg. Pour mon information, quelles piles rechargeables avez-vous pris ? 

Pour le moment toujours pas de remontée de température mais je vais m'y remettre demain car mon diviseur n'a pas été pris en compte, mais des valeurs remontent bien, et ce n'était plus le cas depuis longtemps (du fait probablement du défaut de tension comme indiqué par fairecasoimeme - étonnant que ca ait marché si longtemps même !)

biwizit, 2024-09-24:
Pour les LR20 rechargeables, j'ai tenté de trouver des "LSD" (Low Self Discharge) mais il n'y a pas de Eneloop dans ce format malheureusement. Je me suis donc rabattu sur les Amazon Baiscs ([ici](https://www.amazon.fr/Amazon-Basics-rechargeables-visuel-varier/dp/B07PHCWNH1)). Même avec de la décharge, en partant de 10 000mAh, on est bien à 10x l'équivalent d'une pile bouton (dans les 400mAH) sur plusieurs mois. 
[image]

biwizit, 2024-09-28:
Quel bonheur un ZiPulses qui fonctionne depuis une semaine déjà ! Preuve qu'il y avait quand même de grosses améliorations possibles côté firmware. Par contre, j'aimerais avoir confirmation par @helepeut (ou d'autres) qu'il réussit bien à voir des températures et tensions remonter. C'est le seul device Zigbee qui a du mal à s'appairer du premier coup de mon côté. Souvent des erreurs pendant le process d'inclusion. Je n'ai pas envie de toucher à quoi que ce soit pour le moment, surtout si je ne suis pas le seul, mais j'aimerais à terme avoir la tension, pour être prévenu que le changement de piles est pour bientôt. Merci.

helepeut, 2024-09-28:
Merci @biwizit pour les références de piles, pour le moment je vais rester sur mes 2 LR20 qui suffisent pour l'alimentation par batterie, car je n'ai pas non plus de chargeur de pile pour cette dimension : j'ai de quoi voir venir avant de rentabiliser les rechargeables, merci !

Pour ma part, j'ai refait l'apparairage plusieurs fois pour les informations de température et de batterie : mais toujours rien, c'est toujours votre cas aussi @biwizit ?
Par ailleurs, je n'ai jamais réussi à modifier les diviseurs et unité d'énergie. Si je n'appuie pas sur le bouton, j'ai bien le message d'erreur rouge qui m'indique que c'est raté. Mais quand j'appuie sur le bouton pour prendre en compte la modification, aucun message (ni rouge, ni vert, rien du tout). J'ai tout laissé par défaut pour virtualiser un sensor sur la base des valeurs par défaut du ZiPulses. 

Mais sinon, jusqu'ici tout va bien, seulement 2 jours mais pas de soucis pour le moment (d'habitude je le perdais entre 10 jours et 3 semaines)

biwizit, 2024-09-28:
@helepeut, oui je comprends pour l'investissement des LR20 rechargeables, surtout si vous n'avez pas le chargeur.

Pour la remontée d'information de la température et de la tension, j'avais vraiment tenté plein d'appairages après l'upload du 4001-0007, et j'ai même essayé de le faire avec des clés/réseaux différents, en vain. Ce n'était pas la première fois que l'inclusion pose souci à la première tentative, mais c'est la première fois que j'abdique suite à un grand nombre de tentatives.

Comme vous avez le même problème, j'imagine qu'il faudra un peu de temps pour que @fairecasoimeme soit convaincu que quelque chose a peut-être changé par rapport à avant... En tout cas, je vais vivre comme ça jusquà ce que mon ZiPulses rencontre un autre problème.

Mais clairement, ca faisait très très longtemps que je n'avais pas atteint une semaine sans aucune coupure. Et surtout, je vois bien que le ZiPulses s'adapte au réseau. Après avoir creusé un peu, il s'avère que le Sonoff ZBMini que j'utilise comme relais n'est pas connu pour sa grande capactité d'émission (surtout vers le coordinateur dans ma configuration) et le ZiPulses arrive à s'y reconnecter parfaitement bien quand le Sonoff revient après une petite heure d'absence (il offre au ZiPulses un LQI de 120 à 145 contre bien moins pour une liaison directe entre le ZiPulses et le coordinateur, liaison qui arrive à tenir malgré tout pendant cette période). Le switch de l'un (le relais) à l'autre (le coordinateur) se passe désormais à merveille.

Je continue à croiser les doigts car en efet, ça ne fait qu'une semaine.

Pigouchet, 2024-09-29:
> Normalement,il faut attendre 2h pour la remontée de ces infos température et tension. Et c'est ensuite mis à jour toutes les deux heures. Je confirme que tout va toujours très bien de mon côté avec ce 4001-0007, (ça en est presque frustrant de ne pas avoir été écouté plus tôt car il y avait bel et bien une grosse amélioration possible côté firmware pour la communication Zigbee ! :-) Je conseille vivement 4 piles LR20 (en rechargeable) pour être vraiment serein sur la tension et sur l'autonomie (qui reste malgré tout à confirmer).

Ou est-ce que l'on peut trouver cette version 4001-0007, pour ma part celà fait près d'un an que je cherche en vain des solutions pour avoir des remontées de mon compteur d'eau au-delà d'une journée... Je voudrais pouvoir essayer !!!

Merci de votre aide.

P.S.: La version de PCB est-elle visible directement sur la carte ?

biwizit, 2024-09-29:
> @biwizit @helepeut pouvez-vous retester avec le firmware suivant :
>
> * Augmentation du nombre de messages simultanés + ACK
> * Amélioration de la gestion du mode sommeil
> * Reconnexion automatique au [...]

@Pigouchet, le firmware à uploader est le fichier zip ci-dessus. Il vous faut un dongle pour le faire (https://github.com/fairecasoimeme/ZiPulses?tab=readme-ov-file#mise-%C3%A0-jour-du-firmware-non-ota)

fairecasoimeme, 2024-10-01:
Bonjour,

Normalement, cette version devrait corriger le problème des reports sur la température et la tension.

[ZiPulses_v7.2.zip](https://github.com/user-attachments/files/17202993/ZiPulses_v7.2.zip)

helepeut, 2024-10-01:
Bonjour @fairecasoimeme, merci pour la nouvelle version, je viens de la passer et c'est ok j'ai retrouvé la température et les informations de batterie, c'est parfait !

fairecasoimeme, 2024-10-03:
Merci pour le retour.
Je vais reprendre les tests sur la perte de comptage. J'ai un générateur d'impulsions et je vais tenter de mettre en place une plateforme de tests avec Z2M.

Sur le débat de l'autonomie, c'est assez compliqué de trouver le bon paramétrage. En effet, pour GAZPAR, si on remonte les infos de manière régulière, on maitrîse mieux l'autonomie mais lorsque l'été arrive et qu'il n'y a pas de consommation, on va envoyer toutes les X minutes des impulsions qui n'auront pas bougé. Peux -tu ouvrir une issue sur ce sujet ? (pour ne pas mélanger )

Ce que je peux faire c'est faire des remontées toutes les X minutes uniquement si le compteur a augmenté depuis la dernière fois. Le délai de remontée pourra être paramétrable mais nécessitera une MAJ des plugins.
Je vais réfléchir sur la bonne méthode. 

Concernant les déconnexions, si tout est OK, je vais publier une MAJ officielle avec la version 4000-0007

Et merci à tous pour votre participation au projet

biwizit, 2024-10-03:
Merci pour votre retour. Je pense que faire des remontées toutes les X minutes uniquement si le compteur a augmenté depuis la dernière fois, avec le délai de remontée paramétrable serait l'approche idéale. Je vais ouvrir un ticket en ce sens.

Pour les déconnections, ça aura été bien d'avoir le retour de @Pigouchet mais en ce qui me concerne, c'est absolument parfait depuis 15 jours (le 20 septembre). C'est un record de longévité donc c'est quand même bon signe. 

Merci !

biwizit, 2024-10-06:
Petite bizarrerie au bout de 17 jours : il  y a eu bégaiement (=envoi de la même valeur plusieurs fois) puis le device a été "annoncé" pour ensuite envoyer des valeurs plus petites que celles d'avant...

Etonnant...

Je vais en profiter pour mettre le 4002-0007 qui rajoute température et tension.

[image]

biwizit, 2024-10-06:
Firmware mis à jour ! Et effectivement, l'inclusion s'est super bien passée du premier coup avec ce 4002-0007, j'ai bien la tension et la température maintenant.

fairecasoimeme, 2024-10-07:
> Petite bizarrerie au bout de 17 jours : il y a eu bégaiement (=envoi de la même valeur plusieurs fois) puis le device a été "annoncé" pour ensuite envoyer des valeurs plus petites que celles d'avant...
> 
> Etonnant...
> 
> Je vais en profiter pour mettre le 4002-0007 qui rajoute température et tension.
> 
> ![Screenshot 2024-10-06 170731](https://private-user-images.githubusercontent.com/46035286/373957367-463edd69-a3d7-4bf6-9e1c-4acdeb874f4e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyODYwMDgsIm5iZiI6MTcyODI4NTcwOCwicGF0aCI6Ii80NjAzNTI4Ni8zNzM5NTczNjctNDYzZWRkNjktYTNkNy00YmY2LTllMWMtNGFjZGViODc0ZjRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDA3VDA3MjE0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhODg2NjY1NGM1YjNmZTkxZTM3YTg3OTlkODQ5ZTIyM2NjN2M3MmY1ZDYyYjE3YjAyODE0Y2NhMDdiZWMzZDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bp0OT7bBqs_xRNmAUb_cKLwbROWwYx7v_vwr8sMDdkk)

je pourrais avoir le log complet svp ?

biwizit, 2024-10-07:
Oui, j'essaie de faire ça le plus vite possible. De ce que je vois dans les logs envoyés à mon NAS par Zigbee2mqtt, j'ai de nouveau un bégaiement en cours (impulsion 225, idem avec la 220 quelques temps avant), ça ne me l'avait jamais fait avant hier... :-( A voir si j'ai bien une mise à jour plus tard mais j'ai peur que non.

[image]

biwizit, 2024-10-07:
Voici le log. Ca commence par une séquence normale qui tenait depuis 17 jours (j'ai donc raccourci le avant...). On voit le ZBMini sortir du réseau quelques minutes (ça lui arrive une fois tous les 2 ou 3 jours, pendant une petite demi-heure), il revient, on voit les impulsions reprendre leur cours normalement (le ZiPulses n'a pas eu le temps de switcher sur le coordinateur, il arrive à la faire de temps en temps) puis la bizarrerie apparait (compteur qui perd environ 1500 impulsions). On voit que je sors le device du réseau, que je le réinclue, on voit l'arrivée des températures et tensions avec le 4002-0070. Et on voit de nouveau un blocage à 225 impulsions depuis ce matin. Le problème est toujours en cours. Il y a bien quelque chose qui semble vivre sur le réseau mais malheureusement, ça reste bloqué. C'est presque plus frustrant qu'un device qui aurait complètement disparu...

A 19h30, toujours pareil (il faut rajouter 2 heures au last seen):
`{
    "battery": 100,
    "divisor": 1,
    "energy": 225,
    "last_seen": "2024-10-07T17:30:04.316Z",
    "linkquality": 0,
    "multiplier": 1,
    "temperature": 11.98,
    "unitOfMeasure": "kWh",
    "voltage": 3400
}`

[log.txt](https://github.com/user-attachments/files/17282783/log.txt)

Pigouchet, 2024-10-10:
> Merci pour votre retour. Je pense que faire des remontées toutes les X minutes uniquement si le compteur a augmenté depuis la dernière fois, avec le délai de remontée paramétrable serait l'approche [...]

Bonjour,

Pour ma part j'ai flashé dans la foulée en [ZiPulses_v7.1.zip](https://github.com/user-attachments/files/17062831/ZiPulses_v7.1.zip) mais le Zipulse s'est déconnecté une fois dans son emplacement (trappe dans le trottoir devant le garage) alors qu'il fonctionnait plutôt bien avant que je le remette en place.

J'ai attendu un peu car j'avais commandé le support de piles LR20 recommandé ainsi que les piles mentionnées par @biwizit. 

J'ai tenté l'installation aujourd'hui mais le module ne semble plus répondre. Plus de Led bleu quand je connecte les piles et idem lors d'une tentative de flash pour le mettre à jour en version ZiPulses_v7.0.

Je crois qu'il ne communiquera plus quel que soit son firmware 😇💀 !!!

fairecasoimeme, 2024-10-10:
> Voici le log. Ca commence par une séquence normale qui tenait depuis 17 jours (j'ai donc raccourci le avant...). On voit le ZBMini sortir du réseau quelques minutes (ça lui arrive une fois tous les 2 ou 3 jours, pendant une petite demi-heure), il revient, on voit les impulsions reprendre leur cours normalement (le ZiPulses n'a pas eu le temps de switcher sur le coordinateur, il arrive à le faire de temps en temps) puis la bizarrerie apparait (compteur qui perd environ 1500 impulsions). On voit que je sors le device du réseau, que je le réinclue, on voit l'arrivée des températures et tensions avec le 4002-0007. Et on voit de nouveau un blocage à 225 impulsions depuis ce matin. Le problème est toujours en cours. Il y a bien quelque chose qui semble vivre sur le réseau mais malheureusement, ça reste bloqué. C'est presque plus frustrant qu'un device qui aurait complètement disparu...
> 
> A 19h30, toujours pareil (il faut rajouter 2 heures au last seen): `{ "battery": 100, "divisor": 1, "energy": 225, "last_seen": "2024-10-07T17:30:04.316Z", "linkquality": 0, "multiplier": 1, "temperature": 11.98, "unitOfMeasure": "kWh", "voltage": 3400 }`
> 
> PS : linkquality de 0 signifie passage par le ZBMini, a priori ce device ne sait pas "transférer" l'info, limitation connue pour certain routeurs.
> 
> [log.txt](https://github.com/user-attachments/files/17282783/log.txt)

Bonjour,

J'ai regardé les logs et la première chose qui m'interpelle, ce sont les valeurs de LQI qui parfois reste à 0 pendant un long moment puis revient à 30-40 (ce qui reste très bas). Votre réseau Zigbee a réellement un problème. Pas sur que le ZBmini soit un bon routeur.

On voit bien les nombreuses réémissions. 3-4 fois la même trame (avec le même nombre de pulses) espacées de quelques millisecond. Ces trames sont reçu par le coordinateur mais l'acquittement ne parvient pas au ZiPulses (qui du coup réémet)

Concernant le nombre d'impulsions qui repart en arrière, je n'ai pas trouvé d'explications pour le moment. mais j'y travaille dessus. (je suis en train de monter une plateforme de tests)

fairecasoimeme, 2024-10-10:
> > Merci pour votre retour. Je pense que faire des remontées toutes les X minutes uniquement si le compteur a augmenté depuis la dernière fois, avec le délai de remontée paramétrable serait l'approche idéale. Je vais ouvrir un ticket en ce sens.
> > Pour les déconnections, ça aura été bien d'avoir le retour de @Pigouchet mais en ce qui me concerne, c'est absolument parfait depuis 15 jours (le 20 septembre). C'est un record de longévité donc c'est quand même bon signe.
> > Merci !
> 
> Bonjour,
> 
> Pour ma part j'ai flashé dans la foulée en [ZiPulses_v7.1.zip](https://github.com/user-attachments/files/17062831/ZiPulses_v7.1.zip) mais le Zipulse s'est déconnecté une fois dans son emplacement (trappe dans le trottoir devant le garage) alors qu'il fonctionnait plutôt bien avant que je le remette en place.
> 
> J'ai attendu un peu car j'avais commandé le support de piles LR20 recommandé ainsi que les piles mentionnées par @biwizit.
> 
> J'ai tenté l'installation aujourd'hui mais le module ne semble plus répondre. Plus de Led bleu quand je connecte les piles et idem lors d'une tentative de flash pour le mettre à jour en version ZiPulses_v7.0.
> 
> Je crois qu'il ne communiquera plus quel que soit son firmware 😇💀 !!!
> 
> Il aura fonctionné en un peu plus d'un an 10 jours sur pile et ~3 semaines d'affilés sur batterie externe.

Vous pouvez me le retourner pour analyse et échange

Pigouchet, 2024-10-10:
> Vous pouvez me le retourner pour analyse et échange

Je viens d'en commander un autre depuis votre boutique... ça marche je vous le renvoi dès que j'ai reçu le nouveau ! (j'avais un PCB 0.5 est-ce que le nouveau sera d'une génération plus récente ?)

Merci.

biwizit, 2024-10-10:
> Bonjour,
>
> J'ai regardé les logs et la première chose qui m'interpelle, ce sont les valeurs de LQI qui parfois reste à 0 pendant un long moment puis revient à 30-40 (ce qui reste très bas). Votre [...]

En réalité, quand vous voyez des valeurs non nulles, c'est que la connection se fait en direct entre le device et la clé Sonoff ZDongle-P, et les valeurs sont faibles en effet, d'où l'intérêt du routeur à l'extérieur.

Les liaisons device<->routeur et routeur<->clé sont entre 100 et 140, le routeur fait donc un bon relais de ce point de vue-là. Et la valeur linkquality du device est à 0 quand la connection est faite à travers le routeur. Je ne sais pas qui est responsable de ce problème de forward d'info (ZBMini ou Z2M) mais la carte graphique du réseau dans  z2m montre que les vraies valeurs entre eux sont dans un bonne fourchette.

En tout cas, encore maintenant, je continue de recevoir des trames "ZiPulses", désespéremment bloquées à 255. Je vais d'abord essayer de débrancher et rebrancher le relais pour voir ce que ça fait. Et si ça ne donne rien, j'essaierai de trouver du courage pour retenter un appairage. Au prochain souci, "je m'amuserai" à remettre un thermometre Aqara à 5€ sans antenne dans le coffre du compteur pour voir combien de temps il tient avec ce combo clé-routeur Sonoff...

fairecasoimeme, 2024-10-16:
Bonjour à tous,

Voici la procédure en cours sur la plateforme de test:
J'utilise un générateur d'impulsions : 1 impulsion de 100 ms toutes les secondes
J'ai branché le Zipulses sur pile

[image]

https://github.com/user-attachments/assets/a497d709-7f28-407d-b45a-e4e89fb4d011

Actuellement je suis à 30 heures de fonctionnements : soit 108 000 secondes (soit 108 000 impulsions)
Sur Z2M le module avait déjà 925 impulsions (dû à des tests précédents)

J'ai donc récupéré sur Z2M à 16h23:00 précise, 108925 impulsions (sur Z2M)

[image]
Bon j'ai fait la capture juste avant 16h23:00

La prochaine étape est de couper Z2M puis relancer puis enlever le coordinateur puis relancer.

biwizit, 2024-10-16:
Statistiques intéressantes.

Sur les quelques jours durant lesquels il a été possible de comparer ZiPulses et GRDF, les résultats sont assez bien alignés. Pour 6 jours, j'ai 29.58m3 pour le ZiPulses, et 29.82m3 d'après le fournisseur de gaz. Ca fait moins de 1% d'erreur ((29.82-29.58)/29.82=0.008).
[image]

fairecasoimeme, 2024-10-17:
Bonjour,

J'ai trouvé un problème de comptage d'impulsions lors des déconnexions. S'il y a beaucoup d'impulsions rapprochés il y a des loupés. Je travaille dessus

[62 more replies]