fairecasoimeme/ZiPulses #39: ZiPulse passe en "bleu clignotant" ...

alexicna92, 2024-11-17:
Bonjour 

J'ai 2 Lixee ZiPulse proche assez proches l'un de l'autre. 
Le premier mesure ma conso d'eau , il est en firmware 4, je n'ai pas de soucis avec lui. En remontant les "last seen" via Z2M , s'il n'est pas sollicité entre temps, il fait bien "coucou" toutes les 2 heures.
Le second mesure ma conso de gaz (relié au compteur Gazpar donc) , il est en firmware 6 . Lorsqu'il est bien connecté , les remontées d'impulsions sont OK. Par contre, il arrive fréquemment qu'il ne donne plus signe de vie (je m'en rends compte grâce à une alarme qui m'indique s'il n'a pas donné de signe de vie depuis plus de 2h). Dans ce cas , il clignote bleu , et un appui long ou court n'a aucun effet. La seule façon de le reconnecter est de retirer et remettre la pile . Les impulsions du gazpar durant cette déconnexion semblent perdues. 

Une mise à jour du firmware peut elle régler ce soucis  ?
Dois je passer en 7.0 ? 7.1 ? 8.0 ? ou en 4 (comme l'autre qui fonctionne ??).

Merci d'avance pour le retour

alexicna92, 2024-11-21:
Bonjour, le soucis semble réglé avec le passage du firmware de la version 6 à la version 7. 
Je n'ai pas eu de déconnexion depuis plus de 36h , à voir dans le temps. 

J'en profite pour soulever une autre question : Il m'avait semblé qu'en l'absence de toute sollicitation , le ZiPulse envoyait un message "Last Seen" toutes les 2 heures. Est bien le cas ? 
- J'ai l'impression que oui pour celui qui est sur le gazpar en firmware 7 (bien qu'en ce moment il est rarement inactif 2 heures ...)
- Pour celui en firmware 4 sur le compteur d'eau , il a été plus de 6h sans sonner signe de vie (durant la nuit) mais s'est bien réveillé lorsqu'il a été sollicité.

Pourriez vous me refaire un point sur la fréquence des messages "Last Seen" ? Est ce différent selon le firmware ?
J'en ai besoin pour monitorer leur état et savoir au plus vite s'il y a un dysfonctionnement.

Merci d'avance

neoback45, 2025-01-02:
> Bonjour
>
> J'ai 2 Lixee ZiPulse proche assez proches l'un de l'autre. Le premier mesure ma conso d'eau , il est en firmware 4, je n'ai pas de soucis avec lui. En remontant les "last seen" via Z2M , [...]

Bonjour,
J'ai le même soucis que vous sur mon Zipulses qui relève ma conso de gaz (Gazpar) et je suis en Version 8.0! Avez vous toujours des "déconnexion"?
Merci de votre retour
Pour le zipulses qui releve ma conso d'eau je n'ai pas ce soucis.