fairecasoimeme/ZiPulses #40: Inclusion impossible avec Zigbee2Mqtt sur Jeedom

kouros38, 2024-12-03:
Bonjour,

 Depuis hier, j'essaye d'inclure le compteur d'inclusion Lixee ZiPulses sur Jeedom avec le plugin ZigbeeLinker.
Il me remonte sur zigbee2Mqtt : Non pris en charge avec erreur d'inclusion.
Après recherche, cet équipement est bien pris en compte par zigbee2mqtt. J'ai bien mis a jour zigbee2mqtt avec la version 1.42.0.
et pas d'inclusion.
Mon ZiPulses est en version V1 avec Pcb 0.5. la dernière version est la V8    Est cela mon problème ?
Faut il mettre a jour le firmware, chose que j'ai jamais faite.

fairecasoimeme, 2024-12-05:
Bonjour,
Lors de l'appairage, il faut appuyer sur le bouton "link" toutes les secondes pour le maintenir éveillé lorsque Z2M l'interroge.

kouros38, 2024-12-05:
Bonjour,
J'ai bien appuyé sur le bouton Link des multitudes de fois et lorsque il s'appareille, Z2M récupère bien son identifiant mais me signale qu'il n'est pas pris en charge, bien que Z2M le référence depuis la version 1.30.3.
[image]

fairecasoimeme, 2024-12-06:
Désolé alors je ne sais pas. Essayez peut-être de mettre à jour le firmware. Mais normalement s'il est marqué non pris en charge, c'est que ya un problème avec Z2M.