fairecasoimeme/ZiPulses #41: Mise à jour du ZiPulses depuis macOS ?

MilesTEG1, 2024-12-06:
Bonjour @fairecasoimeme 
Est-il possible de faire la MAJ du firmware depuis macOS ?
Avez-vous un lien vers une version macOS du binaire qui permet de flasher ?

Merci d'avance.

fairecasoimeme, 2024-12-06:
Bonjour,

Désolé mais l'application de flashage n'est compilé que pour Windows. J'ai les sources du constructeur mais je ne peux pas les fournir.

closed 2024-12-06 by fairecasoimeme

MilesTEG1, 2024-12-06:
Ha mince...
Et vous n'avez pas de mac, je suppose, pour compiler une version macOS ? 😭

MilesTEG1, 2024-12-06:
Est-ce que ceci permettrait de flasher depuis macOS ?
https://github.com/dfu-programmer/dfu-programmer

fairecasoimeme, 2024-12-06:
malheureusement non :( pas de mac.
Non dfu-programmer ne permet pas de flasher car le chip est un NXP JN5189

MilesTEG1, 2024-12-06:
Bon je vais devoir me résigner à utiliser un pc windows 😓