fairecasoimeme/ZiPulses #45: Pas de remontée d'index automatique en V8

boulgom, 2025-01-13:
Bonjour,

Ayant des soucis en V6 et V7 j'ai tenté un upgrade complet de ma config.

J'ai donc la désormais la config suivante:
- Raspberry pi 
- Dongle sonoff ZBDongle-E (CH9102F) flashé avec ember
- Zigbee2mqtt en V2.0.0
- ZiPulses acheté en aout dernier sur lixee flashé en V8 alimenté en 5V permanent, il est à côté de 4 routeurs (1 ZLinky_TIC , et 3 telerupteurs Legrand) et se connecte à l'un des 4. J'ai 12 autres équipements zigbee qui fonctionnent tous très bien.

Le comportement du ZiPulses est le suivant (branché à un compteur d'eau)
- On voit bien les impulsions (led blue s'allume et s'éteint) mais pas de mise à jour de la valeur
- On voit bien les connexions toutes les minutes dans les logs
- Si on appuie sur le bouton link, là je récupère une nouvelle valeur sur le broker mqtt

Ce comportement est systématique et reproduisible

Voici les logs zigbee2mqtt associé que j'ai nettoyé pour ne pas voir les autres appareils
- 19:30 j'ai fait couler l'eau pour avoir des impulsions, rien ne remonte, 
- on voit bien les connexions toutes les minutes (je n'ai mis que les connexions de 19:31 et 19:32 dans la pièce jointe)
- 19:42 appuie sur le bouton link et là je récupère le nouvel index

[log2.log](https://github.com/user-attachments/files/18401862/log2.log)

Merci d'avance

fairecasoimeme, 2025-01-15:
je regarde les logs et reviens vers vous ASAP.

fairecasoimeme, 2025-01-16:
Bonjour,

je viens de refaire les tests de mon côté et la v8 permet de remonter les infos d'impulsions toutes les minutes.
Si ce n'est pas le cas, il y a peut-être eu un problème lors de l'appairage sur un bind ou autre.

Avez-vous essayé de relancer la config (bouton orange) et en appuyant toutes les secondes sur le bouton "link" du zipulses jusqu'à ce qu'un message de Z2M renvoie config ok ?

boulgom, 2025-01-18:
Bonjour,

J'ai essayé de sortir l'appareil du réseau et le remettre et j'ai déjà essayé la config. 
J'ai réessayé la config hier mais toujours pareil.

Il y a une méthode p1our faire un reset plus poussé ?

fairecasoimeme, 2025-01-20:
Vous avez bien noté que sur la V8, les remontées d'impulsions ne sont pas instantannées ? il faut attendre 1 minute s'il y a eu des impulsions.

boulgom, 2025-01-20:
Oui oui, je n'ai aucune remontée (pendant plusieurs heures) sauf si j'appuie manuellement sur le bouton link

fairecasoimeme, 2025-01-21:
Désolé mais du coup, je ne vois pas ce qui peut engendrer ce phénomène. Si le bind est bien fait, que la led s'allume lors d'une impulsion et que la trame part bien lors d'un appui, je ne vois pas bien ce qui ne va pas. J'ai essayé de reproduire mais impossible de mon côté.
Vous pouvez passer par le mail : contact[at]lixee.fr pour me faire un renvoi de l'appareil pour essayer de reproduire. Mais j'avoue que je ne pense pas que ce soit matériel ou firmware.

boulgom, 2025-01-22:
Ok, donc plutôt un soucis d'intégration avec zigbee2mqtt. Bon je vais essayer de faire un reset complet du réseau. Je vous dirais ce qu'il en est.

fairecasoimeme, 2025-01-22:
Pour être honnête, je ne pense pas à un problème d'intégration.
Pouvez-vous essayer de refaire une mise à jour en effaçant la mémoire flash (option : -e FLASH)
retélécharger la version : https://github.com/fairecasoimeme/ZiPulses/releases/download/v8.0/ZiPulses_v8.0.bin

neoback45, 2025-02-25:
> Ok, donc plutôt un soucis d'intégration avec zigbee2mqtt. Bon je vais essayer de faire un reset complet du réseau. Je vous dirais ce qu'il en est.

Bonjour,
J'ai exactement le même soucis que vous. J'ai même tenter de le re-flasher mais idem...J'ai en revanche un zipulses sur le compteur gaz et cela fonctionne très bien.
Lorsque je le sort du réseau ou que je le reconfigure cela fonctionne quelque heure puis plus rien, obliger d'appuyer sur le bouton pour récupérer la valeur. J'ai meme eu encore aujourd'hui, le zipulses qui ne réagit plus, la led s'allume plus comme si il était planté. Le fait d'enlever et remttre la pile le reveil...
Je vois également la batterie qui oscille entre 50% et 100% avec une pile neuve.
Je pense plutôt a un soucis de V8 raccorder au compteur d'eau...car au gazpar RAS, et les valeurs sont plutot coérante.

fairecasoimeme, 2025-02-26:
Désolé mais je ne pense pas que ce soit le même soucis ....

boulgom, 2025-02-26:
alors moi j'ai reflashé le dongle zigbee et le zipulse après l'avoir sorti du réseau et quand je l'ai remis j'ai cliqué assez frénétiquement sur le bouton link et pour une fois j'ai eu zéro message dans zigbee2mqtt.

ça fait 2 semaines et ça à l'air stable.

Je dirais que c'était un soucis de bind. C'est pas si simple de l'intégrer et de le configurer j'ai l'impression qu'il s'endort vite

closed 2025-02-26 by boulgom

neoback45, 2025-02-27:
> Désolé mais je ne pense pas que ce soit le même soucis ....

Ca a l'air pourtant similaire...

neoback45, 2025-02-27:
> Non dans son cas, c'est un problème de bind.

J'insiste mais j'ai observé le même phénomène que le message du début....