fairecasoimeme/ZiPulses #46: [ZHA] Probleme pour changer l'unité

zoic21, 2025-01-29:
Bonjour,

Depuis la dernière version de HA il y a un soucis pour configurer l'unité ou seul la valeur 0 et 1 est acceptée mais plus les autres : 

[image]

Dans les logs on trouve ca : 

```
Enregistreur: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
intégration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [problèmes](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
S'est produit pour la première fois: 16:47:34 (9 occurrences)
Dernier enregistrement: 18:18:34

[140243285369920] Unexpected exception
[140242094374000] Unexpected exception
[140242015180624] Unexpected exception
[140242029996368] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1109, in _async_admin_handler
    await result
  File "/usr/src/homeassistant/homeassistant/components/zha/websocket_api.py", line 1333, in set_zigbee_cluster_attributes
    response = await zha_device.write_zigbee_attribute(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zha/zigbee/device.py", line 878, in write_zigbee_attribute
    value = convert_zcl_value(value, attr_def.type)
  File "/usr/local/lib/python3.13/site-packages/zha/application/helpers.py", line 156, in convert_zcl_value
    field_type[value.replace(" ", "_").split(".", 1)[-1]]
    ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/enum.py", line 789, in __getitem__
    return cls._member_map_[name]
           ~~~~~~~~~~~~~~~~^^^^^^
KeyError: '7'
```

J'ai l'impression que c'est plus une valeur autorisée.

En vous remerciant d'avance,

fairecasoimeme, 2025-01-30:
Bonjour,
je ne sais pas du tout. Peut-être voir avec home assistant ou les dev de ZHA. Mais je ne vois pas pourquoi cette valeur ne serait pas plus autorisée.
De mon côté, je vais faire le test et je reviens vers toi si j'ai la même chose.

zoic21, 2025-01-30:
J'ai vu avec ZHA ils considèrent que cet attribut et en lecture seule donc pas autorisé a l'écriture. Je leur ait expliqué pourquoi je voulais le faire a voir si ils corrigent.

fairecasoimeme, 2025-01-30:
En effet, normalement dans la norme, l'attribut est en Read Only mais j'avais modifié le type pour pouvoir rendre l'unité de mesure paramétrable.
je vais réfléchir à comment faire autrement.

julien969a, 2025-05-03:
L'argument n'est pas en Read Only, car il est tout à fait possible de le changer de 0 (kWh, la valeur par défaut) à 1 (m3) mais les autres valeur échouent. 
Dans la documentation il est indiqué en RW tout comme les multiplier et le diviser

0x0300 | unitOfMeasure | **RW** | Enum8 | 0 (kWh) par defaut | Unité de mesure pour les impulsions. Pour choisir la bonne unité de measure cf Tableau de mesures
0x0301 | Multiplier | RW | Uint24 | 1 par defaut | Coefficient multiplicateur pour les impulsions
0x0302 | Divisor | RW | Uint24 | 1 par defaut | Coefficient diviseur pour les impulsions

J'ai écrit au support pour leur demander des précisions.

julien969a, 2025-05-03:
L'argument n'est pas en Read Only, car il est tout à fait possible de le changer de 0 (kWh, la valeur par défaut) à 1 (m3) mais les autres valeur échouent.
Dans la documentation il est indiqué en RW tout comme les multiplier et le diviser

ça serait génial de pouvoir choisir les unités comme dans la doc
0 | kWh
1 | m3
2 | ft3
3 | ccf
4 | US gl
5 | IMP gl
6 | BTUs
7 | L (litre)
8 | kPA (jauge)
9 | kPA (absolu)
10 | kPA (absolu)
11 | sans unité
12 | MJ
13 | kVar