fairecasoimeme/ZiPulses #47: Zipulse Compteur bloqué

net1957, 2025-02-25:
Suite aux soucis initiaux qui ont été réglés par la mise à jour en version 14, le fonctionnement de ce module est devenu plus fiable.

Un seul soucis est récurent:

. Le boîtier communique toujours avec Zigbee2MQTT, mais ni l’énergie ni la température n’évolue.
. en appuyant sur le bouton, la led bleu s'allume et le last-seen de Zigbee2MQTT se met a jour, mais pas pas les autres valeurs

Pour récupérer un fonctionnement "normal" la procédure suivant fonctionne:
. enlever la pile, la remettre en place, appuyer sur le bouton et le compteur est transmis à Zigbee2MQTT avec la bonne valeur

Exemple de ce jour:

vers midi le compteur se bloque a une valeur de 25267.
vers 18H, j'effectue la manip décrite et la valeur récupérée est de 25299

Le valeurs sont cohérentes avec le relevé manuel du compteur.

Ce ce produit de temps en temps avec une périodicité très variable entre quelques jours a quelques semaines.

Une idée ?

Zigbee2MQTT:  2.11
ZiPulse: ZiPulses_v8.0.bin

neoback45, 2025-02-26:
> Suite aux soucis initiaux qui ont été réglés par la mise à jour en version 7, le fonctionnement de ce module est devenu plus fiable.
> 
> Un seul soucis est récurent:
> 
> . Le boîtier communique toujours avec Zigbee2MQTT, mais ni l’énergie ni la température n’évolue. . en appuyant sur le bouton, la led bleu s'allume et le last-seen de Zigbee2MQTT se met a jour, mais pas pas les autres valeurs
> 
> Pour récupérer un fonctionnement "normal" la procédure suivant fonctionne: . enlever la pile, la remettre en place, appuyer sur le bouton et le compteur est transmis à Zigbee2MQTT avec la bonne valeur
> 
> Exemple de ce jour:
> 
> vers midi le compteur se bloque a une valeur de 25267. vers 18H, j'effectue la manip décrite et la valeur récupérée est de 25299
> 
> Le valeurs sont cohérentes avec le relevé manuel du compteur. Cela indique que le Zipulses compte toujours les impulsions, mais la valeur transmise ne bouge plus...
> 
> Ceci se produit de temps en temps avec une périodicité très variable entre quelques jours a quelques semaines.
> 
> Une idée ?
> 
> Zigbee2MQTT: 2.11 ZiPulse: ZiPulses_v8.0.bin

Bonjour, J'ai exactement le meme soucis! Meme version et Z2M aussi.
J'attend un retour du fabricant....

fairecasoimeme, 2025-02-26:
Ok je vais refaire les tests en ce sens avec les mêmes conditions.
Je vous tiens au courant.

neoback45, 2025-02-27:
> Ok je vais refaire les tests en ce sens avec les mêmes conditions. Je vous tiens au courant.

Merci ;)

gblanchet, 2025-03-02:
mon zipulses est bloqué depuis 1 jour.
Je vais aller enlever puis remettre la pile.
Version du firmware
    4000-0008

neoback45, 2025-03-02:
> mon zipulses est bloqué depuis 1 jour.
> Je vais aller enlever puis remettre la pile.
> Version du firmware
>     4000-0008

J'ai ça depuis plusieurs semaine. Ça repart et ça décroche

gblanchet, 2025-03-02:
C'est reparti. Je suis avec Z2M. Dans MQTT energy qui était bloqué à 1133 est passé d'un coup à 2506 !!
En la manipulant j'ai un peu chauffé la pile c'est peut-être car dans la rue où est mon gazpar ça caille !!

neoback45, 2025-03-03:
> C'est reparti. Je suis avec Z2M. Dans MQTT energy qui était bloqué à 1133 est passé d'un coup à 2506 !! En la manipulant j'ai un peu chauffé la pile c'est peut-être ça, car dans la rue où est mon gazpar ça caille !!

Moi idem depuis hier. Le zipulses ne réagit même plus lorsque j’appuie sur le bouton. J’enlève la pile et je la remet et ça repart....pour 1 journée!

fairecasoimeme, 2025-03-03:
Désolé mais cela ressemble beaucoup à un problème de pile... 
Les piles plates au lithium ont ce comportement ... même si au voltmètre la tension semble OK, si la pile est morte, dès que l'on va tirer dessus de l'énergie, la tension va s'écrouler.
Ne pouvez-vous pas, pour les tests utiliser une alimentation externe ou une batterie externe. à brancher sur le bornier avec une tension entre 5v et 12v

De mon côté je teste mais pas encore reproduit le phénomène.

gblanchet, 2025-03-03:
C'est bien une histoire de température. Quand il gèle la batterie ne doit plus fournir de courant pour que le bouton soit actif. Mais si on sort la pile et qu'on la réchauffe un peu du coup ça repart. Heureusement le zipulses garde en mémoire les impulsions manquées.
ça fait dans ma partie energie de HA une grosse consommation de gaz d'un coup.
@fairecasoimeme Est-ce qu'une pile 9v qu'on pourrait emmailloter ferait l'affaire ?

fairecasoimeme, 2025-03-03:
Oui un pile 9V alcaline fait l'affaire.

neoback45, 2025-03-03:
> Oui un pile 9V alcaline fait l'affaire.

J'ai effectivement essayer avec une alimentation externe mais même soucis :( 
J'avoue ne pas comprendre...

fairecasoimeme, 2025-03-03:
Et en appuyant sur le bouton "reset", ça repart ?

neoback45, 2025-03-03:
> Et en appuyant sur le bouton "reset", ça repart ?

Oui mais pas pour longtemps.
Mon zipulses est dans le garage ou la température descend pas plus bas que 10 degrés du coup je pense pas que ce soit le froid

fairecasoimeme, 2025-03-04:
ok je continue les tests

skydarc, 2025-03-09:
Même problème ici... Mais le reset ne fait pas repartir le compteur. Obligé de supprimer le pulse et de le reacoupler a z2m... Au début j'avais z2m couplé avec zha... Je pensais que l'un parasité l'autre... J'ai viré zha... Même problème. Passerelle zigbee Slzb06. Zipulse sur transformateur (la pile se vidait trop vite). V firmware 4000-0006

gblanchet, 2025-03-09:
[image]
si ça peut aider

fairecasoimeme, 2025-03-10:
> Même problème ici... Mais le reset ne fait pas repartir le compteur. Obligé de supprimer le pulse et de le reacoupler a z2m... Au début j'avais z2m couplé avec zha... Je pensais que l'un parasité [...]

Bonjour,

Non ce n'est pas le même problème. il faut mettre à jour votre zipulses en v8

Fred

skydarc, 2025-03-12:
En effet... Firmware mit à jour en v8. Je vous tiens au courant si. Pour info z2m m'annonce que le firmware est tjr en V6. (Pendant un temps j'avais eu le même problème sur le zlinky d'ailleurs... Mis en v14 mais toujours affiché en v12 ou v.qqch.)

gblanchet, 2025-03-14:
Bonjour,
Je viens de mettre une pile 9V en alimentation extérieure, après avoir enlevé la pile CR2450 et basculé le petit switch.
On va voir si ça tient le coup dans la durée.
A suivre ...

gblanchet, 2025-03-21:
Depuis que j'ai mis la pile 9V, il y a plus d'une semaine il n'y a plus de blocage.
Il semble bien que ce soit un problème de puissance de la pile CR2450.
Qu'en dit @skydarc ?
Gérard

jessycler, 2025-04-24:
Bonjour j'ai le même souci depuis la V8.
Le zipulse arrête de transmettre les infos à Z2M sans que je ne comprenne pourquoi. Pour le relancer, je dois retirer la pile, 
ça refonctionne pendant 3j ou 1 semaine puis cela recommence...
Ce qui est étrange c'est que je le vois bien apparaître dans la "carte" de Z2M, relié à un routeur et avec un LQI correct. Par contre les valeurs température et Energy n'évoluent plus...

gblanchet, 2025-04-24:
Bonjour,

J'attends un autre Zipulses commandé directement chez Lixee. Je vais lui brancher un connecteur 9V (avec les 2 boutons pressions) mais sans la pile. De même du côté câble vers le compteur Gazpar. Tout refermer des deux côtés. Mettre tout ça dans le boitier gaz dans la rue. Par mon smartphone connecté à home assistant zigbee2MQTT je vais "autoriser l'appairage (tout)" et remettre la pile. Pas loin du compteur gaz il y a 2 prises extérieures qui font routeur zigbee. Les autres zipulses (2) que j'avais utilisé précédemment ont été désinstallés à la fois de Z2M et de MQTT.
Je vous tiens au courant dès que j'ai fait la manip. En partant propre et en faisant l'appairage in situ j'espère qu'il n'y aura plus de déconnexion sauf lorsque la pile sera morte.

gblanchet, 2025-04-25:
Bonjour,
Je l'ai reçu ce matin. Bien appairé. Bon la chaudière ne tourne pas. J'ai consommé du gaz sur ma cuisinière.
Un truc intéressant : la tension en mVolt.

[image]
Gérard