fairecasoimeme/ZiPulses #48: Probléme version mise a jour zipulse zigbee2mqtt

geogeozib57, 2025-02-27:
Bonjour, je rencontre un souci avec mon zipulse, je l'ai mis à jour en v7 en suivant les instructions du github, tout se passe bien pour la MAJ. Par contre, quand je le connecte à mon zigbee2mqtt n°1, la version affichée dans zigbee2mqtt est toujours la v6. Ce qui est bizarre, c'est qu'en connectant le zipulse à mon deuxième réseau zigbee2mqtt, il est bien en v7

Pour info : zigbee2mqtt n°1 en v1.40.1 / Type de coordinateur :ConBee2 en v0x26780700

[image]

                 zigbee2mqtt n°2 en v1.39.0 / Type de coordinateur:EmberZNet en v7.4.3 [GA]

[image]

fairecasoimeme, 2025-02-27:
`les versions ne sont consultés et donc envoyée uniquement lors de l'appairage.
C'est pour cette raison que dans le premier cas la version n'a pas été mise à jour

closed 2025-02-27 by fairecasoimeme

geogeozib57, 2025-02-27:
j'ai supprimé le zipulse de zigbe2mqtt et refait un appareillage

geogeozib57, 2025-02-27:
Mais j'ai le même resultat

fairecasoimeme, 2025-02-27:
il faut forcer la suppresion il me semble

geogeozib57, 2025-02-27:
Je l'ai fait aussi 

[image]

fairecasoimeme, 2025-02-27:
Lors de l'appairage, il faut appuyer plusieurs fois sur le bouton d'appairage car le ZiPulses se rendort assez vite et peut louper les requetes