fairecasoimeme/ZiPulses #49: V8 re join echec

devildant, 2025-03-06:
Bonjour, 
J'ai mi a jour mon module en v8 il y a quelque semaine pensant que ca résoudrait mon probleme de sortie du reseau que j'ai.

Le probleme c'est reproduit de nouveau today, le module n'arrive plus a communiquer avec mon reseau zigbee et passe hors ligne sous z2m.

Le click sur le bouton n'aide pas, seul solution que j'ai est d'enlever la pile puis la remettre, a se moment le module re join mon reseau sans souci.

Mais c'est relativement pénible. :(

devildant, 2025-03-06:
Pour info le module est a quelque mettre de mon appartement, la pile est une cr123a avec 60% de batterie
Lqi 83

Mon maillage est solide avec 43 routeur

devildant, 2025-03-06:
Du peu que j'ai pu observer, lorsque cela se produit, et que j'enlève et remet la pile, le zipulse ne se reconnecte pas au même routeur, cela peut arrivé que selon les perturbations il soit nécessaire de passer par un autre routeur disponible, la plus part des devices a pile le font assez facilement, mais pas le zipulse.

Il est capable de le faire au démarrage électrique mais pas par la suite, peut être une piste de ce coté la

fairecasoimeme, 2025-03-07:
Merci pour les pistes. Le produit est actuellement en test à ce sujet.

besbar, 2025-04-16:
Bonjour, 
Je viens d'acquérir 2 modules Zigbee.

- 1 lixee TIC pour le linky qui marche parfaitement, 
-  1 lixee Zipulse (version 4000-0008)) qui malheureusement se déconnecte aussi très régulièrement. La connexion du Zipulse tient généralement quelques heures avant de disparaître. Ce dernier est situé dans mon entrée d'appartement, juste derrière ma porte d'entrée.

Pour info la pile est CR2430, neuve et le signal aux alentours de 100-120 LQI. En effet le clic sur le bouton ne "ranime" pas le capteur.
Je confirme qu'en retirant et replaçant la pile, le capteur s'appaire à nouveau, sans que la valeur de l'energie soit réinitialisée (mais celle-ci connaît un saut plus ou moins important).

fairecasoimeme, 2025-04-17:
Oui déjà que certaines CR2450 peuvent poser problèmes. Les piles boutons Lithium ne peuvent pas délivrer beaucoup d'énergie
la CR2430 peut délivrer de mémoire 10mA et la CR2450 15mA, les émissions radios consomment autour des 10mA.
Il est possible que ce soit trop juste et que la chute de tension engendre ce phénomène.

devildant, 2025-04-17:
Je rajoute quelques pierre a l'edifice.

Dans mon cas sur pile 3v cr123a, le zipulse est a l'exterieur de chez moi, il passe donc par un routeur (ampoule, prise ou inter connecté)
Je peux avoir des souci plusieurs fois par semaine ou rien sur plusieurs mois, par exemple depuis que j'ai creer ce poste je n'ai pas eu de déconnexion mais il semble toujours connecté au même routeur (actuellement une prise nous a1z), cela semble moins stable avec les ampoule (ikea)
Le problème que j'ai constaté c'est une perte de connexion et le faites d'enlever et remettre la pile permet au zipulse de switch de routeur.

Concernant l'utilisation avec une alimentation secteur 5v pour mon cas je n'ai pas encore rencontrer le souci qu'il soit connecté a une ampoule ou un autre coordinateur, cela fait a peu près 1 an que j'ai 2 zipulse pour l'eau avec cette alimentation 5v et pas de souci pour le moment.

Outre l'alimentation 5v ces 2 zipulse sont a l'interieur donc au plus proche de mon reseau, c'est celui a pile le plus éloigné qui pause le plus de souci (et loi de murphy oblige c'est le plus pénible d'acces)

besbar, 2025-04-17:
@fairecasoimeme => d'après nos deux témoignages, il semblerait que le Zipulses ne pose pas de problème de connexion, lorsqu'il est alimenté en externe.

Que peut-on faire pour assister à la résolution du problème?
Est-ce que cela est utile si je teste avec une pile CR2450?

@devildant, je suis aussi intéressé par connecter un 2ème Zipulses à mon compteur d'eau.
Est-ce que comme pour le Gaspar, il faut un câble spécifique, pour se connecter au compteur d'eau. Je n'ai pas encore regardé du tout ce point. Merci par avance.

devildant, 2025-04-17:
Je reponds partiellement, a mon avis tu aura un mieux avec une cr2450 mais malgré tout, j'ai le probleme avec une cr123a...

Pour le compteur d'eau moi j'ai fait installer des compteur a impulson a l'interieur de chez moi, que j'ai ensuite connecté au zipulse (impossible sur le compteur d'eau principal)

Voila la ref du compteur (tout est fournis) : GIOANOLA COMPTEUR D'EAU CHAUDE À IMPULSION - 3/4" 1 IMP/LITRE

Mincka, 2025-04-19:
> [@fairecasoimeme](https://github.com/fairecasoimeme) => d'après nos deux témoignages, il semblerait que le Zipulses ne pose pas de problème de connexion, lorsqu'il est alimenté en externe.

Malheureusement, je ne peux pas en dire autant. J'ai l'alimentation externe et tout s'est bien passé pendant l'hiver.
J'ai dû le débrancher et maintenant, j'ai vraiment de grosses galères pour qu'il reste connecté.
Il se connecte, j'arrive à faire l'interview et après quand je le laisse, il ne se reconnecte plus. Appuyer sur le bouton ne fait rien.

elafargue, 2025-07-07:
Même problème ici avec une CR2450 Duracell neuve. Y a-t-il plus de nouvelles sur ce sujet? Le symptôme a l'air d'être identique aux précédents, le Zipulses n'envoie plus de données, un appui sur le bouton fait clignoter la LED en bleu, mais rien n'est reçu sur le réseau Zigbee. Retirer la pile ou simplement commuter brièvement le switch de "Pile" à "Reg" fait revenir le Zipulses sur le réseau instantanément.

Dans mon cas, il communique avec le coordinateur directement. Est-qu'il y a une meilleure chance de garder la connexion avec une pile 9V, par exemple?

elafargue, 2025-07-07:
@fairecasoimeme : des idées? Y a-t-il quoique ce soit que je peux faire pour aider? Est-ce que le ZiPulses comprend que le rejoin ne fonctionne pas quand le problème se produit? N'y a-t-il pas possibilité d'avoir un compteur d'essais au bout duquel le ZiPulses fait un reset et redémarre?

fairecasoimeme, 2025-07-08:
Bonjour, et merci pour tout les retours. Je vais me consacrer à temps plein sur le sujet à la mi juillet. 
Pour résumer : 
* problème existant avec pile ou secteur 
* Problème existant en passant par routeur ou en direct.
* Pas de lien significatif avec le coordinateur zigbee.

De mon côté, je n'ai jamais réussi à reproduire le problème, même en jouant avec le rejoin. Les  remontées repartent direct, même avec 2 semaines hors ligne.

Je vais revoir toute la partie rejoin et appairage pour améliorer/optimiser. Je vais développer un garde fou pour reset lorsque le zipulses est sorti du réseau.

N'hésitez pas à me relancer si j'ai pas donner de nouvelles d'ici début août.

Merci encore pour la contribution.

elafargue, 2025-07-09:
Si ça peut aider, mon coordinateur est: ZStack3x0. Revision: 20240710, sur un ZBDongle-P.

fairecasoimeme, 2025-07-11:
Si ce n'est pas déjà fait, pouvez-vous regarder si vous êtes sur la dernière version du firmware de votre coordinateur. Il me semble que sur ce dongle il y a eu des corrections pour avoir un réseau plus stable

devildant, 2025-07-11:
> Si ce n'est pas déjà fait, pouvez-vous regarder si vous êtes sur la dernière version du firmware de votre coordinateur. Il me semble que sur ce dongle il y a eu des corrections pour avoir un réseau plus stable

c'est l'avant dernière version, une nouvelle version est sortie hier, va falloir attendre avant de la faire ^^ 
la 20240710 est la plus stable.

En complément je peux ajouter des précisions sur la dernière fois ou j'ai rencontré le bug avec une alimentation secteur (5v)
le zipulse a commencer a déraillé lorsque j'ai eu 2 coupure de courant a 1h d'interval.
j'ai un maillage assez gros (40 routeur pour un 40m2 ^^) et la il a commencer dans un premier temps a ne plus comptabilisé les impulsion, il est apparu hors ligne au bout de 24h du coup j'ai perdu une journée de stat mais c'est comme ca, suite a quoi j'ai du le couper électriquement pour qu'il se réamorce et se remettent a compter les impulsion et a communiquer, au bout d'une semaine, il a de nouveau été considéré comme hors ligne il ne communiquait plus avec les routeur, j'ai du le re paired

j'en ai 2 cote a coté et c'est seulement l'un des 2 qui est partie en sucette.

a noté également que pendant cette phase j'avais des messages de join assez fréquent sur Z2M pour ce zipulse.
je soupçonne que le zipulse tente des join fréquemment si il rencontre un souci avec le routeur mais passé un certains nombre il part au fraise, ce problème est un peux différent du poste initial mais je pense qu'il sont lié.

selon la fluctuation du réseau (interférence du voisinage/wifi et j'en passe) si la connexion est pas trop parasité tout va bien, mais des que le zipulse commence a rentré dans un cycle de join ca part très vite en offline sur le reseau zigbee, j'avais fait le même constat sur mon zipulse sur pile

j'espère que cette observation pourra aider un peu :)

fairecasoimeme, 2025-07-28:
> c'est l'avant dernière version, une nouvelle version est sortie hier, va falloir attendre avant de la faire ^^ la 20240710 est la plus stable.
>
> En complément je peux ajouter des précisions sur la [...]

Merci la piste est très intéressante. Je travaille dessus

fairecasoimeme, 2025-07-30:
Bonjour à tous,

est-il possible que vous testiez ce firmware qui je l'espère corrige le phénomène sur le rejoin ?
[ZiPulses_v9.0.zip](https://github.com/user-attachments/files/21504111/ZiPulses_v9.0.zip)

neoback45, 2025-08-01:
> Bonjour à tous,
> 
> est-il possible que vous testiez ce firmware qui je l'espère corrige le phénomène sur le rejoin ? [ZiPulses_v9.0.zip](https://github.com/user-attachments/files/21504111/ZiPulses_v9.0.zip)

Bonjour,
Je testerais a l'occasion. Depuis le remplacement de mon zipulses j'ai également fait une mise a jour de mon coordinateur pour avoir une compatibilité avec un nouvelle équipement zigbee. Depuis je n'ai pas eu de soucis avec le zipulses...
Cordialement

urbatecte, 2025-08-15:
Bonjour,

@fairecasoimeme j'ai testé la v9 suite au problème de vidage de CR4050 sur un zipulse acheté en juin 😓.
Alors, ça fonctionne quelques heures puis la led bleue reste allumée en permanence et plus de transmission, un appui sur RST rétablie immédiatment la trnsmission.
J'en ai profité pour mettre à jour mon coordinateur avec son dernier firmware mais j'ai le même phénomène.
Je peux transmettre des logs (mais je ne sais pas comment les récupérer 🤦‍♂️).

En tout cas, merci pour le boulot 🙏

devildant, 2025-10-04:
hi @fairecasoimeme,
je me suis mi aussi en teste de la v9 suite a une nouvelle déconnection...
par contre malgré le flash toujours en 4000-0008 ? 
sinon malgré le flash il est toujours capricieux, la led s'allume quand elle veut quand j'appui sur le bouton (le plus souvent elle s'allume pas), il y a un moyen de transférer la valeur de compteur energy a un nouveau zipulse ?

neoback45, 2025-10-08:
> Bonjour, Je testerais a l'occasion. Depuis le remplacement de mon zipulses j'ai également fait une mise a jour de mon coordinateur pour avoir une compatibilité avec un nouvelle équipement zigbee. [...]

Finalement même soucis. Je n'ai pas tester la V9 car je vois que les problèmes de déconnexion sont toujours la.

fairecasoimeme, 2025-10-08:
> > hi [@fairecasoimeme](https://github.com/fairecasoimeme),
> > je me suis mi aussi en teste de la v9 suite a une nouvelle déconnection...
> > par contre malgré le flash toujours en 4000-0008 ?
> > sinon malgré le flash il est toujours capricieux, la led s'allume quand elle veut quand j'appui sur le bouton (le plus souvent elle s'allume pas), il y a un moyen de transférer la valeur de compteur energy a un nouveau zipulse ?
> 
> Sinon ma question ? :p

Non pas de possibilité de transférer la valeur de compteur energy. (il faut que je rende Writable le cluster/attribut mais la norme Zigbee 3.0 ne le permet pas normalement)
le fait que la version ne change pas sur Z2M c'est parce qu'il n'y a pas eu de read attribute après la MAJ. Peut etre en relancant la config (bouton orange) et en appuyant sur le bouton du ZiPulses pour le réveiller devrait normalement corriger si Z2M lance le read attribute. Peut etre aussi en utilisant la console dev et en lançant manuellement le read attribute sur le cluster 0x0000 et l'attribut SWBuild 0x4000
le fait que la Led ne s'allume pas est parfois dû à un ZiPulses trop sollicité. Vous avez la possibilité avec la V9.0 de brancher le module TTL sur un ordinateur et voir les debug sur le port COM avec la config 115200 8N1

devildant, 2025-10-08:
Observation mon zipulse capricieux semble bien se comporter si je le reset complètement, il répond de nouveau correctement, je me demande si les maj faites en preservant les donné ne rendent pas le zipulse instable

devildant, 2025-10-08:
> Non pas de possibilité de transférer la valeur de compteur energy. (il faut que je rende Writable le cluster/attribut mais la norme Zigbee 3.0 ne le permet pas normalement)
> le fait que la version ne [...]

Merci pour ses infos, j'en remplacer mon zipulse problématique par un de mes spares et il se comporte normalement.

Comme cité dans mon message precedent, celui problématique je l'ai complètement reset en refaisant la maj et depuis il répond bien.
Dorénavant je ferais des reset a chaque fois avec les maj, bien que le zipulse reparte a zero le dashboard energy de ha ne prend que les ajouts donc pas de fluctuation foireuse (avec une grosse valeur negative)

fairecasoimeme, 2025-10-16:
Après plus de 10h de déconnexion avec des impulsions toutes les secondes, le ZiPulses n'a pas planté.
J'ai rebranché le sonoff dongle P, relancer Z2M et le ZiPulses a bien rejoint. Côté Z2M, les données ont été mises à jour.
Donc voici le firmware pour que vous puissiez tester : 

[ZiPulses_v10.0.zip](https://github.com/user-attachments/files/22942866/ZiPulses_v10.0.zip)

devildant, 2025-10-16:
> Après plus de 10h de déconnexion avec des impulsions toutes les secondes, le ZiPulses n'a pas planté. J'ai rebranché le sonoff dongle P, relancer Z2M et le ZiPulses a bien rejoint. Côté Z2M, les [...]

j'ai mi a jours 2 de mes 3 zipulses en version 10, et je confirme deja que c'est deja moins galère (je peine toujours a faire rétablir la connexion apres une maj, la ca a galéré quelque seconde et c'est repartie)

[image]

le 3ieme est dans les partie commune et faut tout que je démonte pour le mettre a jours :(

neoback45, 2025-10-16:
Bonjour,
Zipulses à jour en v10.0! 🤞

fairecasoimeme, 2025-10-26:
> Top ! Existe-t-il une possibilité de mettre à jour depuis MacOS (je n’ai pas de Windows et que des Linux virtuels)

non la méthode recommandée par le fabricant du MCU est d'utiliser l'outil de flash sous windows. les sources du flasher ne sont pas disponible au public ...

> Bonjour, Moi de mon coté en V10 pas de déconnexion pour le moment! En revanche j'ai environ 1 perte d'impulsions tous les 2 jours environs.

Je vais investiguer mais compliqué à débugger