fairecasoimeme/ZiPulses #50: Batterie épuisé après quelques heures

ArthurKnoep, 2025-03-07:
Bonjour,

J'ai acheté un ZiPulses il y a quelques semaine pour pouvoir mesurer la consommation de gaz chez moi. Ma chaudière est branché sur un compteur Samgas G4/2001, j'ai donc acheté en même temps cet émetteur d'impulsion: https://www.compteur-energie.com/gaz-emetteurs-impulsions-samgas-samgas-g4.htm (simple interrupteur ILS)

Après avoir galéré quelques jours pour trouver une pile CR2450 (difficile à trouver en magasin), j'ai mis en place le ZiPulses en branchant mon emetteur d'impulsion sur les bornes `pulse` et en positionnant le switch de l'alimentation sur BAT.

J'ai eu quelques difficulté pour régler le multiplier/divisor (mon compteur envoyant une impulsion pour 0.01 m3, je dois régler le diviser sur 100). Zigbee2mqtt ne cessé d'afficher une erreur de timeout lorsque j'utilisais le bouton "reconfigure" dans l'interface après avoir set mon divisor (j'ai bien fait en sorte d'appuyer sur le bouton link après le reconfigure). Mais j'imagine que ça a fini par fonctionner car j'obtenais bien des données cohérentes après quelques minutes d'utilisation.

Cependant, après seulement une journée, la report du pourcentage de la batterie est tombé à 0% et j'ai cessé de recevoir la mesure de la consommation de gaz dans l'heure qui a suivi.

Voici le graph de la batterie de jeudi à vendredi:
[image]

J'ai utilisé de nombreuse fois le bouton link lors de mes tests d'appairage et de configuration pour le multiplier.
La pile que j'ai utilisé est une pile neuve de la marque Varta.
La version logiciel rapporté par zigbee2mqtt est la 4000-0008

Est il réellement possible que j'ai vidé la pile en utilisant le bouton link (je pense l'avoir utilisé entre 100 et 200 fois maximum) ? Y a t il un problème avec mon unité ?

ArthurKnoep, 2025-03-08:
Pour l'update, j'ai retiré la pile ce matin et je mesure 1.6V au voltmètre, elle semble en effet épuisé

fairecasoimeme, 2025-03-10:
bonjour,
Oui utilisé la fonction Link est très consommateur.
Fred

closed 2025-03-10 by fairecasoimeme

ArthurKnoep, 2025-03-24:
Bonjour,

J'ai installé la deuxième pile du pack que j'avais acheté (neuve aussi), j'ai bien fait attention cette fois ci à ne pas utiliser le bouton link outre mesure (juste une ou deux fois pour join mon réseau)
Et pareil, après 1 jours de fonctionnement environ j'ai cessé de recevoir le comptage des impulsions. En enlevant la pile hier matin je mesurait 2.4V, elle semble donc morte aussi.

Serait il possible que la capteur que j'utilise soit le responsable ? (https://www.compteur-energie.com/gaz-emetteurs-impulsions-samgas-samgas-g4.htm)

fairecasoimeme, 2025-03-24:
Bonjour,
2 possibilités, soit les piles utilisées (chinoises) ne sont pas de bonnes factures, soit l'entrée du zipulses est endommagée et du coup consomme plus qu'à la normale.
Vous pouvez me contacter sur mon mail contact[at]lixee.fr pour faire un échange ou analyse de votre produit.

ArthurKnoep, 2025-03-25:
Bonjour,

Les 2 piles venait d'un pack Varta acheté en grande surface, a priori c'est une marque assez réputé. Je pencherai plutôt pour l'autre solution mais je ne semble pas voir de problème de soudure sur la carte.
Je vous recontacte sur votre adresse mail pour organiser une analyse/échange éventuellement